lvgl/lvgl #3060: unknown driver letter

lulersoft, 2022-01-29:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

output:

debug letter: 4,34; A, 41
[Warn]	(0.807, +807)	 lv_fs_open: Can't open file (A:/spiffs/bg.png): unknown driver letter 	(in lv_fs.c line #68)

guess：
The ASCII 41 character of 'A' seems to be truncated, leaving only 4?

### To Reproduce

#define CONFIG_LV_FS_STDIO_LETTER "A"
#define CONFIG_LV_FS_STDIO_PATH ""
#define CONFIG_LV_FS_STDIO_CACHE_SIZE "128"

```c

lv_fs_drv_t * lv_fs_get_drv(char letter)
{
    lv_fs_drv_t ** drv;

    _LV_LL_READ(&LV_GC_ROOT(_lv_fsdrv_ll), drv) {   
        printf("debug letter: %c,%x; %c, %x\n",(*drv)->letter,(*drv)->letter,letter,letter);             
        if((*drv)->letter == letter) {
            return *drv;
        }
    }

    return NULL;
}

```
### Expected behavior

### Screenshots or video

embeddedt, 2022-01-29:
I think `CONFIG_LV_FS_STDIO_LETTER` should be a character, not a string (so use single quotes).

lulersoft, 2022-01-29:
> I think `CONFIG_LV_FS_STDIO_LETTER` should be a character, not a string (so use single quotes).

I can manually modify the SDK config file generated automatically by IDF menuconfig, and it is normal. thank you.

Normally, it is not necessary to manually modify the generated file. Is it a bug???

kisvegabor, 2022-01-29:
The type of that field should be `int` and you can set the ascii code of the character.

closed 2022-02-03 by lulersoft