lvgl/lvgl #3065: V8.1 SDL GPU Img always cached, no way to invalidate

iankellogg, 2022-02-01:
### Perform all steps below and tick them with [x]
- [ x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x] Update lvgl to the latest version
- [ x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
When using latest SDL GPU commits that fix SDL GPU drawing, images are created into a texture cache regardless of image cache state. That itself makes sense and shouldn't be an issue except it breaks any animation or other widget/library that uses image containers.
Example, FFMPEG only shows first frame.
A button that updates an image source when pressed never updates

My Current workaround is to add another function to lv_draw_ctx_t that provides cache invalidation on the texture

### To Reproduce
Enable SDL GPU
Create FFMPEG player

or
Create image
set source to one image
add button with event handler that updates image source

### Expected behavior
When updating image source, expected behavior is to show updated image

kisvegabor, 2022-02-01:
cc @mariotaku

mariotaku, 2022-02-01:
I have a few suggestions to this situation.

1. Use `frame_id` in `lv_draw_img_dsc_t`.
2. Pass `SDL_Texture` directly. This is highly recommended although not properly documented yet, and I'll explain below.

## Use SDL_Texture as image source

`lv_sdl_img.h`

```c
#pragma once

#include "lvgl.h"

#include <SDL.h>

#define LV_SDL_IMG_HEAD "!SDLIMG:"

typedef enum {
    LV_SDL_IMG_TYPE_PATH,
    LV_SDL_IMG_TYPE_CONST_PTR,
    LV_SDL_IMG_TYPE_TEXTURE,
} lv_sdl_img_src_type_t;

typedef struct {
    lv_coord_t w, h;
    lv_img_cf_t cf: 8;
    lv_sdl_img_src_type_t type: 8;
    union {
        void *pointer;
        const void *constptr;
        const char *path;
        SDL_Surface *surface;
        SDL_Texture *texture;
    } data;
    unsigned int data_len;
    SDL_Rect rect;
} lv_sdl_img_src_t;

#define LV_SDL_IMG_LEN (8 + sizeof(lv_sdl_img_src_t) * 4 / 3 + 4)

void lv_sdl_img_decoder_init(lv_img_decoder_t *decoder);

void lv_sdl_img_src_stringify(const lv_sdl_img_src_t *src, char *str);

void lv_sdl_img_src_parse(const char *str, lv_sdl_img_src_t *src);
````

`lv_sdl_img.c`

```c
#include "lv_sdl_img.h"

#include <mbedtls/base64.h>

#include <SDL_image.h>
#include "draw/sdl/lv_draw_sdl_utils.h"
#include "draw/sdl/lv_draw_sdl_texture_cache.h"

lv_res_t sdl_img_decoder_info(struct _lv_img_decoder_t *decoder, const void *src, lv_img_header_t *header);

lv_res_t sdl_img_decoder_open(struct _lv_img_decoder_t *decoder, struct _lv_img_decoder_dsc_t *dsc);

void sdl_img_decoder_close(struct _lv_img_decoder_t *decoder, struct _lv_img_decoder_dsc_t *dsc);

static bool is_sdl_img_src(const void *src);

static lv_draw_sdl_dec_dsc_userdata_t *lv_gpu_sdl_dec_dsc_userdata_new();

void lv_sdl_img_decoder_init(lv_img_decoder_t *decoder) {
    IMG_Init(IMG_INIT_JPG | IMG_INIT_PNG);
    lv_img_decoder_set_info_cb(decoder, sdl_img_decoder_info);
    lv_img_decoder_set_open_cb(decoder, sdl_img_decoder_open);
    lv_img_decoder_set_close_cb(decoder, sdl_img_decoder_close);
}

void lv_sdl_img_src_stringify(const lv_sdl_img_src_t *src, char *str) {
    SDL_memcpy(str, LV_SDL_IMG_HEAD, 8);
    size_t olen;
    int ret = mbedtls_base64_encode((unsigned char *) &str[8], LV_SDL_IMG_LEN - 8, &olen,
                                    (const unsigned char *) src, sizeof(lv_sdl_img_src_t));
    LV_ASSERT(ret == 0);
    (void) ret;
}

void lv_sdl_img_src_parse(const char *str, lv_sdl_img_src_t *src) {
    size_t olen, slen = SDL_strlen(str) - 8;
    int ret = mbedtls_base64_decode((unsigned char *) src, sizeof(lv_sdl_img_src_t), &olen,
                                    (unsigned char *) &str[8], slen);
    LV_ASSERT(ret == 0);
    (void) ret;
}

lv_res_t sdl_img_decoder_info(struct _lv_img_decoder_t *decoder, const void *src, lv_img_header_t *header) {
    if (!is_sdl_img_src(src)) {
        return LV_RES_INV;
    }
    lv_sdl_img_src_t sdl_src;
    lv_sdl_img_src_parse(src, &sdl_src);
    header->w = sdl_src.w;
    header->h = sdl_src.h;
    header->cf = sdl_src.cf;
    return LV_RES_OK;
}

lv_res_t sdl_img_decoder_open(struct _lv_img_decoder_t *decoder, struct _lv_img_decoder_dsc_t *dsc) {
    if (!is_sdl_img_src(dsc->src)) {
        return LV_RES_INV;
    }
    lv_sdl_img_src_t sdl_src;
    lv_sdl_img_src_parse(dsc->src, &sdl_src);
    lv_draw_sdl_dec_dsc_userdata_t *userdata = lv_gpu_sdl_dec_dsc_userdata_new();
    switch (sdl_src.type) {
        case LV_SDL_IMG_TYPE_TEXTURE: {
            userdata->texture = sdl_src.data.texture;
            userdata->texture_managed = true;
            break;
        }
        case LV_SDL_IMG_TYPE_PATH: {
            lv_disp_drv_t *driver = _lv_refr_get_disp_refreshing()->driver;
            SDL_Renderer *renderer = ((lv_draw_sdl_drv_param_t *)driver->user_data)->renderer;
            SDL_Texture *texture = IMG_LoadTexture(renderer, sdl_src.data.path);
            userdata->texture = texture;
            break;
        }
        case LV_SDL_IMG_TYPE_CONST_PTR: {
            lv_disp_drv_t *driver = _lv_refr_get_disp_refreshing()->driver;
            SDL_Renderer *renderer = ((lv_draw_sdl_drv_param_t *)driver->user_data)->renderer;
            SDL_RWops *src = SDL_RWFromConstMem(sdl_src.data.constptr, (int) sdl_src.data_len);
            SDL_Texture *texture = IMG_LoadTexture_RW(renderer, src, 1);
            userdata->texture = texture;
            break;
        }
        default: {
            SDL_free(userdata);
            return LV_RES_INV;
        }
    }
    userdata->rect = sdl_src.rect;
    dsc->img_data = NULL;
    dsc->user_data = userdata;
    return LV_RES_OK;
}

void sdl_img_decoder_close(struct _lv_img_decoder_t *decoder, struct _lv_img_decoder_dsc_t *dsc) {
    if (!is_sdl_img_src(dsc->src)) {
        return;
    }
    lv_sdl_img_src_t sdl_src;
    lv_sdl_img_src_parse(dsc->src, &sdl_src);
    lv_draw_sdl_dec_dsc_userdata_t *userdata = dsc->user_data;
    if (!userdata) {
        return;
    }
    if (!userdata->texture_referenced) {
        SDL_DestroyTexture(userdata->texture);
    }
    SDL_free(userdata);
}

static inline bool is_sdl_img_src(const void *src) {
    return ((char *) src)[0] == '!' && SDL_memcmp(src, LV_SDL_IMG_HEAD, 8) == 0;
}

static lv_draw_sdl_dec_dsc_userdata_t *lv_gpu_sdl_dec_dsc_userdata_new() {
    lv_draw_sdl_dec_dsc_userdata_t *userdata = SDL_malloc(sizeof(lv_draw_sdl_dec_dsc_userdata_t));
    SDL_memset(userdata, 0, sizeof(lv_draw_sdl_dec_dsc_userdata_t));
    SDL_memcpy(userdata->head, LV_DRAW_SDL_DEC_DSC_TEXTURE_HEAD, 8);
    return userdata;
}

```

You will need to register image decoder above to lvgl, and construct `lv_sdl_img_src_t` like below:

```c
lv_sdl_img_src_t frame_src = {
  w = 480, h = 270,
  cf = LV_IMG_CF_TRUE_COLOR,
  type = LV_SDL_IMG_TYPE_TEXTURE,
  data.texture = texture,
};
```

Then stringify struct with `lv_sdl_img_src_stringify`, and pass it as image source.

Since you are using FFMPEG, `SDL_UpdateYUVTexture` is much more efficient than processing pixel array.

embeddedt, 2022-02-23:
@kisvegabor Not sure if you missed the message above, but I don't think this is resolved yet.

kisvegabor, 2022-02-24:
Sorry, somehow I completely missed this conversation.

> Pass SDL_Texture directly. This is highly recommended although not properly documented yet, and I'll explain below.

This solution seems local to the SDL backend. Is this true? If so I suggest going into this direction for now. There is a [PR about reworking the image decoder interface](https://github.com/lvgl/lvgl/pull/2904#issuecomment-1047616125) in which `lv_img_set_src` will be updated to `lv_img_set_src(img, src, LV_IMG_SRC_FILE/RAW/SYMBOL)`. Once it's merged we can add `LV_IMG_SRC_SDL_TEXTURE` too. 

If it will be added we really should have a documentation for the SDL backend too, as suggested in https://github.com/lvgl/lvgl/issues/3126

closed 2022-03-18 by lvgl-bot