lvgl/lvgl #3069: Gridnav does not provide a way to set the initially focused object

FoundationKen, 2022-02-03:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
The `gridnav` feature does not provide a way to set the initially focused object.

### To Reproduce
Create a `gridnav` with `lv_gridnav_add()` and add some objects to it.  The first focusable object always receives focus, even if you call `lv_group_focus_obj(desired_obj)`.
 
### Expected behavior
Option 1: Add a parameter like `focused_obj` `to` `lv_gridnav_add()`.  Not ideal as it changes the API.
Option 2: Add a new function like `lv_gridnav_add_with_focus()` that accepts an additional `focused_obj` parameter and sets `dsc->focused_obj = focused_obj`.  The existing `lv_gridnav_add()` can forward to this new function.
Option 3: Have `lv_gridnav_add()` look to see if there is a child in the object that is focused already and set `focused_obj` to that child.

### Screenshots or video
I tried Option 2 quickly since it was easy.  This worked, but having two APIs like this is not so clean.  It does avoid having to search through the children though.

```c
void lv_gridnav_add(lv_obj_t *obj, lv_gridnav_ctrl_t ctrl)
{
    lv_gridnav_add_with_focus(obj, ctrl, NULL);
}

void lv_gridnav_add_with_focus(lv_obj_t *obj, lv_gridnav_ctrl_t ctrl, lv_obj_t *focused_obj)
{
    lv_gridnav_remove(obj); /*Be sure to not add gridnav twice*/

    lv_gridnav_dsc_t *dsc = lv_mem_alloc(sizeof(lv_gridnav_dsc_t));
    LV_ASSERT_MALLOC(dsc);
    dsc->ctrl = ctrl;
    dsc->focused_obj = focused_obj;
    lv_obj_add_event_cb(obj, gridnav_event_cb, LV_EVENT_ALL, dsc);

    lv_obj_clear_flag(obj, LV_OBJ_FLAG_SCROLL_WITH_ARROW);
}
```

kisvegabor, 2022-02-07:
What bout a function like `lv_gridnav_set_focused(lv_obj_t * cont, lv_obj_t * to_focus)`?

FoundationKen, 2022-02-10:
That sounds great as long as it takes effect before the first rendering, which I think it would.

kisvegabor, 2022-02-17:
I've just added `lv_gridnav_set_focused()`. Yes, it will be applied before the first rendering.

FoundationKen, 2022-02-23:
Awesome!  Thank you!  I'm using it now.