lvgl/lvgl #3072: Missing EVENT_CLOSE

X-Ryl669, 2022-02-03:
### Describe the feature request

Since LVGL is more and more used as an desktop app too, I think it's necessary to have a clean application exit event too.
Right now, all example have a `while(1) { }` in their main function so when you close the app (by clicking the X icon, or Alt+F4 or ...), you're actually leaking all the application objects. 
For most of them, it's not important, but some object might have LV_EVENT_DELETE handler that are not called. Those handler might, in turn, have side effects (like flushing some file buffer to disk, shutting down a nuclear reactor before leaving, or just printing "Bye world" on the console).

So, I think an LV_EVENT_QUIT should be added that'd be triggered by SDL_WINDOWEVENT_CLOSE (for SDL) or whatever closing protocol. An a default handler storing a boolean (`should_quit`) should be attached to this event.
A `lv_should_quit()` function should be added, returning the boolean, and examples' *while* loop be modified to call this function.

embeddedt, 2022-03-09:
I think it can be left open; it's an excellent first contribution for someone to make if they're looking for an idea.

kisvegabor, 2022-03-09:
I agree!

geekheart, 2022-06-27:
Hi, I think [x-track](https://github.com/FASTSHIFT/X-TRACK/tree/main/Software/X-Track/USER/App/Utils/PageManager)'s page management framework might be helpful for your discussion. I tried porting the pure c version on [7.11](https://github.com/geekheart/page_manager).This page management framework draws on the page management framework of ios, making page loading and exiting very convenient

kisvegabor, 2022-06-28:
@geekheart how is it connected to `EVENT_QUIT`?

geekheart, 2022-06-29:
The page manager divides each page into a life cycle.  Users need to connect the following functions: on_custom_attr_config(), on_view_load(), on_view_did_load(), on_view_will_appear(), on_view_did_appear(), on_view_will_disappear(), on_view_did_disappear(), on_view_did_unload().

geekheart, 2022-06-29:
The page manager divides each page into a life cycle. Users need to connect the following functions: <br/> on_custom_attr_config(),  <br/> on_view_load(), on_view_did_load(), <br/> on_view_will_appear(),  <br/> on_view_did_appear(), <br/> on_view_will_disappear(), <br/> on_view_did_disappear(), <br/> on_view_did_unload() <br/> PageManager::push and PageManager::pop for page entry and exit. <br/>

geekheart, 2022-06-29:
At the beginning of page creation, on_custom_attr_config() will be run. At this time, the root page is not created. In this callback, the attributes of the page are mainly modified. For example, the animation method of entering the page and exiting the page, whether the page is cached, etc. <br\> At this point the page has two modes. The first is to not clear the page control cache, which is suitable for those with more memory and focusing on switching efficiency. There is also a clear page control cache, suitable for MCUs with less memory. on_view_load() and on_view_did_load() in the first method will only run on the first load, and will not enter this function after that. They represent callbacks for what the user should do before and after the page root is created, respectively.<br/>on_view_will_appear() and on_view_did_appear() are called before the page animation plays and after the page animation plays. Of course, this part of the animation has been set internally, and the user only needs to modify some properties to easily set the animation.<br/>After the page's PageManager::pop is triggered by a button callback or a timer callback, the page enters on_view_will_disappear() and
in the on_view_did_disappear() function. on_view_will_disappear() will be called by the user before the control and page disappear. on_view_did_disappear() will be called after the page switching animation. Users can implement EVENT_QUIT through the implementation of these two functions (I don't know if I understand EVENT_QUIT and your discussion are the same)<br/>Finally, if no page controls are cached, on_view_did_unload() is called to unload the page. Users can free some variables created by themselves here. If there is a cached page control it will never go to on_view_did_unload(). <br/>Thank you very much for being here, since I am using Google Translate to explain the problem with you guys, so I don't know if the translated content can express what I want to explain. I don't know if the EVENT_QUIT I understood before is the same as the page exit. Just for the sharing of this page manager, I hope it can help you get more inspiration.

kisvegabor, 2022-07-01:
What you described is quite similar to [Fragment](https://docs.lvgl.io/master/others/fragment.html), or am I missing something?

kisvegabor, 2023-12-02:
It can be resolved like this: https://github.com/lvgl/lvgl/blob/master/src/dev/sdl/lv_sdl_window.c#L257

So I close this issue

closed 2023-12-02 by kisvegabor