lvgl/lvgl #3075: obj_test_task_cb: mem leak since start: 56, frag: 1% (in lv_demo_stress.c)

Viatorus, 2022-02-05:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Possible memory leak when running the lvgl_stress demo under visual studio.

Possible because: I am new to LVGL and I am unsure if this is supposed to be as it should.

### To Reproduce

1. clone: https://github.com/lvgl/lv_sim_visual_studio
1. comment lvgl_demo_widget() in `LVGL.Simulator`
1. uncomment lvgl_demo_stress() in `LVGL.Simulator`
1. run application in x86 Debug or Release

### Expected behavior

No memory leaks at all.

BUT if this should be the case, a comment on this line would be great:
https://github.com/lvgl/lvgl/blob/master/demos/stress/lv_demo_stress.c#L77

### Screenshots or video

[image]

kisvegabor, 2022-02-07:
It seems normal, the gist is that it's not growing.

> BUT if this should be the case, a comment on this line would be great

True! I added a comment about it here:
https://github.com/lvgl/lvgl/blob/master/demos/stress/lv_demo_stress.c

Viatorus, 2022-02-07:
Great! Thank you for your fast reply and improvement!

closed 2022-02-07 by Viatorus