lvgl/lvgl #3078: GCC bug 80753 means missing lv_conf.h is silently ignored with LV_CONF_INCLUDE_SIMPLE

bsilver8192, 2022-02-08:
### Perform all steps below and tick them with [x]
- [X ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ X] Update lvgl to the latest version
- [X ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

https://gcc.gnu.org/bugzilla/show_bug.cgi?id=80753 causes the operations performed in `lv_conf_internal.h` to silently ignore a missing `lv_conf.h`. It uses `__has_include("lv_conf.h")` and then `#include "lv_conf.h"` even if that fails, and GCC will silently ignore the later `#include` not finding a file.

I ran into this due to path confusion in my build system that placed my `lv_conf.h` in the wrong place. Due to this behavior, it took a long time to figure out what was happening.

### To Reproduce

1. Configure with `-DLV_CONF_INCLUDE_SIMPLE=1` and GCC.
2. `make` (without creating a lv_conf.h)

### Expected behavior

Fails to build because the file is missing.

### Screenshots or video

embeddedt, 2022-02-08:
Hmm. The cleanest workaround I can think of that won't inconvenience too many custom `lv_conf.h` files is something like this:

```diff
    #elif defined(LV_CONF_INCLUDE_SIMPLE)         /*Or simply include lv_conf.h is enabled*/
        #include "lv_conf.h"
    #else
        #include "../../lv_conf.h"                /*Else assume lv_conf.h is next to the lvgl folder*/
    #endif
#endif

+#if !defined(LV_CONF_H) && !defined(LV_CONF_SUPPRESS_DEFINE_CHECK)
+#warning LV_CONF_H define not found, have you encountered GCC bug #80753?
+#endif
```

Anyone who gets the warning and doesn't have the issue can just add `LV_CONF_SUPPRESS_DEFINE_CHECK` to their build system flags.

bsilver8192, 2022-02-08:
That's better than anything I came up with. It's a really annoying bug...

Maybe also add a comment explaining how that bug leads to __include_next suppressing any failures from including that file, and a link to the bugzilla entry because some quick Google searching based on "GCC bug #80753" doesn't find it for me. Also, maybe avoid the `,` and `?` in the `#warning` message to minimize compatibility issues, I'm not sure about this specifically but in general C preprocessors can be really finicky about punctuation.

kisvegabor, 2022-02-08:
@embeddedt Wonderful idea!

A few comments:
- Be sure the new check is in the `#ifndef LV_CONF_SKIP` block
- IMO the warning should say `#warning Failed to include lv_conf.h.  
- Add comments above the warning as @bsilver8192 suggested

What do you think?

embeddedt, 2022-02-08:
See #3082.

commit cced17fb (2022-02-08): fix(conf) work around GCC bug

Fixes #3078
commit ae138316 (2022-02-08): fix(conf) work around GCC bug

Fixes #3078
PR #3082 (2022-02-11): fix(conf) work around GCC bug
Fixes #3078

### Checkpoints
- [x] Follow the [styling guide](https://github.com/lvgl/lvgl/blob/master/docs/CODING_STYLE.md)
- [x] Run `code-format.py` from the `scripts` folder. [astyle](http://astyle.sourceforge.net/install.html) needs to be installed.
- [ ] Update the documentation
closed 2022-02-11 by kisvegabor