lvgl/lvgl #3080: The __STDC_VERSION__ macro in lv_draw_rect_dsc_t causes from cpp language to c language pass parameters error

guoweilkd, 2022-02-08:
### Describe the bug

__STDC_VERSION__  macro  is not defined  when compiling C++. both c++ and c are used in my project.  I get into trouble when I write the following code in CPP.
```cpp
void draw_event_cb(lv_event_t* e)
{
    lv_obj_t* obj = lv_event_get_target(e);
    lv_event_code_t code = lv_event_get_code(e);
    if (code == LV_EVENT_DRAW_MAIN) {
        lv_draw_ctx_t* draw_ctx = lv_event_get_draw_ctx(e);
        lv_area_t test_coords;
        lv_area_set(&test_coords, 50, 50, 100, 100);
        lv_draw_rect_dsc_t test_dsc;
        lv_draw_rect_dsc_init(&test_dsc);
        test_dsc.bg_opa = LV_OPA_MIN;
        test_dsc.outline_opa = LV_OPA_MIN;
        test_dsc.shadow_opa = LV_OPA_MIN;
        test_dsc.bg_img_opa = LV_OPA_MIN;
        test_dsc.border_color = LV_COLOR_MAKE(255,0,0);
        test_dsc.border_width = 1;
        test_dsc.border_opa = 10;
        lv_draw_rect(draw_ctx, &test_dsc, &test_coords);
    }
}

void obj_draw_test(void)
{
    lv_obj_t * obj = lv_obj_create(lv_scr_act());
    lv_obj_set_size(obj, 200, 200);
    lv_obj_add_event_cb(obj, draw_event_cb, LV_EVENT_ALL, NULL);
}
```
The `lv_draw_rect_dsc_t` structure in the CPP file is different from this structure in C, causing The member value is incorrect.

Why add this macro in [lv_draw_rect_dsc_t](https://github.com/lvgl/lvgl/blob/master/src/draw/lv_draw_rect.h#L38) ？

kisvegabor, 2022-02-08:
This `#if` is added as anonymous unions are not supported in all C versions. 

IMO this optimization is not critical because `lv_draw_rect_dsc_t`s lives only a for a short time and we save only a few bytes with it anyway.

X-Ryl669, 2022-02-08:
Right, that's old code I forgot to remove. We'll need to remove it (and have both bg_color and bg_grad values). Maybe when reaching LVGL9 the former bg_color member can be removed too so at least there's no more a waste of 16 or 32 bits here.

X-Ryl669, 2022-02-08:
As a side note, anonymous union is allowed in C++ too so it might be worth changing the test to 
```cpp
#if __STDC_VERSION__ >= 201112L || __cplusplus >= 201103L
```
So both code will see the same layout.

lishaoenxm, 2022-02-09:
@X-Ryl669 it's better to remote the _#if_ and union in the _lv_draw_rect_dsc_t_, keep all the _bg_color_ and _bg_grad_ in the structure.

guoweilkd, 2022-02-09:
> ```c++
> #if __STDC_VERSION__ >= 201112L || __cplusplus >= 201103L
> ```
It still seems impossible to avoid discrepancies between CPP and C versions, such as using C99 and C ++11.

X-Ryl669, 2022-02-09:
You're right. I'll remove the macro here.

lishaoenxm, 2022-02-16:
@X-Ryl669 Could you provide a patch for this issue?

closed 2022-02-18 by guoweilkd