lvgl/lvgl #3083: [meta] Use pre-commit to run astyle on staged files when commiting

C47D, 2022-02-09:
### Introduce the problem
I tend to forget about formatting my code before pushing it to the remotes, then CI will fail and I will end up having multiple emails complaining about it. I've also pushed commits with test cases that doesn't pass.

### Examples and cases

### Suggested solution

I've found a 'tool' named [pre-commit](https://pre-commit.com/#install) in [Interrupt this post ](https://interrupt.memfault.com/blog/pre-commit) and I've set it up locally, I'm pretty pleased with how it works.

I've recorded a explanation video, please forgive the record setup, let me know if you have additional questions https://youtu.be/E3874kFISbM

kisvegabor, 2022-02-14:
It seems useful. We can add a description to it in the docs and link it in the PR template. 

Can you share your config file too?

C47D, 2022-02-15:
Sure, I can send the configuration file in a PR.

What about mentioning it after the [Commit message format](https://docs.lvgl.io/master/CONTRIBUTING.html#commit-message-format) in the docs?

As this is an optional work flow I'm not sure if it's worth mentioning it in every PR. I use it just to don't forget formatting the code, other people can use other tools. 

What do you think?

kisvegabor, 2022-02-16:
> What about mentioning it after the [Commit message format](https://docs.lvgl.io/master/CONTRIBUTING.html#commit-message-format) in the docs?

I suggest this (too?): https://github.com/lvgl/lvgl/blob/master/docs/CODING_STYLE.md

This file is linked [here](https://docs.lvgl.io/master/CONTRIBUTING.html#introduction) and in the [PR template](https://github.com/lvgl/lvgl/blob/master/.github/pull_request_template.md) too.

> As this is an optional work flow I'm not sure if it's worth mentioning it in every PR.

My experience is that people sometimes ignore notes even in the PR templates. I can hardly believe the vast majority of people will find it in the docs.

PR #3092 (2022-02-24): Add initial pre-commit configuration for code formatting
### Description of the feature or fix

Add pre-commit configuration for code formatting. Fixes #3083 

### Checkpoints
- [ ] Follow the [styling guide](https://github.com/lvgl/lvgl/blob/master/docs/CODING_STYLE.md)
- [ ] Run `code-format.py` from the `scripts` folder. [astyle](http://astyle.sourceforge.net/install.html) needs to be installed.
- [ ] Update the documentation
closed 2022-02-24 by kisvegabor