lvgl/lvgl #3086: Text color style prop doesn't get inherited from a grandparent object

FoundationKen, 2022-02-11:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version (not tested under latest version, but the style inheritance code looks the same)
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
If you create a grandparent object and add style with `text_color` set to red, then add a child object to it and add a label to that child, the label should inherit the red text color from the grandparent, but it does not.

It DOES work if the text color prop is in the immediate parent, but does not work if the text color prop is in the grandparent.
 
### To Reproduce
Simulator link with example:

https://sim.lvgl.io/v8.1/micropython/ports/javascript/index.html?script_startup=https://raw.githubusercontent.com/lvgl/lvgl/0b5a1d4b23975b920ff841ea9cd038802f51711b/examples/header.py&script=https://raw.githubusercontent.com/lvgl/lvgl/0b5a1d4b23975b920ff841ea9cd038802f51711b/examples/styles/lv_example_style_8.py&script_direct=8fb1c2b59e4164f867198e062ed23179ae69db58

### Expected behavior
Style properties should be inherited from the whole parent chain.

### Screenshots or video
N/A

kisvegabor, 2022-02-11:
It's not working because `parent` also sets the text color to black. The inheritance algorithm finds the first parent on which the property is set. 

To achieve what you want you can remove all styles from the parent with `parent.remove_style_all()` (it already makes the text red) and add new styles to the parent which doesn't set the text color.

FoundationKen, 2022-02-11:
Thanks for the explanation.  Makes sense now.

Is there any less aggressive way to clear just the text color prop.  I guess this is a bit tricky since multiple styles could be applied and you don't necessarily want to remove the prop from a style completely (maybe it's shared and the change could affect other objects).

Is there a value you can set for a prop that could mean "inherit from parent" even if other styles are on this object?

That would let you add another style to the object to override any previous styles.  The recursive prop search would find this special value, but instead of returning it, it would recurse to the parent.

kisvegabor, 2022-02-17:
> Is there a value you can set for a prop that could mean "inherit from parent" even if other styles are on this object?

It'd be the ideal "API" for this feature. E.g. in CSS you can do `color:inherit;` So in theory we can use special values. But there are 3 variable types for style properties:
- integer: fine `INT32_MAX` or so should do the job
- pointer: using a pointer to `char inherit_pointer` should work too
- colors: this is the problem. E.g. in ARGB32 there is no space for special values. If you have any idea for this I'd be very happy to hear it.

embeddedt, 2022-02-17:
This is just a couple minutes worth of thought on the issue, but what if we do some refactoring:

1. Currently we store flags for style properties in the property index itself (e.g. [here](https://github.com/lvgl/lvgl/blob/6c005526295aeb277edad42b3a05b0c7e6d72eaf/src/misc/lv_style.h#L165)). Maybe I've missed something, but I don't see why we can't move this information to a `const uint8_t[]` lookup table that is then referenced as needed. It doesn't need to be stored in every single style.
2. Styles are stored in RAM as a list of properties followed by a list of values. However, we now have 5(?) bits of storage per property since we freed up those bits in the property index. We can use those bits to implement `initial` and `inherit`.

With that solution we don't need to store `inherit` as a proper value at all.

FoundationKen, 2022-02-17:
@embeddedt That makes sense to me as a way to store the flag.

The API changes might be more daunting.  There are so many style functions like `lv_style_set_bg_color()`, `lv_style_set_pad_left()`, etc. that don't have the ability to pass in a flag like that.

The alternative might be to just say you have to drop down to a new function like `lv_style_set_prop_inherit(style, LV_STYLE_BG_COLOR)` if you want to do this.  Not sure if that function and constants would be exposed in MicroPython off the top of my head.

Or maybe there's a cleaner way.

For my particular case, I've rearranged the hierarchy to avoid this problem for the moment, but I wouldn't be surprised if I come across it again.

kisvegabor, 2022-02-17:
Sounds like a great idea! We should also consider cases like [this](https://github.com/lvgl/lvgl/blob/master/src/extra/layouts/flex/lv_flex.c#L97), i.e. when registering a  property and adding flags to it. So we should have a function like 
```c
lv_style_prop_set_flags(LV_STLYE_BG_COLOR, LV_STYLE_FLAG_X | LV_STYLE_FLAG_Y)
```
The only drawback is that the lookup will require 256 byte extra flash (assuming max 256 properties).

> The alternative might be to just say you have to drop down to a new function like lv_style_set_prop_inherit(style, LV_STYLE_BG_COLOR) if you want to do this.

This API looks good. It'd be better to have special values for inherit, e.g. `lv_style_set_bg_color(&style, LV_STYLE_INHERIT);` but I think it's not really possible. ...how simple would it be C++ with function overload... 

>  Not sure if that function and constants would be exposed in MicroPython off the top of my head.

Yes, it's exposed as `lv.STYLE.BG_COLOR`.

embeddedt, 2022-02-17:
> We should also consider cases like [this](https://github.com/lvgl/lvgl/blob/master/src/extra/layouts/flex/lv_flex.c#L97), i.e. when registering a property and adding flags to it.

I started refactoring it locally and added a parameter to `lv_style_register_prop` with the flags. (That also opens the possibility of having a regex to help refactor code.) Does that work?

kisvegabor, 2022-02-17:
> I started refactoring it locally and added a parameter to lv_style_register_prop with the flags. (That also opens the possibility of having a regex to help refactor code.) Does that work?

Yes, definitely. :+1:

kisvegabor, 2022-05-28:
@embeddedt it seems only the final step is missing from here. Do you plan to implement it in the near future? If not  I can add it as well.

closed 2022-06-10 by embeddedt

kisvegabor, 2022-06-10:
Awesome! :clap: :clap: :clap:

commit 9a48de0f (2022-06-10): feat(style) add 'inherit' and 'initial' CSS properties (#3390)

Fixes #3086
commit f7d809cc (2022-09-20): feat(style) add 'inherit' and 'initial' CSS properties (#3390)

Fixes #3086