lvgl/lvgl #3087: STM32 DMA2D and display driver flush

patricklaf, 2022-02-11:
### Introduce the problem
I am using LVGL with Arduino on STM32. Prior to last update, I used DMA2D in the display driver with _lv_gpu_stm32_dma2d_copy_ function. With latest 8.2.0, this function doesn't exist anymore and all public functions from DMA2D code are based on _lv_draw_ctx_t_ use.

### Examples and cases
Code working in 8.0.2:
```c
void lvgl_disp_flush(lv_disp_drv_t *display, const lv_area_t *area, lv_color_t *color_p) {
#if LV_USE_GPU_STM32_DMA2D
	lv_gpu_stm32_dma2d_copy(
			(lv_color_t*) tft.getBuffer() + area->y1 * LV_HOR_RES + area->x1,
			(lv_coord_t) LV_HOR_RES, color_p, area->x2 - area->x1 + 1,
			area->x2 - area->x1 + 1, area->y2 - area->y1 + 1);
#else
	tft.drawBuffer(area->x1, area->y1, area->x2 - area->x1 + 1, area->y2 - area->y1 + 1, (uint16_t*)color_p);
#endif
	lv_disp_flush_ready(display);
}
```
### Suggested solution
Is it possible to add a public function to GPU drivers to achieve accelerated buffer copy usable from display driver flush function?

patricklaf, 2022-02-18:
I am currently away from my office and from my board, so I can't test anything for a week.
I will do it when I am back.

I think that it can be something like this in `lv_gpu_stm32_dma2d.c`:
```c
/**********************
 *   GLOBAL FUNCTIONS
 **********************/

void lv_draw_stm32_dma2d_buffer_copy(lv_draw_ctx_t *draw_ctx, void *dest_buf,
		lv_coord_t dest_stride, const lv_area_t *dest_area, void *src_buf,
		lv_coord_t src_stride, const lv_area_t *src_area) {
	LV_UNUSED(draw_ctx);
	lv_draw_stm32_dma2d_blend_map(dest_buf, dest_area, dest_stride, src_buf,
			src_stride, LV_OPA_MAX);
}
```
And in display driver `flush` callback:
```c
void lvgl_disp_flush(lv_disp_drv_t *display, const lv_area_t *area,
		lv_color_t *color_p) {
#if LV_USE_GPU_STM32_DMA2D
	lv_draw_stm32_dma2d_buffer_copy(NULL, tft.getBuffer(),
			(lv_coord_t) LV_HOR_RES, area, color_p, 0,
			(lv_area_t) (0, 0, 0, 0));
#else
	tft.drawBuffer(area->x1, area->y1, area->x2 - area->x1 + 1, area->y2 - area->y1 + 1, (uint16_t*)color_p);
#endif
	lv_disp_flush_ready(display);
}
```
But how to be sure we wait for DMA to be finished as it was done in previous `lv_gpu_stm32_dma2d_copy`?

kisvegabor, 2022-02-21:
> I will do it when I am back.

Great, thanks!

> But how to be sure we wait for DMA to be finished as it was done in previous lv_gpu_stm32_dma2d_copy?

If you mean DMA2D then `ctx->wait_for_finish` should do the job.

patricklaf, 2022-03-04:
I'm back.

It mostly works but with some glitches. I think that it is related to DMA2D, I can not call `ctx->wait_for_finish` because `ctx` is not available from my `disp_flush` function.

Is there a global function to get the current `ctx` to be able to call `wait_for_finish`.

kisvegabor, 2022-03-04:
> Is there a global function to get the current ctx to be able to call wait_for_finish.

There is no API for it now, but you can get the `ctx` like this: `disp_drv->draw_ctx`

patricklaf, 2022-03-13:
Issue solved.
Thanks!

closed 2022-03-13 by patricklaf