lvgl/lvgl #3090: Can't display gif that has width more than 140px

phamvant, 2022-02-13:
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

kisvegabor, 2022-02-14:
Can it be an out of memory issue?
Please enable logging as described here: https://docs.lvgl.io/master/porting/log.html

phamvant, 2022-02-14:
[image]

I got this

embeddedt, 2022-02-14:
Your heap size is too small.

kisvegabor, 2022-02-16:
What it the current size? 
It should be  at least 40 * 1024.

embeddedt, 2022-02-16:
Hmm, @kisvegabor, in the log it appears that it tries to allocate 308KB, so I'm not sure if that's enough.

kisvegabor, 2022-02-16:
Oh, I missed a digit. 

Note that, GIFs require a  lot of RAM. See here: https://docs.lvgl.io/master/libs/gif.html#memory-requirements

phamvant, 2022-02-16:
I use ``` CUSTOM_MEM_SIZE``` and although with 120*240 gif it only use 6.3% RAM

phamvant, 2022-02-16:
> What it the current size? 
> It should be  at least 40 * 1024.

Even 200 doesn't work

closed 2022-03-11 by lvgl-bot