lvgl/lvgl #3093: Improve the drawing accuracy of arc

FASTSHIFT, 2022-02-15:
### Introduce the problem
The current arc representation angle uses the `uint16_t` type, and the range is 0~360. When the arc size is large, the result of the motion does not seem to be continuous.

### Examples and cases
```c
static void timer_cb(lv_timer_t * timer)
{
  lv_obj_t * arc = timer->user_data;
  static int i = 0;
  lv_arc_set_value(arc, i++);
}

void arc_test(void)
{
  lv_obj_t * arc = lv_arc_create(lv_scr_act());
  lv_obj_set_size(arc, 480, 480);
  lv_arc_set_range(arc, 0, 1000);
  lv_obj_center(arc);

  lv_timer_create(timer_cb, 100, arc);
}
```
https://user-images.githubusercontent.com/26767803/153990584-95a88b98-b547-45ab-b7c2-e9562c83789e.mp4

### Suggested solution
To improve the representation accuracy of arc, can refer to `lv_img_set_angle()` to use 0~3600 as the angle range?

kisvegabor, 2022-02-21:
It looks really odd with large arcs. 

I agree to change the resolution but it would be a larger  breaking change so we can do it only in v9.

BesitzeRuf, 2022-02-22:
Hi all, what about just add a new function for ex. lv_img_set_angle_precise(), which accepts bigger range of angle and internally refactor lv_img_set_angle() (angle parameter shall be multiplied by 10 in this case)?

kisvegabor, 2022-02-22:
@BesitzeRuf It's nice to see you again! :slightly_smiling_face: 

It's a great idea! In v9 we can remove `lv_img_set_angle_precise` and keep only `lv_img_set_angle` using the 0.1° resolution.

embeddedt, 2022-02-22:
> In v9 we can remove lv_img_set_angle_precise and keep only lv_img_set_angle using the 0.1° resolution.

Personally, I think we could still keep `lv_img_set_angle` as a macro in v9, as there's really not much maintenance needed for that and it'd be one less function to refactor on a major version upgrade.

kisvegabor, 2022-02-22:
> Personally, I think we could still keep lv_img_set_angle as a macro in v9, as there's really not much maintenance needed for that and it'd be one less function to refactor on a major version upgrade.

True, but this way it won't follow the same logic as `lv_img_set_angle()`.

GorgonMeducer, 2022-03-15:
I highly recommend using fixed-point (Q15) to represent angle. So you can keep using uint16_t and not necessarily force yourself to choose a precision (either 1° or 0.1°)

kisvegabor, 2022-03-21:
@GorgonMeducer Do you mean 0.0 would mean 0° and 1.0  mean 360°?

GorgonMeducer, 2022-03-21:
> @GorgonMeducer Do you mean 0.0 would mean 0° and 1.0 mean 360°?

No 1.0° means 1 degree

kisvegabor, 2022-03-22:
In LVGL no `float` is used but it's just a minor difference. 

It'd great to have a way to keep it working with normal `int` argument too. 

It sounds ugly but I'm thinking about something like this:
```c
#define LV_FRACT(i, f) (i + (f << 16))

set_angle(10); //10°
set_angle(LV_FRACT(10, 5); //10.5°

void set_angle(uint32_t angle)
{
  uint32_t angle_int;
  uint32_t angle_fract;
  angle_int = angle & 0xffff;
  angle_fract = (angle >> 16) & 0xffff;
}
```

GorgonMeducer, 2022-03-22:
@kisvegabor why not use Q16 directly?

kisvegabor, 2022-03-24:
> why not use Q16 directly?

How do you mean directly?

E.g.
```c
set_angle(10.5);  //Float, not allowed in lvgl
```

GorgonMeducer, 2022-03-25:
How about introduce a macro to solve this problem:

```c
#define f2q16(__value)    (int32_t)((__value) * (float)((int32_t)1<<16))
```

Hence 

```c
set_angle( f2q16(10.5f)); 
````

kisvegabor, 2022-03-29:
I feel like it's too much hack to avoid the obvious solution: `float`. 

Maybe we can approach it differently by introducing an `LV_USE_FLOAT` config option which changes the API of some widgets from `int` to `float`.

FASTSHIFT, 2023-04-08:
@kisvegabor 
Hi gabor, I hope this feature can be done on v9, can we reopen this issue?

kisvegabor, 2023-04-11:
I've reopened it. 

We can implement it for sure. I think the blocking thing now it agree on a fractional parameter support format which is:
- not `float` to keep LVGL integer only
- not too verbose, such as `LV_FRACT(100, 25)`

Actually maybe we can use `float` but make it optional. E.g. 
```c
#if LV_USE_FRACTIONAL_COORDS
typedef lv_coord_fract_t float;
#else
typedef lv_coord_fract_t lv_coord_t;
#endif
``` 

And use `lv_coord_fract_t` in APIs. This way to user can pass `100.25f` or `100` easily. 

The other option could be to have 2 APIs, e.g. `lv_arc_set_start_angle(arc, 100)` and `lv_arc_set_start_angle_fract(arc, 100, 25)`

This shoul affect line drawing too where we should be able to use fractional coordinates. 

What do you think?

W-Mai, 2023-04-12:
Since we want to increase the precision but don't want to introduce float, why not extend the angle to int32_t and use [0, 36000) to represent [0, 360)? This way we can achieve a precision of one hundredth of a degree, and the performance will be more outstanding in any situation. This is much better than the previous 10% precision.😺

What do you think?

C47D, 2023-09-07:
I've seen this issue when working with arcs, is there some work/idea done for this?

kisvegabor, 2023-09-08:
I think there should be a way to disable `float`s to have  smaller memory footprint if needed.

In case of arcs we need to define an angle with higher resolution.  We need an angle for image rotation too. Are the any  units other then degrees where we need higher resolution?

If not, what if add this to `lv_conf.h`:
```c
#define LV_ANGLE_FLOAT   1
```

Somewhere:
```c
#if LV_ANGLE_FLOAT
    typedef float lv_angle_t;
#else
    typedef int32_t lv_angle_t;
#endif
``` 

And in `lv_arc`:
```c
lv_arc_set_bg_start_angle(lv_obj_t * arc, lv_angle_t angle);
```

This way `lv_arc_set_bg_start_angle(arc, 40)` always works, but if you have enabled `LV_ANGLE_FLOAT`, the parameter can be e.g.  `40.3f` too.

Internally in drawing routines and a few other places we can add `#if LV_ANGLE_FLOAT` to handle the angles differently if needed. 

What do you think?

kisvegabor, 2023-10-26:
Done by `LV_USE_FLOAT` and `lv_value_precise_t`

closed 2023-10-26 by kisvegabor