lvgl/lvgl #3094: Slider example 3 (Slider with extended drawer) doesn't compile, after fix the label is not printed

SgtPepperFTW, 2022-02-15:
`lv_coord_t* size = lv_event_get_param(e);` should be
`lv_coord_t* size = (lv_coord_t*)lv_event_get_param(e);`

`lv_obj_draw_part_dsc_t* dsc = lv_event_get_param(e);` should be
`lv_obj_draw_part_dsc_t* dsc = (lv_obj_draw_part_dsc_t*)lv_event_get_param(e);`

` lv_draw_label(dsc->draw_area, &label_draw_dsc, &label_area, buf, NULL);` should be
` lv_draw_label(dsc->draw_area, &label_area, &label_draw_dsc, buf, NULL);`

I would have made a PR, but I still can't see the label, can anyone check if the example works correctly with these fixes?

SgtPepperFTW, 2022-02-15:
Ok ,so the label is printed with

`lv_draw_label(&label_area, &label_area, &label_draw_dsc, buf, NULL);`

only I don't know if it is intended to work like that AND the label is printed black even if the dark theme is standard.

kisvegabor, 2022-02-16:
Thanks, I've pushed a fix for both the warnings and the not visible label on dark theme.

commit 4a766c51 (2022-02-16): example(slider): make lv_example_slider_3 work with dark theme too

fixes: #3094