lvgl/lvgl #3097: Support for start and repeat delay for circular scrolling labels

andresovela, 2022-02-16:
### Introduce the problem

LVGL should support these kinds of animations out of the box:
[image]

The gif above shows a label that is displayed at the starting position for X seconds (animation start delay), then it does the circular scrolling animation and when it comes back to the starting position it stays there for Y seconds (repeat delay) before starting to scroll again.

See LVGL forum post [here](https://forum.lvgl.io/t/how-to-customize-the-circular-scrolling-of-a-label/8011/14).

### Examples and cases
See the gif above.

### Suggested solution
I think there needs to be a way to set the `start_delay` and `repeat_delay` properties of the scrolling animation.

kisvegabor, 2022-02-16:
As discussed in the Forum I think a new style property called `anim` would be the most flexible solution. The animations of the widgets can consider this animation as a template. 

To add a new style property it should be listed in [style_api_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/style_api_gen.py) and the script needs to be run to generate API for the new property. 

In `lv_label` we can use `lv_obj_get_style_anim(label, LV_PART_MAIN)` to get the template, copy it and overwrite the required fields. `lv_obj_get_style_anim_speed()` also can be used to set the duration.

andresovela, 2022-02-17:
I implemented this in a local branch and it seems to be working fine. I had a couple of issues though:

The start delay gets overwritten by this piece of code:
https://github.com/lvgl/lvgl/blob/fe4caaff0e4b65cfa37d34ff5077960b6ed47ac7/src/widgets/lv_label.c#L1056-L1061

I'm not exactly sure what that code is doing, but commenting it out fixes the start delay issue. I'm not sure what exactly is it that I'm breaking by commenting it out though. How do I go about that?

The other thing I ran into is that after running the `style_api_gen.py` script, it didn't add a new entry to the `lv_style_prop_t` enum:
https://github.com/lvgl/lvgl/blob/fe4caaff0e4b65cfa37d34ff5077960b6ed47ac7/src/misc/lv_style.h#L263-L266

I added the entry manually but I feel like it should have been taken care of automatically.

Lastly, I didn't know what `style_type` to choose when adding the new property to the `style_gen_api.py` script. I chose `'style_type' : 'ptr'`, and using adding the property to the style requires the user to pass a pointer to the `lv_anim_t` structure and keep that struct around for as long as the style is in use. Not sure if that's the way to go either.

I can make a PR after those points are clear @kisvegabor

kisvegabor, 2022-02-17:
> I'm not exactly sure what that code is doing, but commenting it out fixes the start delay issue. I'm not sure what exactly is it that I'm breaking by commenting it out though. How do I go about that?

This code ensures that if you update the label's text the scroll position is maintained. Without it when the the animation starts from the beginning if you update the text.   Maybe we can skip it if `delay != 0` in the template animation.

> The other thing I ran into is that after running the style_api_gen.py script, it didn't add a new entry to the lv_style_prop_t enum:

I forgot the mention that. It's really maintained manually because the options has a specific order and indices to lvgl apply some optimizations when searching properties.

This order seems fine:
```c
 LV_STYLE_ANIM_SPEED              = 102,
 LV_STYLE_ANIM                    = 103, 
 LV_STYLE_TRANSITION              = 104, 
```

> Lastly, I didn't know what style_type to choose when adding the new property to the style_gen_api.py script. I chose 'style_type' : 'ptr', and adding the property to the style requires the user to pass a pointer to the lv_anim_t structure and keep that struct around for as long as the style is in use. Not sure if that's the way to go either.

I should do the same way :slightly_smiling_face:

PR #3128 (2022-02-25): feat(label): added animation style property to apply it to circular scrolling animation of label widget
### Description of the feature or fix

Added an animation style property to be used as animation template for different use cases in different widgets.
This commit implements using this property to set the start and repeat delay of the circular scrolling animation of the label widget.

Closes #3097.

I think the new `ANIM` property should replace `ANIM_TIME` wherever that is used, but I think that's beyond the scope of this PR.

### Checkpoints
- [x] Follow the [styling guide](https://github [... cut]
commit 340d45cf (2022-02-25): feat(label): added animation style property to apply it to circular scrolling animation of label widget (#3128)

* feat(label): added animation style property to apply it to circular scrolling animation of label widget

Added an animation style property to be used as animation template for different use cases in different widgets.
This commit implements using this property to set the start and repeat delay of the circular scrolling animation of the label widget.

Closes #3097

* fix(label): changed animation style property's var_type to `const lv_anim_t *`

* example(label): added example show [... cut]
closed 2022-02-25 by kisvegabor