lvgl/lvgl #3098: LV_OBJ_FLAG_OVERFLOW_VISIBLE doesn't draw overflow in y-directions correctly

Viatorus, 2022-02-16:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version -> 8.2.0 and master
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Using LV_OBJ_FLAG_OVERFLOW_VISIBLE  seems to have a bug with correctly drawing overflow in y-direction.
Even without scrollbar, the problem exists.

Tested on Windows 10, SDL2 and MicroPython online.

See here: 

### To Reproduce

```c
lv_obj_t* obj = lv_obj_create(lv_scr_act());
lv_obj_set_size(obj, lv_pct(100), lv_pct(100));

lv_obj_t* obj2 = lv_obj_create(obj);
lv_obj_set_size(obj2, 240, 220);
lv_obj_set_pos(obj2, 0, 100);
lv_obj_add_flag(obj2, LV_OBJ_FLAG_OVERFLOW_VISIBLE);

lv_obj_t* obj3 = lv_obj_create(obj2);
lv_obj_set_size(obj3, 240, 100);
lv_obj_set_pos(obj3, 0, -50);
```

[image]

--

```python
import display_driver
import lvgl as lv

scr1 = lv.obj()
scr2 = lv.obj(scr1);
scr3 = lv.obj(scr2);
scr3.set_pos(2, 2);

scr2.add_flag(lv.obj.FLAG.OVERFLOW_VISIBLE)

lv.scr_load(scr)
```

https://sim.lvgl.io/v8.1/micropython/ports/javascript/index.html?script_direct=4945be4fb597588372101912885c52730a3fa4bd

[image]

### Expected behavior

Correctly drawn of OVERFLOW_VISIBLE in x and y direction.

kisvegabor, 2022-02-16:
Hi,

Thanks for the report. I've just pushed a fix.

Viatorus, 2022-02-16:
Thank you for your fast action but I think you break something:
[image]

kisvegabor, 2022-02-16:
Ups, a debug settings left enabled. Fixed here: fe4caaff

Viatorus, 2022-02-16:
Great! Know it is working. Thank you very much. :)

commit c742f2c8 (2022-02-16): fix(obj): fix LV_OBJ_FLAG_OVERFLOW_VISIBLE

fixes: #3098