lvgl/lvgl #3100: lvgl\src\draw\lv_draw_rect.h(44,6): warning C4201: nonstandard extension used: nameless struct/union

geert-KLA-BE, 2022-02-17:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
When building custom font files, generated through lv_font_conv, I get a warning in visual studio for each font file:
       lvgl\src\draw\lv_draw_rect.h(44,6): warning C4201: nonstandard extension used: nameless struct/union

`
    /*Background*/
    lv_opa_t bg_opa;
#if __STDC_VERSION__ >= 201112L
    union {
#endif
        lv_color_t bg_color;        /**< First element of a gradient is a color, so it maps well here*/
        lv_grad_dsc_t bg_grad;
#if __STDC_VERSION__ >= 201112L
    };
#endif
`
This should be simple to solve by assigning a name to the union here.

### To Reproduce
Just build lvgl with VS2019 using standard C17.

### Expected behavior

No warnings

kisvegabor, 2022-02-17:
Hi,

Should be fixed here: https://github.com/lvgl/lvgl/pull/3099

closed 2022-02-17 by kisvegabor

geert-KLA-BE, 2022-02-17:
Ok thanks I didn't find that through the search.