lvgl/lvgl #3111: A new porting for Raspberry Pi Pico in MDK

GorgonMeducer, 2022-02-19:
### Introduce the problem
I have created a new porting for the Raspberry Pi Pico in MDK.

https://github.com/GorgonMeducer/lv_port_mdk_raspberry_pi_pico

It uses a popular 1.3inc LCD module

[image] 

https://www.waveshare.com/wiki/Pico-LCD-1.3

### Examples and cases

### Suggested solution
Could you please create a repo as `lv_port_mdk_raspberry_pi_pico`?

Thank you.

kisvegabor, 2022-02-21:
Hi,

The link to you repo is not working. Did you mean https://github.com/GorgonMeducer/Pico_Template ?

As a name I'd swap it as `lv_port_raspberry_pi_pico_mdk`. Is there a way to build project with `make` or `CMake` too?

GorgonMeducer, 2022-02-21:
@kisvegabor 
I have verified the link, it is 
https://github.com/GorgonMeducer/lv_port_mdk_raspberry_pi_pico

I can open it without problem. It is not https://github.com/GorgonMeducer/Pico_Template which is purely a Pico template without LVGL.

It is ok to swap it as `lv_port_raspberry_pi_pico_mdk`

MartinK7, 2022-02-21:
Check if your repository is marked as as "public" instead of "private".

GorgonMeducer, 2022-02-21:
> Check if your repository is marked as as "public" instead of "private".

Thank you, you are right, it was private. I created this repo using Github Desktop, I didn't notice that this tool set the repo as private by default.

GorgonMeducer, 2022-02-21:
@kisvegabor by the way, I have to emphasize that:
Although this project uses the official pico-sdk (which is based on gcc), it uses a different compiler, i.e. arm compiler 6. 

Furthermore, the LVGL is deployed using cmsis-pack, I don't think a cmake or make project can benefit from LVGL cmsis-pack easily without the help of MDK. If you have MDK, then why don't you use MDK directly. 

In the future, maybe someone could contribute a cmake or makefile project. Or probably, someone could create a separated repo and use cmake+gcc+pico-sdk as the toolchain.

GorgonMeducer, 2022-03-08:
@kisvegabor are there any suggestions?

kisvegabor, 2022-03-08:
Sorry, I missed your previous answer. 

I see you point and agree to keep it MDK specific. So, shall I open a `lv_port_raspberry_pi_pico_mdk` repo, right? :slightly_smiling_face:

GorgonMeducer, 2022-03-08:
YES. Thank you. @kisvegabor

kisvegabor, 2022-03-08:
Opened, and invitation sent:
https://github.com/lvgl/lv_port_raspberry_pi_pico_mdk

GorgonMeducer, 2022-03-08:
@kisvegabor Thank you very much. I will upload the repo today.

closed 2022-03-08 by GorgonMeducer

GorgonMeducer, 2022-03-08:
Done