lvgl/lvgl #3116: Suggestion: Dynamic Global LVGL Cache

amirgon, 2022-02-21:
Today LVGL maintains a number of caches for different purposes:
- Shadows
- Circle drawing
- Gradient
- Images
- lv_fs_read

Today these caches are allocated statically (on the data segment) and their size is configurable through macros.  
These statically-allocated caches have number of disadvantages:
- The ram is always "locked out" for other purposes. Some applications might have some modes where more ram is temporariliy needed, at the exspense of gui optimizations.
- On some platforms (such as ESP32), heap can be allocated on external RAM (spi-ram) while the main ram is used for DMA etc. Allocating cache buffers on external ram could save precious ram and still be effective since the cpu-cache backs the external ram as well, so on periods of time when the cache buffers are used extensively they can still be very fast even though they are on external ram.
- In case of Micopython or other dynamic bindings, these caches are preconfigured today per firmware instead of being user-configurable at runtime.

Instead of statically allocated caches, it could be useful to allow the user allocate and configure these caches dynamically on runtime in order to provide the user more control over when they are allocated/deallocated and in which size.

Also, to make things simpler to the user, instead of configuring each cache individually, the user could dynamically define a "global" cache according to the available memory on the device and let LVGL manage all caches internally inside this global cache.  
This way, the user could simply say "use more cache now" or "release all cache now" without worrying which cache he should be twicking.  
Advanced users might want to control how this global cache is split between specific LVGL caches (this can be specified, for example, by using a factor for each individual cache type which would control how much it consumes out of the global cache)

kisvegabor, 2022-02-24:
I like the idea of using run time configurable cache. Something like:
```c
static uint8_t cache[1024];
set_cache(cache, 1024);
```

will have the behavior as now. 

@xiaoxiang781216 proposed to have hook on malloc "Out of memory" that asks the the user to free some memory and try malloc-ing automatically again. The user could simply reduce the cache size in such a hook.

However, if there is an Out of memory error in the middle of e.g. image decoding and we change the cache size (i.e. drop the cache and start with a smaller empty one) some previously allocated data of the current decoding will be lost too. So there should be a way  to restart it in such event somehow.

About the global cache, it an interesting idea. I still can't how can it be managed in predictable way but let's see what can we figure out. Just as a topic starter: having something like cache priority and set it like:
```c
lv_cache_set_priority(LV_CACHE_TYPE_GRADIENT, LV_CACHE_PRIORITY_HIGHEST);
lv_cache_set_priority(LV_CACHE_TYPE_FONT, LV_CACHE_PRIORITY_LOWEST);
lv_cache_set_priority(LV_CACHE_TYPE_IMAGE, LV_CACHE_PRIORITY_NONE);
```  

cc @X-Ryl669

X-Ryl669, 2022-02-24:
I thought that's more or less what the memory allocator in LVGL already does? 

I completely agree with @amirgon remark above. There are multiple memory area with different behavior on MCU, and it might be interesting to take advantages of those. To follow his example, if rendering a Lottie animation to internal ESP32's 320kB of DRAM instead of the PSRAM, you'll gain a factor 10x in speed (almost). Right now, there is only a malloc/new in the code so it's asking global ESP32 heap allocator and it's very likely it'll hit the PSRAM first (since the allocation requested size is large and 8 bits strided. The ESP32 does not give IRAM memory that's only accessible in 32bits words, even if it has plenty of it).

If there was a low level API to help make good decision in LVGL (with fallbacks, as you said) for memory allocation and releasing, then it might help a lot in performance.

Not speaking of dealing with caches, but mainly being able to "advice" the allocator what kind of memory we need, then the allocator could make good decisions by itself. 

So, in addition to cache API above, it would also need some modification in the code that's using the cache to add error reporting (like "E_MEM_RECLAIMED") so the operation:
 1. Can be retried when the cache is dropped (for example, the decoder will try to allocated some buffer for the image, and the allocator will figure out there's not enough memory, it'll call the hook to release the caches, and return E_MEM_RECLAIMED. The decoder will check the return code and re-query the (now empty) cache to find a free entry)
 2. There's some multithreading issue going on here, if some thread is using a pointer on the cache. Since LVGL is not thread safe, I don't know if it's a concern, but it might be a good idea to provide a function pointer that's called when the cache is evicted/dropped, so that the user can free heap space on her own, and/or signal the thread/process that the cache is invalid.
 

I don't like the cache "priority" here, since, it's a lot of complexity and without much interest. 
If your decoder tries to allocate from a low priority cache and there no such place anymore, should it trigger dropping the high priority cache after having dropped the low priority cache ? 
Or, on the opposite, if the decoder need high priority cache and it's full but there's enough space in a middle priority cache, why should we drop low priority cache that's still valid ?

Honestly, it would be better if there was only one cache (shared between all *consumers*) and that the allocation was *typed* with the feature (high speed RAM, DMA RAM, low speed RAM, GPU RAM, whatever) and *maybe* a priority field. When full, the linked list would be search for the lowest priority and low priority item is freed. The iteration continue until there's enough space in the allocator.

 In the end, it's almost what LVGL does now with it's own allocator (except that it does not depend on low level allocator... yet).

If pushing the algorithm to its limit, each cache's item should have a callback function pointer that's being called when *forced* free (so we can call decoder_close, gradient_cache_clean, etc...) a bit like a dumb garbage collecting in Java with its dispose() callback.

kisvegabor, 2022-02-24:
So there are 2 questions:
### How to decide which entry to free when the cache is full

In the image decoder's cache we measure the "time-to-open" and try to make the harder to decode images leave longer.  

Maybe we can use the same policy. So that data should be cached whose reopening takes the most time. 
However, let's say we cache character bitmaps too (FreeType has its own cache, so let's talk about a custom imaginary font decoder). If it takes 0.1 ms for each bitmap probably it never will be cached. But is the a letter is used 200 times caching would save 20 ms in the end. 

So the life of the cache entries should be decreased with the estimated (or real) time-to-open. If we open a png once in 10 ms and then draw 100 letters in 100x0.1ms, the PNG life will be zero and it can be replaced by character bitmaps. 

But it also can be misleading. If the all the 100 letters and the upcoming 100 letters were different, the PNG's life is still over but still no cache entry would be used twice.

Maybe storing a hash from the entries to see how many times they were opened? 

### How to handle the "malloc hook"

In the end we need to check after any `malloc` (direct or hidden in an other function) and return if the cache was reset. It's very error prone.

X-Ryl669, 2022-02-25:
> How to handle the "malloc hook" ?

As I understand it, it would be better if there was no impact on the calling code. This is possible if inside the lv_malloc function, a strategy/algorithm was used, like this:
1. Try to allocate from the heap (if ok, return the pointer)
2. If heap is full, start parsing the cache(s) items and evict the less useful items until the required alloc size is available. Defining the usefulness is a complete algorithm/problem, as you stated above
3. If after evicted the complete cache items, there's still not enough memory, call the user's `need_memory_cb`
4. If even after the call there's still no memory, it's game over, return NULL.

That way, there's no need to change a single line of the calling code. 

However, as you said, defining what we can clean first in the cache is an art, and I can't provide a rule here. Maybe it should be priority based (although I don't vote for this, too hard to get right), maybe it should be "usage counter" based, maybe it should be "difficulty counter" based, I don't know. 

Also, there's the issue of moving cache entries in memory. For gradient, it's completely ok to memmove the entries since there's no outside pointer, we can keep the cache's heap unfragmented (the current implementation does this). For fonts, I'm not sure it's possible, but probably too. 
But for images, I don't think it's possible. This means that we'll rely on the allocator to figure out how to merge freed block, typically a case where 2+2 != 4, that is, it's not because we free 2x 1024 bytes buffer that we can allocate a 2048 bytes buffer.
Again, it's not impossible to implement, it just means we have to be very precise when writing the code (maybe by checking the allocator's internal state, or something like this).

And finally, there's the issue of destructors to be called (for example: `decoder_close`). I think it's easily solved by a function pointer on the cache entry's structure.

kisvegabor, 2022-02-25:
Actually, what and how do we want to store in the cache? 

E.g. for an image decoder (just an example):
- uncompressed image: 100 kB
- something about the decoding session: 64 bytes
- some other temporary data: 20 bytes

For a font bitmap:
- the bitmap: 160 byte
- pointer to the font: 4 bytes
- unicode value: 4 bytes
- some other temporary data: 12 bytes 

I think all these needs to be cached in one chunk.
```c
typedef struct {
   uint32_t a;
   uint32_t b;
   uint32_t c;
   uint32_t d;
   lv_color_t * raw_img;
}my_img_decoder_t;

my_img_decoder_t * dec = cache_alloc(sizeof(my_img_decoder_t) + width * height * color_depth)
dec->raw_img = ((uint8_t * )dec) + sizeof(my_img_decoder_t);
``` 

Same for the bitmap.

So if we establish a policy to malloc from the cache only in one chunk there won't be half allocated cases.

In other words it should be avoided:
```c
void decode(void) 
{
  a = malloc(10);
  b= cache_malloc(20);
  c = malloc(20);
  d = cache_malloc(200*200);
```

Instead:
```c
x = cache_malloc(10 +20 + 20 + 200 * 200);
```

X-Ryl669, 2022-02-28:
Right, exactly. Having a single allocation as the *enormous* advantage of being either successful or failed. No half state where `c` fails but both `a` and `b` succeed which implies a lot of code to deal with freeing previous allocation before returning a error.

I did this in the gradient cache and, I think, it works well because the usual memory size of the gradient item is small. 
When evicting an item in the cache, I'm memcpy'ing the next gradient's items data and since it's small, that's acceptable.

But for the picture case... If you have a 1080p picture, that's 6MB of data. That can't be memcpy'd around without a performance hit. So, instead, these should be left in the TLSF allocator (up to the allocator to merge the block when released). That's a very specific case, but we have many such case (for example, when evicting a image cache entry, we have to free the decoder context and it might contain a file descriptor that needs to be closed to avoid leaking in kernel memory). 

So for these items, it might make sense to have a pointer to a destructor function that's in charge of actually freeing the heap (not cache) memory and closing the file descriptors, network streams, whatever.

This is not that hard to do, as this limits the error code testing to specific low level functions (in the image decoders' code and the image cache), so hopefully we can make it error-free.

X-Ryl669, 2022-02-28:
Yes that's a good idea, and it shouldn't be hard to enforce.
Should it be shared with LVGL's allocator buffer ?

I'm not sure about this. Should the cache be a different memory from the LVGL's own heap buffer, or should it use it too ?
If the former, then there's the issue of image's cache picture buffer that should be in LVGL heap, IMHO (we won't perform better than TLSF). Also the code that report the amount of memory used by LVGL should be modified.

If the latter, then yes it makes sense to split the cache's type and have some limit on the "global out of memory killer" so all other user can work too without having to kill the cache for every malloc.

What do you think ?

X-Ryl669, 2022-03-02:
It don't think it's a good idea to have multiple TLSF instance, see this [bug](https://github.com/espressif/esp-idf/issues/7822) for reference. The overhead of the TLSF control structures is likely of the same order of the cache's size.

kisvegabor, 2022-03-02:
After reading the bug you linked I made some tests in LVGL. 
`SL_INDEX_COUNT_LOG2 = 5` right now in LVGL.
This is with PC using `lv_demo_widgets()`:

`LV_MEM_SIZE (4 * 1024 * 1024U)`
| SL_INDEX_COUNT_LOG2 | sizeof(control_t) | mem used |
|--------------------------------------|------------------------|---------------|
| 5 | 3936 | 61.2kB |
| 4 | 2152 | 59.5kB |
| 3 | 1192 | 58.6kB |
| 2 | 688  |  58.1kB |
| 1 | 578  |  57.8kB |
    
    

`LV_MEM_SIZE (80 * 1024U)`
| SL_INDEX_COUNT_LOG2 | sizeof(control_t) | mem used |
|--------------------------------------|------------------------|---------------|
| 5 | 2640 | 60.0kB |
| 4 | 1488 | 58.8kB |
| 3 | 856   |  58.2kB |
| 2 | 504   |  57.9kB | 
| 1 | 320 |  57.7kB |

I don't know what could be the drawback of using a smaller `SL_INDEX_COUNT_LOG2`. @xiaoxiang781216 do you have any ideas?

X-Ryl669, 2022-03-03:
In the patch linked in the bug above, the idea is to merge the instances and adapt the size of the control structure to the size of the heap (a small heap is less likely to have plenty of allocations, so it does not make sense to have larger controls). On ESP32, there are many heaps, and cumulative save is something like 20kB with the patch. That's not negligible.

kisvegabor, 2022-03-03:
I haven't reviewed TLSF in such detail but do you know how the max entry count depends on  `SL_INDEX_COUNT_LOG2`?

X-Ryl669, 2022-03-03:
The SL_INDEX_COUNT_LOGX defines the number of subdivision in the block (there are FL_INDEX_COUNT initial blocks). Once all the divisions have happened, you can't get smaller block for your allocation, so you start wasting memory like crazy.
For example, if you set a LOG2 define, you can only divide a block 4 times (IIUC), so if your initial block is 1MB size, you'll only be able to allocate 4x 256kB (best case), or 1x256kB + 1x128kB + 1x64kB + 1x32kB items (only 480kB of memory total). I might be wrong here, since I haven't looked at the algorithm underneath, but that's what's described in the documentation.

kisvegabor, 2022-03-04:
I made a test with 1M pool size with different  `SL_INDEX_COUNT_LOG2` settings and alloc sizes.

The algorithm:
```c
    size_t alloc_size = 1 or 8 or 1024;
    void * p;
    uint32_t c = 0;
    do {
        p = lv_mem_alloc(alloc_size); 
        c++;
    }while(p);

    lv_mem_monitor_t m;
    lv_mem_monitor(&m);

    printf("count: %d (from tlsf: %d), free: %d\n", c, m.used_cnt, m.free_size);
    return 0;
```

The results:
**`alloc_size=1`**
-  `SL_INDEX_COUNT_LOG=5  count: 32415 (from tlsf: 32496), free: 0`
- `SL_INDEX_COUNT_LOG=1  count: 32510 (from tlsf: 32591), free: 0`

**`alloc_size=8`**
-  `SL_INDEX_COUNT_LOG=5  count: 32415 (from tlsf: 32496), free: 0`
- `SL_INDEX_COUNT_LOG=1  count: 32510 (from tlsf: 32591), free: 0`

**`alloc_size=1024`**
-  `SL_INDEX_COUNT_LOG=5  count: 1006 (from tlsf: 1087), free: 80`
- `SL_INDEX_COUNT_LOG=1  count: 1009 (from tlsf: 1090), free: 24`

So with `SL_INDEX_COUNT_LOG=1` always more blocks were allocated, I guess because of the smaller overhead.  

I opened a [PR](https://github.com/mattconte/tlsf/issues/25) in TLSF's repo with these result.

X-Ryl669, 2022-03-04:
Am I missing something ? You've made a 1MB pool and you've only allocated 32K block of 1 byte ? Or 32K block of 8 bytes (so 256kB) ? Something is clearly wrong here.

kisvegabor, 2022-03-07:
> (and it wouldn't make sense at all, since it would mean you only have 2 blocks to allocate)

With this logic LOG=5 allowed only 32 blocks which certainly not the case.  Anyway, it's independent from this  issue, but I suggest lowering it to 3 (1 feel too small :smile: ) and see if someone reports any issues.

X-Ryl669, 2022-03-07:
Not sure I get you here. In your comment above you were only able to allocate 32k blocks of 1 bytes. It's exactly 1MB / 32. So clearly, you have 2⁵ levels of allocations (said differently, the 1MB pool was divided by 32 to create 32k blocks of 32kB minus the TLSF overhead). In each block of 32kB you only "used" 1 byte, but the block was marked as used and can't be used for anything else, hence the result of **only** 32415 allocations. 

In that trivial case, the allocator is very inefficient, since on the 1MB of initial pool, you only have 32kB of actual data (efficiency of 3.12%).

For the 8 bytes case, it's similar you'll get only 25% of the initial pool.

When going smaller (LOG=3) you are reducing the efficiency even more for small allocation. Please notice that LOG=1 is likely not supported by the library and it's silently upped to a higher value (at least, on ESP32, 5 is the minimum, and the patch I've linked reduced it to 3, to be used for very very small heaps).

kisvegabor, 2022-03-17:
We have some related open topic, so let's organize them:
- Image decoder rework (@X-Ryl669 #3161)
- Limit the image cache size in bytes instead of image count  (@FASTSHIFT #3124)
- Global cache implementation (this issue)

I think we should finish the image decoder things first and work on the global cache after that. Finally we can use the global cache with images too.

What do yo think?

closed 2022-04-09 by lvgl-bot