lvgl/lvgl #3117: Examples being built when running tests

C47D, 2022-02-22:
### Introduce the problem
The examples are being built when running tests, is this necessary?

### Examples and cases
Run the test cases and see the examples being built.

### Suggested solution
Don't build examples when unit tests are being build.

kisvegabor, 2022-02-22:
There are test cases for the [stress](https://github.com/lvgl/lvgl/blob/master/tests/src/test_cases/test_demo_stress.c) and [widgets](https://github.com/lvgl/lvgl/blob/master/tests/src/test_cases/test_demo_widgets.c) demos so the are required, but I agree that the other demos and examples can be excluded from tests. However, they are required for the "build" tests as we want to see is there area any warnings.

C47D, 2022-02-23:
Hi @kisvegabor, so the required changes would be?

- [ ] Keep demos in unit tests because of the stress and widgets tests
- [ ] Remove examples from unit tests
- [ ] Keep examples in build tests

kisvegabor, 2022-02-23:
> Keep demos in unit tests because of the stress and widgets tests

For demos I suggest:
- [ ] Keep demos in build tests
- [ ] Keep only  stress and widgets tests in unit tests

kisvegabor, 2022-03-10:
Let's continue in the PR.

closed 2022-03-10 by kisvegabor