lvgl/lvgl #3120: A table issue...

ay123net, 2022-02-22:
I create a table with 8 cols 8 rows， in some cases,  I change the table to 4 cols and 4 rows , but it hangup. 

The LVGL version is the lastest release.
`
    lv_obj_t * table;
    uint32_t i, j;
    
    table = lv_table_create(lv_scr_act());

    lv_obj_center(table);

    lv_table_set_col_cnt(table, 8);
    lv_table_set_row_cnt(table, 8);

    for (i = 0; i < 8; i ++) {
        for (j = 0; j < 8; j ++) {
            lv_table_set_cell_value(table, i, j, "00");
        }
    }

    lv_table_set_col_cnt(table, 4);
    lv_table_set_row_cnt(table, 4);

    for (i = 0; i < 4; i ++) {
        for (j = 0; j < 4; j ++) {
            lv_table_set_cell_value(table, i, j, "00");
        }
    }
`

C47D, 2022-02-22:
Is there any interesting output when enabling the logs?

C47D, 2022-02-23:
@ay123net can you set the row count before setting the column count and report back if you still have the issue? Something like this:

```c
lv_obj_t * table;
uint32_t i, j;

table = lv_table_create(lv_scr_act());

lv_obj_center(table);

lv_table_set_row_cnt(table, 8);
lv_table_set_col_cnt(table, 8);

for (i = 0; i < 8; i ++) {
    for (j = 0; j < 8; j ++) {
        lv_table_set_cell_value(table, i, j, "00");
    }
}

lv_table_set_row_cnt(table, 4);
lv_table_set_col_cnt(table, 4);

for (i = 0; i < 4; i ++) {
    for (j = 0; j < 4; j ++) {
        lv_table_set_cell_value(table, i, j, "00");
    }
}
```

ay123net, 2022-02-23:
> @ay123net can you set the row count before setting the column count and report back if you still have the issue? Something like this:
>
> ```c
> lv_obj_t * table;
> uint32_t i, j;
>
> table = [...]

closed 2022-02-23 by ay123net

ay123net, 2022-02-23:
@C47D Hi, It test pass in LVGL version 8.0.3, but the lastest version 8.2.0 is fail

PR #3121 (2022-02-24): test(table): Replicate issue when reducing table cells
### Description of the feature or fix

See #3120 for report
Closes #3120 

Output when running the tests:
```console
==============================
Running tests for test_sysheap
==============================
Test project /home/d3bug/lvgl/tests/build_test_sysheap
      Start  1: test_arc
      Start  2: test_bar
      Start  3: test_checkbox
      Start  4: test_config
      Start  5: test_demo_stress
      Start  6: test_demo_widgets
      Start  7: test_dropdown
      Start  8: test_event
      Start  9: test_font_loader
      Start 10: [... cut]