lvgl/lvgl #3122: slider does not respond to clicks outside the clipping area

FASTSHIFT, 2022-02-23:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
The slider can only respond to drags within the clipping area, not clicks outside the clipping area.

### To Reproduce

```c
static void ui_test()
{
    lv_obj_t * par = lv_obj_create(lv_scr_act());
    lv_obj_remove_style_all(par);
    lv_obj_set_size(par, 200, 10);
    lv_obj_center(par);
    lv_obj_add_flag(par, LV_OBJ_FLAG_OVERFLOW_VISIBLE);
    lv_obj_set_style_border_color(par, lv_palette_main(LV_PALETTE_RED), 0);
    lv_obj_set_style_border_width(par, 1, 0);

    lv_obj_t * slider = lv_slider_create(par);
    lv_obj_set_size(slider, 200, 10);
    lv_obj_set_style_pad_all(slider, 30, LV_PART_KNOB);
}
```

### Expected behavior

The drag of the slider is not affected by the clipping area.

### Screenshots or video

https://user-images.githubusercontent.com/26767803/155279426-328ca9df-a926-486c-8cea-491de357f8a9.mp4

kisvegabor, 2022-02-23:
You can manually increase the click area with `lv_obj_set_ext_click_area(slider,  20)`

FASTSHIFT, 2022-02-24:
@kisvegabor 
Problem solved, thank you :+1:

closed 2022-02-24 by FASTSHIFT