lvgl/lvgl #3123: BUG: Menu can not be deleted.

bsl-jk, 2022-02-24:
### Perform all steps below and tick them with [x]
- [x ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x ] Update lvgl to the latest version
- [ x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Menus cannot be deleted. The page destructor has an incorrect mem free call: 
https://github.com/lvgl/lvgl/blob/ac202e7b96510b9b12beb8a1eee3dfd65bc56a3d/src/extra/widgets/menu/lv_menu.c#L553

### To Reproduce

```
static void back_event_handler(lv_event_t * e)
{
    lv_obj_t * obj = lv_event_get_target(e);
    lv_obj_t * menu = lv_event_get_user_data(e);

    if(lv_menu_back_btn_is_root(menu, obj)) {
        lv_obj_clean(lv_scr_act());
    }
}

void lv_example_menu_2(void)
{
    lv_obj_t * menu = lv_menu_create(lv_scr_act());
    lv_menu_set_mode_root_back_btn(menu, LV_MENU_ROOT_BACK_BTN_ENABLED);
    lv_obj_add_event_cb(menu, back_event_handler, LV_EVENT_CLICKED, menu);
    lv_obj_set_size(menu, lv_disp_get_hor_res(NULL), lv_disp_get_ver_res(NULL));
    lv_obj_center(menu);

    lv_obj_t * cont;
    lv_obj_t * label;

    /*Create a sub page*/
    lv_obj_t * sub_page = lv_menu_page_create(menu, NULL);

    cont = lv_menu_cont_create(sub_page);
    label = lv_label_create(cont);
    lv_label_set_text(label, "Hello, I am hiding here");

    /*Create a main page*/
    lv_obj_t * main_page = lv_menu_page_create(menu, NULL);

    cont = lv_menu_cont_create(main_page);
    label = lv_label_create(cont);
    lv_label_set_text(label, "Item 1");

    cont = lv_menu_cont_create(main_page);
    label = lv_label_create(cont);
    lv_label_set_text(label, "Item 2");

    cont = lv_menu_cont_create(main_page);
    label = lv_label_create(cont);
    lv_label_set_text(label, "Item 3 (Click me!)");
    lv_menu_set_load_page_event(menu, cont, sub_page);

    lv_menu_set_page(menu, main_page);
} ```

### Expected behavior
<!--
A clear and concise description of what you expected to happen.
-->

### Screenshots or video
<!--
If applicable, add screenshots to help explain your problem.
-->

kisvegabor, 2022-02-24:
@ckielstra Could you take a look at it?

ckielstra, 2022-03-01:
@kisvegabor The problem is that we try to free a title object that wasn't allocated using new(). Suggestion is to remove the lv_mem_free() call in line 553, making it responsibility of the calling function to free the memory when necessary. 
A more explicit solution would be to make it a smart pointer([std::unique_ptr](https://en.cppreference.com/w/cpp/memory/unique_ptr)), but I see we don't use this mechanism in the rest of the code.

kisvegabor, 2022-03-07:
Thanks!

I think we should rather internally allocate the `title` in line 127:
```c
    lv_menu_page_t * page = (lv_menu_page_t *)obj;
    page->title = lv_mem_alloc(strlen(title) + 1);
    LV_ASSERT_MALLOC(page->title);
    if(page->title == NULL) return;
    page->title = title;
```

kisvegabor, 2022-03-08:
> Yes. Looks good, except last line should use strcpy() instead of an assignment.

True.

I've opened #3154

PR #3154 (2022-03-08): fix(menu): fix crash on delete
fixes #3123

### Checkpoints
- [x] Follow the [styling guide](https://github.com/lvgl/lvgl/blob/master/docs/CODING_STYLE.md)
- [x] Run `code-format.py` from the `scripts` folder. [astyle](http://astyle.sourceforge.net/install.html) needs to be installed.
- [x] Update the documentation
commit a6c4c134 (2022-03-08): fix(menu): fix crash on delete (#3154)

* fix(menu): fix crash on delete

fixes #3123

* handle NULL

* code formatting
closed 2022-03-08 by kisvegabor