lvgl/lvgl #3126: SDL backend crash

kisvegabor, 2022-02-24:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I'm trying out the SDL backend in a project but it crashes even with the most simple UIs. I believe I'm doing something wrong, but at this moment there is no documentation about the various backends.

### To Reproduce

In `lv_conf.h` set only `LV_USE_GPU_SDL`.

```c
  /* Use the 'monitor' driver which creates window on PC's monitor to simulate a display*/
  sdl_init();

  /*Create a display buffer*/
  static lv_disp_draw_buf_t disp_buf1;
  sdl_gpu_disp_draw_buf_init(&disp_buf1);

  /*Create a display*/
  static lv_disp_drv_t disp_drv;
  sdl_gpu_disp_drv_init(&disp_drv);
  disp_drv.draw_buf = &disp_buf1;
  disp_drv.flush_cb = sdl_display_flush;
  disp_drv.hor_res = SDL_HOR_RES;
  disp_drv.ver_res = SDL_VER_RES;
  lv_disp_t * disp = lv_disp_drv_register(&disp_drv);

   lv_example_btn_1();
```

Gives this (I've clicked "Ignore" a few times)
[image]

### Expected behavior

I sure the issue is on my end but others can have the same issue. So we should have a brief description about the SDL backend here:
https://docs.lvgl.io/master/get-started/index.html

Probably the sub-pages should be grouped as:
- Vendors
- RTOS
- Binding
- Draw backend (I'll write about the SW drawing)

cc @mariotaku 
Could you take a look at this issue?

mariotaku, 2022-02-25:
I think the driver/backend wasn't property configured. I'll check later.

kisvegabor, 2022-03-08:
Is there something I can check meanwhile?

mariotaku, 2022-03-09:
Sorry I was in another project. Just updated a working example: https://github.com/mariotaku/lv_sim_vscode_sdl

kisvegabor, 2022-03-10:
Thanks, however it's still not working :( Just black window.

I've noticed that lvgl in that repo is ~ month old. Could you try with the latest version?

mariotaku, 2022-03-10:
Sure! My project will soon be using lvgl and I'll keep everything up to date:)

rqbh, 2022-03-13:
blow code modify fix the crash

```
void lv_draw_sdl_init_ctx(lv_disp_drv_t * disp_drv, lv_draw_ctx_t * draw_ctx)
{
    _lv_draw_sdl_utils_init();
    lv_memset_00(draw_ctx, sizeof(lv_draw_sdl_ctx_t));
    draw_ctx->draw_rect = lv_draw_sdl_draw_rect;
    draw_ctx->draw_img = lv_draw_sdl_img_core;
    draw_ctx->draw_letter = lv_draw_sdl_draw_letter;
    draw_ctx->draw_line = lv_draw_sdl_draw_line;
    draw_ctx->draw_arc = lv_draw_sdl_draw_arc;
    draw_ctx->draw_polygon = lv_draw_sdl_polygon;
    draw_ctx->draw_bg = lv_draw_sdl_draw_bg;
    lv_draw_sdl_ctx_t * draw_ctx_sdl = (lv_draw_sdl_ctx_t *) draw_ctx;
    draw_ctx_sdl->renderer = disp_drv->user_data;//((lv_draw_sdl_drv_param_t *) disp_drv->user_data)->renderer;
    draw_ctx_sdl->internals = lv_mem_alloc(sizeof(lv_draw_sdl_context_internals_t));
    lv_memset_00(draw_ctx_sdl->internals, sizeof(lv_draw_sdl_context_internals_t));
    lv_draw_sdl_texture_cache_init(draw_ctx_sdl);
}
```

rqbh, 2022-03-13:
but without AA when draw line

kisvegabor, 2022-03-17:
Thanks @rqbh! I merged #3167

Maybe @mariotaku can comment on the line AA issue.

mariotaku, 2022-03-17:
@rqbh @kisvegabor Sorry for being absent for a long time. I'd like to point out the PR #3167 might not be the best way to solve this issue.  Users are supposed to allocate [lv_draw_sdl_drv_param_t](https://github.com/lvgl/lvgl/blob/70c2b57abe640adb70b671d55726cd052daa419e/src/draw/sdl/lv_draw_sdl.h#L42) and assign to userdata of disp_drv. Because more fields might be added to the struct, passing Renderer alone isn't enough.

As for the anti aliasing issue, right now SDL driver draws rectangle texture with rotation for lines. To make sure textures are rotated with anti aliasing, refer to `SDL_HINT_RENDER_SCALE_QUALITY`.

kisvegabor, 2022-03-29:
> Sorry for being absent for a long time. I'd like to point out the PR https://github.com/lvgl/lvgl/pull/3167 might not be the best way to solve this issue. Users are supposed to allocate [...]

I'm also sorry for the late reply. A PR would be welcome :)

kisvegabor, 2022-04-14:
I pin this issue, because it'd be important to have it fixed.

kisvegabor, 2022-05-25:
The [fix](https://github.com/lvgl/lv_drivers/commit/462d9cdeb83df9838f3f02e6f156fc37599c9add) is merged so I close issue.

closed 2022-05-25 by kisvegabor