lvgl/lvgl #3129: Long press from input device

jbarrena, 2022-02-24:
Hi.

I have a keypad with 5 buttons. I am able to get LV_EVENT_LONG_PRESSED of LV_KEY_ENTER (assigned to one input of keypad).

Is there any option to get long press events for the rest of the buttons? Perhaps with a LV_INDEV_TYPE_BUTTON indev driver instead of LV_INDEV_TYPE_KEYPAD type?

Thanks for your help!

Julen.

kisvegabor, 2022-02-24:
Hi,

`LV_KEY_ENTER` is special as it works as if you have clicked the object with a touch pad but really there is no such functionality for `LV_KEY_RIGHT/LEFT/UP/DOWN` or other buttons. 

I'm not sure it works in your case but you can but you do this too:
1. Create a `lv_btn`s a `lv_layer_top()`
2. Create an  `LV_INDEV_TYPE_BUTTON` and target each `lv_btn` with a hardware button
3. Use the events of the `lv_btn`s
4. Make the `lv_btn`s transparent

A proper solution would be to have `LV_EVENT_KEY_LONG_PRESSED` and `LV_EVENT_KEY_LONG_PRESSED_REP` events besides `LV_EVENT_KEY`. If you are interested in contributing with this feature a PR would be highly appreciated :slightly_smiling_face:

jbarrena, 2022-03-01:
Hi.

Finally I modified lv_indev.c file and managed to have all keys' LV_EVENT_LONG_PRESSED event:

/*Long press time has elapsed?*/
        if(i->proc.long_pr_sent == 0 && lv_tick_elaps(i->proc.pr_timestamp) > i->driver->long_press_time) {
            i->proc.long_pr_sent = 1;
            // enable LONG PRESS event for all the buttons if(data->key == LV_KEY_ENTER) {
                i->proc.longpr_rep_timestamp = lv_tick_get();
                lv_event_send(indev_obj_act, LV_EVENT_LONG_PRESSED, indev_act);
                if(indev_reset_check(&i->proc)) return;
            //}
        }

Thank you for the help. Really appreciated.

Regards.

Julen.

kisvegabor, 2022-03-08:
Can you send a Pull request to better see the modifications?

---
As the issue is solved close it now.

Finomnis, 2024-12-19:
`LV_EVENT_LONG_PRESSED` is incorrect for this, as it indicates that the widget itself is long pressed, not a key.
Adding the events `LV_EVENT_KEY_LONG_PRESSED` and `LV_EVENT_KEY_LONG_PRESSED_REP` would be indeed the proper solution.

Was there a feature request for this somewhere? Or was this issue just dismissed now?

kisvegabor, 2024-12-22:
> LV_EVENT_LONG_PRESSED is incorrect for this, as it indicates that the widget itself is long pressed, not a key.
Adding the events LV_EVENT_KEY_LONG_PRESSED and LV_EVENT_KEY_LONG_PRESSED_REP would be indeed the proper solution. It would also require an additional LV_EVENT_KEY_SHORT_PRESSED event do distinguish between long and short presses, and LV_EVENT_KEY_RELEASED.

Very true! There is no PR yet. Opening in one would be very very welcome! :slightly_smiling_face: