lvgl/lvgl #3130: Proposed lv_rlottie improvements

tvanfossen, 2022-02-24:
### Introduce the problem
Extending lv_rlottie animation controls to have more control over lottie animations after they have been created.

### Examples and cases
Case A: A single lottie file with multiple states embedded at different frames (play from frame 0-60 for state 1, play from 61-120 for state 2 etc)
Case B: A single lottie file where frame is tied to some state (this can be controlled with the existing lv_rlottie_set_current_frame() function and the playmode additions)
Case C: Altering the size and position of a lottie animation after creation to match other animations on screen

### Suggested solution
I've gone ahead and put together some stubs for some relevant functions, the below are the scope in my mind right now:

```
void lv_rlottie_set_play_mode(lv_obj_t * rlottie, const lv_rlottie_ctrl_t ctrl);
void lv_rlottie_set_current_frame(lv_obj_t * rlottie, const size_t goto_frame);
void lv_rlottie_set_framerate(lv_obj_t* obj, const int framerate);
void lv_rlottie_set_render_width(lv_obj_t* obj, const int width);
void lv_rlottie_set_render_height(lv_obj_t* obj, const int height);
void lv_rlottie_set_dest_frame(lv_obj_t* obj, const int dest_frame);
void lv_rlottie_set_start_frame(lv_obj_t* obj, const int start_frame);

int lv_rlottie_get_framerate(lv_obj_t* obj);
int lv_rlottie_get_render_width(lv_obj_t* obj);
int lv_rlottie_get_render_height(lv_obj_t* obj);
size_t lv_rlottie_get_start_frame(lv_obj_t* obj);
size_t lv_rlottie_get_dest_frame(lv_obj_t* obj);
size_t lv_rlottie_get_current_frame(lv_obj_t* obj);
lv_rlottie_ctrl_t lv_rlottie_get_play_mode(lv_obj_t* obj);
```

I've implemented set_render_width and set_render_height to be able to resize in real time.
[image]

This was built on an esp32s3 with octal spi, 8mb of flash and 8mb of psram on ESP-IDF4.4. There are 2 50x50 (water drop and fan) lottie animations built from the json as embedded escaped strings which are being resized to 25x25 and re-positioned (both animations attached to the same lv_anim_timeline)

@kisvegabor been sitting on the documentation for rlottie setup for esp for a bit now, will try and get that added in as part of this - I've only been able to get the idf to build when using rlottie as a component, haven't been able to get it going with the installed lib in the same fashion as the current rlottie documentation

```
void lv_rlottie_set_render_width(lv_obj_t* obj, const int width)
{
    lv_rlottie_t * rlottie = (lv_rlottie_t *) obj;
    rlottie->imgdsc.header.w = width;

    rlottie->scanline_width = width * LV_ARGB32 / 8;

    if(rlottie->task) {
        size_t allocaled_buf_size = (width * rlottie->imgdsc.header.h * LV_ARGB32 / 8);

        rlottie->allocated_buf = heap_caps_realloc(rlottie->allocated_buf, allocaled_buf_size+1, MALLOC_CAP_SPIRAM);
        
        rlottie->imgdsc.data = (void *)rlottie->allocated_buf;
        rlottie->imgdsc.data_size = allocaled_buf_size;
        lv_img_set_src(obj, &rlottie->imgdsc);
        
        if(rlottie->allocated_buf != NULL) {
            rlottie->allocated_buffer_size = allocaled_buf_size;
            memset(rlottie->allocated_buf, 0, allocaled_buf_size);
        }
    }
}

void lv_rlottie_set_render_height(lv_obj_t* obj, const int height)
{
    lv_rlottie_t * rlottie = (lv_rlottie_t *) obj;
    rlottie->imgdsc.header.h = height;

    if(rlottie->task) {
        size_t allocaled_buf_size = (rlottie->imgdsc.header.w * height * LV_ARGB32 / 8);

        rlottie->allocated_buf = heap_caps_realloc(rlottie->allocated_buf, allocaled_buf_size+1, MALLOC_CAP_SPIRAM);
        
        rlottie->imgdsc.data = (void *)rlottie->allocated_buf;
        rlottie->imgdsc.data_size = allocaled_buf_size;
        lv_img_set_src(obj, &rlottie->imgdsc);

        if(rlottie->allocated_buf != NULL) {
            rlottie->allocated_buffer_size = allocaled_buf_size;
            memset(rlottie->allocated_buf, 0, allocaled_buf_size);
        }
    }
    
}
```

Additionally, in order to animate between sizes cleanly, I implemented some lv_anim_t helpers

```
static void anim_set_lottie_render_width_cb(void * var, int32_t v)
{
    lv_rlottie_set_render_width(var, v);
};

lv_anim_t set_lottie_render_width(lv_obj_t* obj, int start_width, int target_width, uint32_t anim_speed, lv_anim_path_cb_t anim_path)
{
    lv_anim_t a;

    lv_anim_init(&a);
    lv_anim_set_var(&a, obj);
    lv_anim_set_exec_cb(&a, (lv_anim_exec_xcb_t)anim_set_lottie_render_width_cb);
    lv_anim_set_path_cb(&a, (lv_anim_path_cb_t)anim_path);
    lv_anim_set_delay(&a, ANIM_DELAY_START_MS);
    lv_anim_set_time(&a, anim_speed);
    lv_anim_set_values(&a, start_width, target_width);

    return a;
}

static void anim_set_lottie_render_height_cb(void * var, int32_t v)
{
    lv_rlottie_set_render_height(var, v);
};

lv_anim_t set_lottie_render_height(lv_obj_t* obj, int start_height, int target_height, uint32_t anim_speed, lv_anim_path_cb_t anim_path)
{
    lv_anim_t a;

    lv_anim_init(&a);
    lv_anim_set_var(&a, obj);
    lv_anim_set_exec_cb(&a, (lv_anim_exec_xcb_t)anim_set_lottie_render_height_cb);
    lv_anim_set_path_cb(&a, (lv_anim_path_cb_t)anim_path);
    lv_anim_set_delay(&a, ANIM_DELAY_START_MS);
    lv_anim_set_time(&a, anim_speed);
    lv_anim_set_values(&a, start_height, target_height);

    return a;
}
```

The returned lv_anim_t object is then attached to an lv_anim_timeline, along with other animations which are running on the native LVGL widgets on screen.

kisvegabor, 2022-02-28:
Hi,

@X-Ryl669 is working on something similar in #2904

It's a long discussion and PR but the gist is:
- Add generic controls to `lv_img` to play/pause/seek animations
- Rework the image converters to better support animations (e.g. rlottie)
- Add `lv_animbtn` which can loop parts of the animation in various states

With @X-Ryl669 we agreed on [scenario](https://github.com/lvgl/lvgl/pull/2904#issuecomment-1047616125) how to proceed with these updates. 

@X-Ryl669 what about opening an issue where we describe what is the current state and goals and continue the discussion?

X-Ryl669, 2022-02-28:
We can use this one.

What's in #2904 currently use my fork of Rlottie. 
This fork contains:
1. Partial rendering (also called *windowed* rendering), where a buffer that's smaller than the final picture is used and rendered multiple time, each time moving the ROI. This allow rendering Rlottie animations on ESP32 without a PSRAM, or benefit from DMA memory for rendering (much faster than SPIRAM). 
2. A rewritten JSON parser that's not copying the JSON file to heap (unlike rapidjson). This means that you don't waste your JSON file size in heap like currently. 
3. If you are using a ESPFS partition, you can memory-map the rlottie file directly from flash (with `espfs_faccess` function) so you're as memory efficient as possible
4. Multiple fix to rlottie's bugs (like the painful limitations where you can't render the last frame of the animation)

Based on this, the PR adds:
1. Partial rendering from the rlottie's decoder. This is useful to limit the usage of heap required, understand: it also works on ESP32 with or without SPIRAM (no ESP32s3 required). The rendering can be done line by line or with a limited buffer height
2. Support for animations control (playing forward, backward, looping, event when animation done, going to a specific frame, pausing, etc...)
3. Adding `lv_animbtn` widget that's a button that can play animation (work with Rlottie & others) either on a transition of its state, or when it has reached its state
4. Support for vector format. The image size is no more fixed you can change it anytime and the button will adjust. This cause a re-rendering on rlottie to the new size, obviously.
5. Removing the dependency to run the image converter on the Rlottie files (or any non RAW format)
6. Allow setting any text with symbols in them (so "Ok LV_SYMBOL_OK" works now)
7. Better image caching that does not open the picture multiple time to extract metadata like size and color format
8. Native format for rlottie decoder (no need to call any `lv_rlottie_xxx` function, just call `lv_img_set_src(img, "/path/to/lottie.json") and it'll work. It means that LVGL animations and style works out of the box.

tvanfossen, 2022-02-28:
@X-Ryl669 Should I close this issue and move discussion to #2904 ?

tvanfossen, 2022-02-28:
Animation control covered in #2904, so setting start/end frame covered in that PR?
will the img src for lottie adjust width/height in realtime using standard lv_obj_set_width/height? (i.e. is the render_height/width needed)
Will the PR cover frame rate changes? (slow down/speed up animation)

X-Ryl669, 2022-02-28:
Yes, yes and no. I did not know you could change the framerate of the animation in runtime with rlottie. 

But since the `lv_img_t` now has a timer function where all the work happens, it can be changed by `lv_timer` class of API. 
Since rlottie only uses the frame index to build its frame, it should work. I don't know if it'll work for GIF or ffmpeg through.

tvanfossen, 2022-02-28:
I was thinking of doing it through the next_frame_task callback timer, not in rlottie, but it was looking very choppy when I tried the first time. It would just be calling rlottie at a slower rate than the what is provided by the lottie json through that 

So sounds like everything I was looking at with this change will be covered in your PR, is there anything I can do to help out there?

closed 2022-03-08 by tvanfossen