lvgl/lvgl #3131: lv_label : Abnormal display in Chinese+English

GuoFlying, 2022-02-25:
### Perform all steps below and tick them with [x]
- [ x ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ x ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x ] Update lvgl to the latest version
- [ x ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

- After the label size is set, the Chinese and English labels are not displayed completely
- It is also abnormal in LONG_DOT mode
- **However, in the case of English + Chinese, it is normal**

### To Reproduce

```c
    {
        lv_obj_t *label = lv_label_create(lv_scr_act());
        lv_label_set_text(label, "嵌abcdefghijklnm ");
        lv_obj_set_style_text_font(label, &lv_font_simsun_16_cjk, 0);
        lv_obj_set_size(label, 80, 16);
        lv_obj_align(label, LV_ALIGN_TOP_LEFT, 5, 0);
    }

    {
        lv_obj_t *label = lv_label_create(lv_scr_act());
        lv_label_set_text(label, "abcdefghijklnm嵌 ");
        lv_obj_set_style_text_font(label, &lv_font_simsun_16_cjk, 0);
        lv_obj_set_size(label, 80, 16);
        lv_obj_align(label, LV_ALIGN_TOP_LEFT, 5, 30);
    }
    {
        lv_obj_t *label = lv_label_create(lv_scr_act());
        lv_label_set_text(label, "嵌abcdefghijklnm ");
        lv_label_set_long_mode(label, LV_LABEL_LONG_DOT);
        lv_obj_set_style_text_font(label, &lv_font_simsun_16_cjk, 0);
        lv_obj_set_size(label, 80, 16);
        lv_obj_align(label, LV_ALIGN_TOP_LEFT, 5, 60);
    }
    {
        lv_obj_t *label = lv_label_create(lv_scr_act());
        lv_label_set_long_mode(label, LV_LABEL_LONG_DOT);
        lv_label_set_text(label, "abcdefghijklnm嵌 ");
        lv_obj_set_style_text_font(label, &lv_font_simsun_16_cjk, 0);
        lv_obj_set_size(label, 80, 16);
        lv_obj_align(label, LV_ALIGN_TOP_LEFT, 5, 90);
    }
```

### Expected behavior

- The display length of "Chinese + English" and "English + Chinese" is the same

### Screenshots or video

- The above code produces the following error
[image]

kisvegabor, 2022-03-01:
I've track what could be the issue and I found that the problem comes from here:
https://github.com/lvgl/lvgl/blob/406386dc3f2e6b40d1af600800ca0cc2a2766377/src/misc/lv_txt.h#L170-L173

If you comment out these lines, all cases look normal:
[image]

As the text in your example can't fit in one line LVGL tries to break it and prefers the CJK character to do so.

Feedback from native people about how to handle it would be highly appreciated. 

My opinion is this text:
`This is a nice computer. 计算机被用作各种工业和消费设备的控制系统。`
 
Should wrap like this:
```
This is a nice
computer. 计算
机被用作各种工业
和消费设备的控制
系统。
```

What do you think, how it should work?

GuoFlying, 2022-03-02:
### hi kisvegabor:

thank you for the response!

### Here's what I think:

The multi-line processing is separated from the single-line processing Or opens the interface to the user for configuration(We don't want to change the LVGL source code during development)
And I hope that single line and multiple lines can have a control switch, so as to better control the label.(QLabel->setWordWrap in QT  )

kisvegabor, 2022-03-03:
Actually it's quite flexible in LVGL how you can handle word wrapping with the `lv_label_set_long_mode` you have already discovered.

If you want to avoid word warring just use `lv_obj_set_width(label, LV_SIZE_CONTENT)`. It will set the widget's width to the width of the text so no word wrapping will occure.

If you really want to set the width to specific value without having word wrapping you can use the `LV_LABEL_LONG_CLIP` mode.

GuoFlying, 2022-03-07:
I see. Thank you for your reply.

Back to the original problem.
Keeping the LVGL source code intact helps us to iterate with the LVGL version.  If the source code of LVGL is not changed, do you have any suggestions for the problem of "Chinese + English"?

GuoFlying, 2022-03-07:
our scenario looks like this: the screen width may be smaller than the text, and some of the text is user-configurable, so we have to do truncation or DOT

kisvegabor, 2022-03-07:
It just came to my mind LVGL has a built-in feature to handle long words. Actually a Chinese text looks like long word as there are no spaces in it.

Setting `LV_TXT_LINE_BREAK_LONG_LEN 8` in `lv_conf.h` solved the issue for me.

GuoFlying, 2022-03-08:
That's great. That really helps us. Thank yo.

 
From: Gabor Kiss-Vamosi
Date: 2022-03-07 18:31
To: lvgl/lvgl
CC: Flying; Author
Subject: Re: [lvgl/lvgl] lv_label : Abnormal display in Chinese+English (Issue #3131)
It just came to my mind LVGL has a built-in feature to handle long words. Actually a Chinese text looks like long word as there are no spaces in it.
Setting LV_TXT_LINE_BREAK_LONG_LEN 8 in lv_conf.h solved the issue for me.

kisvegabor, 2022-03-10:
Happy to hear that! :slightly_smiling_face:

closed 2022-03-10 by kisvegabor