lvgl/lvgl #3136: Image color is not normal when using SPI screen to display JPG

LY3T, 2022-02-26:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Image color is not normal when using SPI screen to display JPG
### To Reproduce

My hardware is STM32H743+ 2.0-inch LCD (drive IC is ST7789).  When I run and download the following code, the color of the picture is abnormal.  
```c
lv_obj_t * img = lv_img_create(lv_scr_act());
lv_img_set_src(img, "A:/wallpaper_1.jpg");
```
[image]

### Expected behavior

The original image should look something like this:

[image]

Then I found that the reason was that the decoded image data did not exchange high and low bytes, so I made two changes in the `lv_sjpg.c` file (actually only two macro definitions were added).  

```c
//The first modification is on line 769
#if  LV_BIG_ENDIAN_SYSTEM == 1 || LV_COLOR_16_SWAP == 1

//The second change is on line 833
#if  LV_BIG_ENDIAN_SYSTEM == 1 || LV_COLOR_16_SWAP == 1
```

Then the image color is normal
[image]

### Screenshots or video

that is all.Thank you for reading.
This is the first time FOR me to submit an issue. Please forgive me if THERE is something wrong.  My English is not good, I used translation software, haha.  Also, I think LVGL is a great GUI library, and I hope to support more third-party libraries in the future.

LY3T, 2022-02-27:
> 您是否将宏**_设置_**`LV_COLOR_16_SWAP`为`1`lv_conf.h ？**__**
> 
> ```c
> #定义 LV_COLOR_16_SWAP      1
> ```
Yes, I did.

GorgonMeducer, 2022-02-27:
Why not submit a PR?

LY3T, 2022-02-27:
> 为什么不提交 PR？

Because I don't know how to submit a PR.

kisvegabor, 2022-03-01:
A Pull request would be really great! Here you can see how to send one: https://docs.lvgl.io/master/CONTRIBUTING.html#pull-request

kisvegabor, 2022-03-01:
It seems you have figured it out.  :slightly_smiling_face: 
#3138 

Thank you!

closed 2022-03-01 by kisvegabor