lvgl/lvgl #3142: Can we add low level interface for accelerating rotation

GorgonMeducer, 2022-03-02:
### Introduce the problem
So far, there is no interface in `lv_draw_ctx_t` for rotation. Can we add them?

### Examples and cases

### Suggested solution

GorgonMeducer, 2022-03-02:
Hi @kisvegabor,

Here is the API used in arm-2d, which accelerate rotation:

```c
            arm_2dp_rgb565_tile_rotation_with_alpha(
                                            &(ptItem->tOP),
                                            ptItem->ptTile,     //!< source tile
                                            ptTile,             //!< target tile
                                            &(ptItem->tRegion), //!< target region
                                            ptItem->tCentre,    //!< center point
                                            ptItem->fAngle,     //!< rotation angle
                                            GLCD_COLOR_BLACK,   //!< masking colour
                                            ptItem->chOpacity,  //!< Opacity
                                            ptItem->ptTargetCentre);
```

https://github.com/ARM-software/EndpointAI/blob/master/Kernels/Research/Arm-2D/examples/watch_panel/example_gui.c

I have finished a `lv_gpu_arm2d` and it gives a significant performance uplift for Cortex-M55 (taking advantage of Helium, a SIMD instruction set).  Helium can be used to accelerate rotation, and it is very good. That's why I want an interface from LVGL to for taking advantage of arm-2d rotation APIs.

I am not sure whether 8x8 or 16X16 is a good idea or not. In Arm-2D, you don't have to load a picture (resource) to RAM first, instead, you can create an arm_2d_tile_t header and pointer the resource stored in ROM. The rotation API can accept the source tile (pointer to a resource on ROM), and store the rotated results in the display buffer directly. 

In this process, 

- You don't need extra RAM as a working-ram
- You can only rotate part of the resource

So, if you can pass the whole image resource to arm-2d, you can get the best performance uplift. If you just pass 8x8 or 16*16, arm-2d can work with it (because it supports only rotating a specified part of a given picture), but I doubt that there will be some performance loss.

Let's see how can we find a balanced point between the existing solution in LVGL and a solution which can squeeze the last drop from a GPU (or SIMD based soft-GPU).

What do you think?

kisvegabor, 2022-03-03:
A few questions:
- What is `center point on source picture` and `center point on the display buffer`? The pivot of the rotation?
- Is scaling supported as well?
- If a 20x10 image is rotated by 90° resulting in a 10x20 image, what should be the `target region on the display buffer`?

There is something I can't imagine now. Let's say there is a 2x3 image:
```
A B
C D
E F
```
rotated by 90° (pivot is on `E` (0;2)), resulting in
```
- - -
- - -
E C A
F D B
```

In the complex cases (like rotation) LVGL draws the UI line by line. It know the bounding box of the rotated image (3x2) so for the first pass it wants to calculate the first 3 pixels (`E C A`). The trick is that LVGL (it's common trick actually) scans the destination pixels. For example `A` on the rotated image has (2;0) coordinates (relative to the pivot `E`)  and LVGL asks "Hey image rotate module, what is the color of the 2;0 pixel if I rotate this source image by  90° around that pivot?" The image rotate module rotates the (0;2) coordinate by -90° (note the minus sign) resulting in 0;-2 (relative to the pivot again) which is `A` .

The problem is that I can't see how can do the same with the ARM's API. And it would be important because LVGL knows only the destination area ("Okay, I need to redraw this part of the rotated image"). 

After I wrote all these down I started to suspect we can do it with `target region on the display buffer`. :smile: 
E.g. setting it to 
```c
x1 = 10
x2 = 50
y1 = 30
y2 = 30
```
to get one line from the rotated image. Can you confirm this?

GorgonMeducer, 2022-03-03:
> the trick is that LVGL (it's common trick actually) scans the destination pixels.

Yes, arm-2d uses the same trick.
When using Arm-2d rotation APIs, you only need to tell the API:

- The destination tile 
- The destination area inside the tile (It can be smaller than the destination tile)
- The source tile

Since we are using the same trick, I didn't see any problem.

For those questions:
> What is center point on source picture and center point on the display buffer? 

Consider there are two pieces of paper, one represents the source picture(source tile) and the other represents the target picture (target tile). We use one pin to penetrate the two, so the source picture can rotate on the target picture. 
After removing the pin, you can find holes on both of the papers. That's the rotation centres (pivot).

> Is scaling supported as well?
Planned in the roadmap.

> If a 20x10 image is rotated by 90° resulting in a 10x20 image, what should be the target region on the display buffer?
In arm-2d, all 2d-operations go through a pipeline that includes a standard region clipping feature. Taking advantage of this arrangement, users don't need to care about the size or shape of tiles (source tile and target tiles).
For example, in the case you described, we don't care about the shape or size of the source image. You didn't mention anything about the target area, that's the key to answering your question: 
* If the rotation result is out of the target area (based on the location of the pivot and the size of the rotated result), it will be clipped. 
* If the rotation result is inside the target area, the picture is in one piece. 

In brief, arm-2d APIs are designed in a "Simple and intuitive" manner. 

> After I wrote all these down I started to suspect we can do it with target region on the display buffer. 😄
E.g. setting it to
```c
x1 = 10
x2 = 50
y1 = 30
y2 = 30
```
> to get one line from the rotated image. Can you confirm this?

Big YES.

kisvegabor, 2022-03-08:
Thanks for the great images. Only one question left from my side: Is there a way to handle ARGB8565 format? I.e. RGB565 with and extra alpha channel.

GorgonMeducer, 2022-03-08:
@kisvegabor YES.

kisvegabor, 2022-03-08:
> and you don’t really need a mask for the source picture.

Just to confirm: mask should be passed separately (A8 array + RGB565 array) or as one array?

To have it implemented I see these steps:
1. Agree on an API for `ctx->img_transform()` (together)
2. Use `ctx->img_transform()` in `lv_draw_img` (also refactor the current SW transform for compatibility) (I can do it)
3. Add new transform engines
  - ARM version (@GorgonMeducer) 
  - Update/optimize the SW transform (@kisvegabor)

GorgonMeducer, 2022-03-09:
@kisvegabor 
> Just to confirm: mask should be passed separately (A8 array + RGB565 array) or as one array?

Please pass them separately (A8 Array + RGB565 Array)

By the way, can I pull a PR request and add `lv_gpu_arm2d.c` and `lv_gpu_arm2d.h` as other GPU drivers?

Those files can be found [here](https://github.com/lvgl/lv_port_an547_cm55_sim).  After that, we can find even more opportunities to accelerate LVGL.

kisvegabor, 2022-03-10:
> About the RGBA8888, my suggestion is that pass the mask and the RGBA8888 as separated arrays, but with a special arrangement:

If there are any kind of masking or other "trick" in LVGL, during the image rendering process the ARGB images are separated to two arrays but not exactly in the way you have described. E.g. from ARGB8565 there will be an `uint8_t A8` and an `uint16_t RGB565` array. From `ARGB8888` an `uint8_t A8` and an `uint32_t XRGB8888` array. Can it be used this way? If not probably we can improve this part of LVGL too. 

> By the way, can I push a PR request and add lv_gpu_arm2d.c and lv_gpu_arm2d.h as an optional default GPU driver for all cortex-m processors?

Yes, it'd be great!

GorgonMeducer, 2022-03-10:
> Can it be used this way? If not probably we can improve this part of LVGL too.
YES, no problem at all. The solution I proposed is just a way to save RAM size but not mandatory.

GorgonMeducer, 2022-03-11:
Arm-2D support is added in PR #3162

kisvegabor, 2022-03-18:
Let's find an API for the `transform_cb`. First idea:

```c
transorm(img_src, img_size, img_format, target_area, dest_buf, dest_buf_stride, transform_params);
```
- `img_format`: e.g. RGB565, ARGB8888
- `transform_params`: pivot, angle, zoom

Regarding the alpha channel the best would be if we could pass the `ARGB` source as it is and the result would be stored as `ARGB` in `dest_buf`. Please confirm if it works.

GorgonMeducer, 2022-03-18:
@kisvegabor For RGB565+Alpha, if you pass ARGB together (16bit+8byteAlpha), arm-2d would not able to accelerate it. Please provide a separated alpha. 
For RGBA8888, it is OK that you pass the source directly. 

By the way, the zooming feature has just been added to arm-2d.

kisvegabor, 2022-03-22:
> I notice that in the LVGL official image convertor if you select RGB16 with Alpha, the alpha will be stored together with the RGB channels. I think this has the following drawbacks:

It makes sense. However, in the current way we can handle  ARGB8565 and ARGB8888 with the same code. So it's just more uniform... but really slower...

One thing is still not clear: if we have the RGB + Alpha together (as now) and they needs to passed separately to the `transform_cb()` LVGL needs to create an image sized RGB and an Alpha buffer, right?

Let's consider some common color formats to see what could be a common way to handle them:
- ARGB8888: aligned, standard, all clear
- RGB888: not aligned, shall be padded to 4 bytes?
- ARGB8565: not aligned, non standard, problematic
- ARGB1555: aligned, not that standard,. Separating alpha channel would be wasteful
- ARGB2222: aligned, not that standard,. Separating alpha channel would be wasteful
- ARGB4444: aligned, not that standard,. Separating alpha channel would be wasteful

> It requires runtime decoding (separate alpha from RGB channels)

So there are other formats too where we can't avide decoding

Being inspired by [your comment](https://github.com/lvgl/lvgl/issues/3142#issuecomment-1062460848) a working a pattern would be to define the offset and size of the color and alpha channels for each color format. It also opend the doors for user defined color formats. E.g. 
- ARGB8565: 
  - rgb_size: 16 bit
  - rgb_offset: 8 bit
  - a_size: 8 bit
  - a_offset: 0 bit

> By the way, the zooming feature has just been added to arm-2d.

Amazing!

GorgonMeducer, 2022-03-22:
> Being inspired by https://github.com/lvgl/lvgl/issues/3142#issuecomment-1062460848 a working a pattern would be to define the offset and size of the color and alpha channels for each color format. It also opend the doors for user defined color formats. E.g. ARGB8565

Yes, but ARGB8565 is not a good example. For performance, we should also add a rule that each pixel should align to a natural integer's size, i.e. 2, 4, 8 etc.

> RGB888: not aligned, shall be padded to 4 bytes?

Please padded to 4 bytes. or maybe when padding, put 0xFF to the 4th byte and use as a normal ARGB8888.

> ARGB8565: not aligned, non standard, problematic

I have a suggestion:

a. separate ARGB8565 into two arrays
b. store mask array after the rgb565 array immediately, so we can still use one starting address for the image.
c. Since we know the h and w of each image, it is easy to get the start address of the alpha-mask-array.

> ARGB1555: aligned, not that standard,. Separating alpha channel would be wasteful
> ARGB2222: aligned, not that standard,. Separating alpha channel would be wasteful
> ARGB4444: aligned, not that standard,. Separating alpha channel would be wasteful

Using masks to support the alpha channels of those formats might not be a good idea. If arm-2d would like to support them, arm-2d would handle alpha directly together with those rgb channels.

kisvegabor, 2022-03-24:
I see two problems:
- RGB8888 paddied to 4 bytes wastes +33% memory (and memory bandwidth). If its a screen sized wallpaper it really matters.
- For ARGB8565: I see the advantage of separating the alpha array but it'd be a non generic solution. I.e. in all other formats the A channel is with RGB except with ARGB8565.

If ARM-2D can handle ARGB8565 and other special formats too (e.g. ARGB2222) and the only drawback is loosing some performance I'd go for the generic solution. It can be the user's choice the balance between memory usage (RGB888) or performance (XRGB8888).

GorgonMeducer, 2022-03-25:
@kisvegabor About RGB888, please consider in a real application decision:

- When you have a constraint environment, or in other words, you want to have a cost-sensitive product, you won't choose RGB888 or even RGBA8888. Usually, people will choose SPI/8080 LCD with RGB565 instead.
- When you want to choose RGB888/RGBA888, that means you will choose a resource-rich environment, and usually, those big images will be stored in a very cheap external FLASH hence the 33% memory overhead is nothing important.
    * If you have such concerns, please check with people who design products. 

For ARGB8565, it is usually used in a constraint environment, where performance is also important when memory cost is the same. Providing an unaligned format is not a good idea.

I only provide my input and it's up to you to make any decision. by the way, I don't think ARGB8565 is a standard format that people will accept widely (it's my personal option).

xiaoxiang781216, 2022-03-25:
> I see two problems:
>
> * RGB8888 paddied to 4 bytes wastes +33% memory (and memory bandwidth). If its a screen sized wallpaper it really matters.
> * For ARGB8565: I see the advantage of separating the [...]

@kisvegabor @GorgonMeducer here is my thought:

The separated format is not only more friendly for optimization, e.g.:

1. RGB channel and A channel can always access on the natural alignment
2. CPU could process 8 pixels by SIMD(MVE for M55 and NEON for Cortex A) instruction without the preprocess

but also more compatible with the display panel, since RGB565 panel can ignore the second part of alpha channel directly, otherwise the display driver need drop alpha channel from A8RGB565 before sending to panel which is also a time consuming operation.

So, the performance lost with the interlaced A8RGB565 format may make this format useless in the real product.

kisvegabor, 2022-03-28:
I see your points. So in summarize I see it like this:
- **ARGB8888** is fine
- **RGB888** is ok, but slow
-  **XRGB8888**: (padded RGB888) for performance reasons.  I.e. let the users choose between the 3 bytes and 4 byte variants
- **RGB565**: fine 
- **RGB565A8** (let's call it this way now): store RGB565 and A8 separately because it's faster then compact ARGB8565
- **ARGB8565**: keep the compact for for consistency with other formats
- **ARGB2222** or **ARGB1555** (an other "semi-standard" formats): store them compactly as they are special and we are focusing on the most common formats.

In short, we should support 2 new formats: `XRGB888` and `RGB565A8`. This way the users can select which format they prefer.

GorgonMeducer, 2022-03-28:
@kisvegabor GREAT!!!

I am really happy that you introduce those two new formats !!!

GorgonMeducer, 2022-03-29:
@kisvegabor It's not a push, but just curious, when can we make this happen?

kisvegabor, 2022-03-29:
I'm planning to make some initial work this week (just to see there are no blocking parts) and finish it next week.

kisvegabor, 2022-03-31:
I started to work on it and got a questions.  The stages of the most complex image drawing case are these:
1. transforms (might use GPU)
2. chromakeying (set some pixel's alpha to 0)
3. recoloring (recolor the rgb values)
4. masking (update the alpha mask)
5. blending  (might use GPU)

So there are these cases:
### Pure sw rendering
Clean
- sw transform -> sw blend
- sw transform -> other sw effect -> sw blend

### Pure hw rendering
Transform could blend directly in most of the cases, so there is no point to return the rotate image and blend it after that
- hw transform ->  hw blend (should be one step)
- hw transform -> other sw effect -> hw blend (most be separate steps)

### Mixed case 1
Transform should return the rotated image  
- hw transform ->  sw blend
- hw transform -> other sw effect -> sw blend

### Mixed case 2 
Clean, we are do in this now
- sw transform ->  hw belnd
- sw transform -> other sw effect -> hw blend

---

So it seems there should be two API functions:
- `transform_and_blend`: try this first, if it could do it, we are happy
- `transform_to_buf`: `lv_draw_sw_transform_and_blend` keep calling it because the HW still can be used to speed up transformations.

Do you agree with this?

kisvegabor, 2022-03-31:
And 2 minutes after sending my comment I've realized that basically `transform_and_blend` is the same as the current 
`draw_ctx->draw_img_decoded`.

So you can overwrite `draw_img_decoded` to try drawing the image with all the transformations, masks, color format stuff, blending, etc. and if you can't do it (e.g. can not handle lvgl masks) you can fallback to SW rendering which will call `transform_to_buf`.

GorgonMeducer, 2022-03-31:
> And 2 minutes after sending my comment I've realized that basically transform_and_blend is the same as the current
draw_ctx->draw_img_decoded.

That's exactly what we tried to do as a plan B if you don't introduce a new transform interface.

kisvegabor, 2022-04-01:
> Although it looks like we go back to the starting point, it is at least more clear than before when there is no document that clearly indicates the transform is done inside the draw_img_decoded.

I've just rewritten the `draw_sw_img_decoded` to make it simpler and faster. The core part looks like this now:
https://github.com/lvgl/lvgl/blob/feat/transform_api/src/draw/sw/lv_draw_sw_img.c#L376-L424

I prepared it (and lvgl in general) to handle other image color formats as discussed before. Now the goal is NOT to decode the images to match LVGL's color settings but to decode to a format supported by the image draw function.

The new version is much easier to extend with other formats. Both `convert_cb` and `transform_cb` should provide a separate rgb and alpha array.  

I also re-implemented the SW transformation which now accepts a block to transform. The not antialiased zoom+rotation became 3 times faster and the antialiased version 2 times faster. 

> So, do you still plan to introduce a dedicated low-level interface for transform or do you want to update documents instead?

I definitely would like to keep the new version because it's:
- much faster
- much simpler
- transform_cb can be hooked in complex cases too when e.g. there is a mask, recolor, or other effect. 

Nevertheless the docs should be updated too.

The current implementation is just a proof of concept and not finished yet. Not all color formats are supported, and the transform callback is not added to `draw_ctx` yet. I can finish it next week.

GorgonMeducer, 2022-04-01:
It sounds like a very good re-factory. 
I am looking forward to it.

kisvegabor, 2022-04-04:
I've added some more optimization and support for the current color formats.
The API is also added: 
https://github.com/lvgl/lvgl/blob/45f95a4a01fabd473aae7168e5bdcf4c60f3ce8e/src/draw/sw/lv_draw_sw.h#L71-L73

GorgonMeducer, 2022-04-04:
Looks good. 
But only after starting my porting work, I don't have any confidence to give you useful feedback.

kisvegabor, 2022-04-05:
No problem, I just wanted to let you know about the current state. 

I need to add some backward compatibility macros and after that I'll merge it to master to see if any issues come up.

kisvegabor, 2022-04-26:
Along with the "[widget transformation](https://docs.lvgl.io/master/overview/style.html#opacity-blend-modes-and-transformations)" update I merged the new transform API and improved SW transformation into master.

I close this issue now. If anything comes up about the API, please open an new issue.

closed 2022-04-26 by kisvegabor