lvgl/lvgl #3144: crash occurs when the canvas's buffer setting is small

FASTSHIFT, 2022-03-03:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
When setting a canvas with a size of 1*1 and the color format as `LV_IMG_CF_TRUE_COLOR`,`lv_canvas_set_buffer()` sets a buffer with a size of **8 bytes**, which is passed to the jpeg decoder through the decoder framework. The `is_jpg()` function accesses **10 bytes**, resulting in a memory out of bounds.
The decoder lacks matching data size when image src is `LV_IMG_SRC_VARIABLE`.

```c
static int is_jpg(const uint8_t * raw_data)
{
    const uint8_t jpg_signature[] = {0xFF, 0xD8, 0xFF,  0xE0,  0x00,  0x10, 0x4A,  0x46, 0x49, 0x46};
    return memcmp(jpg_signature, raw_data, sizeof(jpg_signature)) == 0;
}
```

### To Reproduce
In `lv_conf.h`
```
#define LV_COLOR_DEPTH 16
#define LV_USE_SJPG 1
```

```c
#define CANVAS_WIDTH  1
#define CANVAS_HEIGHT  1

static void canvas_test(void)
{
    /*Create a buffer for the canvas*/
    static lv_color_t cbuf[LV_IMG_BUF_SIZE_TRUE_COLOR(CANVAS_WIDTH, CANVAS_HEIGHT)];

    /*Create a canvas and initialize its palette*/
    lv_obj_t * canvas = lv_canvas_create(lv_scr_act());
    lv_canvas_set_buffer(canvas, cbuf, CANVAS_WIDTH, CANVAS_HEIGHT, LV_IMG_CF_INDEXED_1BIT);
}
```

### Expected behavior

### Screenshots or video
```bash
[Info]  (0.000, +0)      lv_init: begin         (in lv_obj.c line #100)
[Warn]  (0.000, +0)      lv_init: Style sanity checks are enabled that uses more RAM    (in lv_obj.c line #168)
[Info]  (0.000, +0)      lv_obj_create: begin   (in lv_obj.c line #202)
[Info]  (0.000, +0)      lv_obj_create: begin   (in lv_obj.c line #202)
[Info]  (0.000, +0)      lv_obj_create: begin   (in lv_obj.c line #202)
[Info]  (0.000, +0)      lv_canvas_create: begin        (in lv_canvas.c line #56)
=================================================================
==18903==ERROR: AddressSanitizer: global-buffer-overflow on address 0x555556545224 at pc 0x7ffff727f31d bp 0x7fffffffd150 sp 0x7fffffffc8f8
READ of size 10 at 0x555556545224 thread T0
    #0 0x7ffff727f31c  (/usr/lib/x86_64-linux-gnu/libasan.so.5+0xd931c)
    #1 0x555555a3c00e in is_jpg /home/vifextech/workpath/lv_sim_eclipse_sdl/lvgl/src/extra/libs/sjpg/lv_sjpg.c:895
    #2 0x555555a306d5 in decoder_info /home/vifextech/workpath/lv_sim_eclipse_sdl/lvgl/src/extra/libs/sjpg/lv_sjpg.c:178
    #3 0x555555b845a7 in lv_img_decoder_get_info /home/vifextech/workpath/lv_sim_eclipse_sdl/lvgl/src/draw/lv_img_decoder.c:99
    #4 0x555555c5a0b7 in lv_img_set_src /home/vifextech/workpath/lv_sim_eclipse_sdl/lvgl/src/widgets/lv_img.c:106
    #5 0x555555c41077 in lv_canvas_set_buffer /home/vifextech/workpath/lv_sim_eclipse_sdl/lvgl/src/widgets/lv_canvas.c:78
    #6 0x55555597eab0 in canvas_test /home/vifextech/workpath/lv_sim_eclipse_sdl/main.c:106
    #7 0x55555597ead1 in main /home/vifextech/workpath/lv_sim_eclipse_sdl/main.c:137
    #8 0x7ffff5f3ac86 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21c86)
    #9 0x55555597e5e9 in _start (/home/vifextech/workpath/lv_sim_eclipse_sdl/demo+0x42a5e9)

0x555556545224 is located 0 bytes to the right of global variable 'cbuf' defined in 'main.c:102:23' (0x555556545220) of size 4
0x555556545224 is located 60 bytes to the left of global variable 'disp_buf1' defined in 'main.c:162:31' (0x555556545260) of size 40
SUMMARY: AddressSanitizer: global-buffer-overflow (/usr/lib/x86_64-linux-gnu/libasan.so.5+0xd931c) 
Shadow bytes around the buggy address:
  0x0aab2aca09f0: f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9
  0x0aab2aca0a00: f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9
  0x0aab2aca0a10: f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9
  0x0aab2aca0a20: f9 f9 f9 f9 00 00 f9 f9 f9 f9 f9 f9 00 00 00 00
  0x0aab2aca0a30: f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9 f9
=>0x0aab2aca0a40: 00 00 00 00[04]f9 f9 f9 f9 f9 f9 f9 00 00 00 00
  0x0aab2aca0a50: 00 f9 f9 f9 f9 f9 f9 f9 00 00 00 00 00 00 00 00
  0x0aab2aca0a60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0aab2aca0a70: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0aab2aca0a80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0aab2aca0a90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==18903==ABORTING
[1] + Done                       "/usr/bin/gdb" --interpreter=mi --tty=${DbgTerm} 0<"/tmp/Microsoft-MIEngine-In-0axpq52i.nlp" 1>"/tmp/Microsoft-MIEngine-Out-5qnyfcrj.hhh"
```

kisvegabor, 2022-03-03:
I seems it can be fixed by:
```c
static int is_jpg(const uint8_t * raw_data, size_t len)
{
    const uint8_t jpg_signature[] = {0xFF, 0xD8, 0xFF,  0xE0,  0x00,  0x10, 0x4A,  0x46, 0x49, 0x46};
    if(len < raw_data, sizeof(jpg_signature)) return false;
    return memcmp(jpg_signature, raw_data, sizeof(jpg_signature)) == 0;
}
```

What do you think?

FASTSHIFT, 2022-03-04:
@kisvegabor 
I agree, but how do I pass the len parameter from src?

kisvegabor, 2022-03-04:
It should be already set in [raw_sjpeg_data_size](https://github.com/lvgl/lvgl/blob/master/src/extra/libs/sjpg/lv_sjpg.c#L161).

FASTSHIFT, 2022-03-05:
> It should be already set in [raw_sjpeg_data_size](https://github.com/lvgl/lvgl/blob/master/src/extra/libs/sjpg/lv_sjpg.c#L161).

PNG libraries should do the same, as should other decoders that rely on file headers to identify files

kisvegabor, 2022-03-08:
True!

kisvegabor, 2022-03-25:
Closed by #3205

closed 2022-03-25 by kisvegabor