lvgl/lvgl #3149: fs_stdio: directly listing fails when using multiple directory handles (WIN32)

Ryzee119, 2022-03-05:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Having multiple directly handles and calling lv_fs_dir_read between them will break the filename listings. Only tested in WIN32 backend, but it appears to be isolated here.

### To Reproduce
Use fs_stdio backend with WIN32
```
lv_fs_dir_t dir_a;
lv_fs_dir_t dir_b;

lv_fs_dir_open(&dir_a, "A:folder1");
lv_fs_dir_open(&dir_b, "A:folder2");
char fname[256];
lv_fs_dir_read(&dir_a, fname); //Read 1st file in dir_a;
lv_fs_dir_read(&dir_a, fname); //Read 2nd file in dir_a;

lv_fs_dir_read(&dir_b, fname); //Read 1st file in dir_b;

lv_fs_dir_read(&dir_a, fname); //SHOULD read 3rd file in dir_a; but it returns the 2nd file in dir_b.
```

### Expected behavior
You should be able to read from different directory handles and keep track of the current files in each directory independently.

Failing due to a shared `next_fn` buffer:
https://github.com/lvgl/lvgl/blob/ff0bf6386d702ff9f32528fe6296c8da326c9914/src/extra/libs/fsdrv/lv_fs_stdio.c#L192

kisvegabor, 2022-03-12:
Hi,

It's great that you have found the reason of this issue. 

Unfortunately, I'm on Linux and I don't really have experience with the WIN32 API so probably others can suggest a good solution. 

@mariotaku do you have any idea for this?

Ryzee119, 2022-03-20:
My fix was to create a directory handle struct which contained `next_fn` which is allocated on each `dir_open` call.

My diff is here:
https://github.com/Ryzee119/lvgl/commit/6d2b6726a384eb435f412e0ccd628ba4fc0eb156

kisvegabor, 2022-03-24:
Thanks @Ryzee119! Could you send a PR with the fix?

mariotaku, 2022-03-25:
@kisvegabor Hi. On Windows I use mingw instead of msvc, and I don't use file IO much, so I can't figure out.

Ryzee119, 2022-03-31:
Raised a PR https://github.com/lvgl/lvgl/pull/3243 :)

kisvegabor, 2022-04-04:
Merged, thank you! :)

closed 2022-04-04 by kisvegabor