lvgl/lvgl #3150: When using freetype to load TTF font library, the program will close when words not in the font library are displayed

wurenjia, 2022-03-07:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

    When using freetype to load TTF font library, the program will close when words not in the font library are displayed
    linux/arm/LVGL8.2/
     but log ：
[Error] (0.490, +40)     lv_font_get_glyph_bitmap: Asserted at expression: font_p != NULL (NULL pointer)        (in lv_font.c line #51)

### To Reproduce

```c
    ui_wifi_switch_text_label = lv_label_create(ui_wifi_switch);
    lv_obj_set_pos(ui_wifi_switch_text_label,0,0);
    lv_obj_set_height(ui_wifi_switch_text_label,LV_PCT(90));
    lv_obj_add_style(ui_wifi_switch_text_label,&style_text_1,0);
    lv_label_set_text(ui_wifi_switch_text_label,"AX3😁😁111111111112222");
    lv_obj_set_style_text_align(ui_wifi_switch_text_label,LV_ALIGN_LEFT_MID,0);
    lv_obj_align_to(ui_wifi_switch_text_label,ui_wifi_switch,LV_ALIGN_LEFT_MID,0,0);
```
### Expected behavior

### Interim solution
```c
const uint8_t * lv_font_get_glyph_bitmap(const lv_font_t * font_p, uint32_t letter)
{
    /**wyw add to solve the bug that the program will be down when displaying words what are not in the font library*/
    if(font_p == NULL){
        return 0;
    }
    /**wyw add end*/
    LV_ASSERT_NULL(font_p);
    return font_p->get_glyph_bitmap(font_p, letter);
}
```

### Screenshots or video

kisvegabor, 2022-03-07:
@guoweilkd knows the LVGL's FreeType integration the best so I cc him. :slightly_smiling_face:

guoweilkd, 2022-03-07:
Seems to be caused by this reason #3000.

wurenjia, 2022-03-09:
emmmm,I'm using the version of January 31 
commit ：
[image]

thanks very much @kisvegabor @guoweilkd

wurenjia, 2022-03-09:
yes

closed 2022-03-09 by kisvegabor

wurenjia, 2022-05-23:
sorry,i use to solved this issue by update commit to [a6c4c13](https://github.com/lvgl/lvgl/commit/a6c4c134902f9a4c156672a70108e809b58fa18c)，not by [release v8.2 0b5a1d4b2](https://github.com/lvgl/lvgl/commit/0b5a1d4b23975b920ff841ea9cd038802f51711b)