lvgl/lvgl #3151: Rendering error when LV_OBJ_FLAG_OVERFLOW_VISIBLE and clip_corner are enabled

FASTSHIFT, 2022-03-08:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Rendering error when `LV_OBJ_FLAG_OVERFLOW_VISIBLE` and `clip_corner` are enabled.

### To Reproduce
```c
static void overflow_test()
{
    lv_obj_t * par = lv_obj_create(lv_scr_act());
    lv_obj_add_flag(par, LV_OBJ_FLAG_OVERFLOW_VISIBLE);
    lv_obj_set_style_clip_corner(par, true, 0);
    lv_obj_set_style_pad_all(par, 0, 0);
    lv_obj_set_height(par, 30);
    lv_obj_center(par);

    lv_obj_t * slider = lv_slider_create(par);
    lv_obj_set_height(slider, 30);
}
```

### Expected behavior
When `LV_OBJ_FLAG_OVERFLOW_VISIBLE` is enabled, `clip_corner` does not need to take effect.

### Screenshots or video

https://user-images.githubusercontent.com/26767803/157190483-273e7eb4-4b5f-422f-92b0-9fc89c05d046.mp4

kisvegabor, 2022-03-08:
It's clearly a not defined case, but I'm not sure which setting should be preferred. 

For me always clip corners sound more intuitive.

FASTSHIFT, 2022-03-08:
> LV_OBJ_FLAG_OVERFLOW_VISIBLE

If `clip_corner` is enable, the priority should indeed be higher than `LV_OBJ_FLAG_OVERFLOW_VISIBLE`.

FASTSHIFT, 2022-03-08:
> It's clearly a not defined case, but I'm not sure which setting should be preferred.
> 
> For me always clip corners sound more intuitive.

I think `LV_OBJ_FLAG_OVERFLOW_VISIBLE` should have a higher priority, and when the user add this flag, there cannot be any clipping behavior.

embeddedt, 2022-03-08:
> I think LV_OBJ_FLAG_OVERFLOW_VISIBLE should have a higher priority, and when the user add this flag, there cannot be any clipping behavior.

That behavior would match HTML:

[image]

<details>
<summary>Code</summary>

```html
<!doctype html>
<html>
    <body>
        <style>
            div {
                width: 300px;
                height: 200px;
                margin: 10px;
                border-radius: 40px;
                border: 4px red solid;
                display: inline-block;
                overflow: visible;
            }
        </style>
        <div>
Lorem ipsum dolor sit amet, consectetur adipiscing elit. Nunc lacus est, accumsan eget lorem id, lobortis lobortis dui. Curabitur lacinia libero odio, non imperdiet elit elementum nec. Etiam in massa sed augue venenatis gravida vel at sapien. Quisque nec iaculis neque. Nam magna urna, scelerisque non scelerisque vitae, condimentum eu orci. Donec euismod dui sit amet ante maximus, sit amet semper ante consequat. Aliquam quis mattis sapien. Aliquam sed sollicitudin neque, quis maximus urna. Curabitur non sapien justo. Sed porttitor, lectus ac consequat lacinia, lacus magna congue justo, ut accumsan tellus nulla eget lectus. Cras nec imperdiet lorem. Quisque rutrum purus et elementum tristique.

Aliquam et fringilla mi, at rutrum purus. Sed pharetra lobortis facilisis. In hac habitasse platea dictumst. Morbi tristique eros sed nulla suscipit, ut sodales nisi posuere. Phasellus a eros ullamcorper, luctus diam eu, blandit lectus. Proin ultrices pharetra enim, eget feugiat risus hendrerit non. Sed non urna lobortis erat auctor feugiat nec et purus.

Curabitur in massa turpis. Fusce ac magna massa. Integer nec velit quis dolor ullamcorper vehicula. Mauris quis odio sem. Vestibulum ante ipsum primis in faucibus orci luctus et ultrices posuere cubilia curae; Pellentesque ipsum velit, convallis ut velit id, ornare suscipit massa. Donec tempus lacinia ultrices. Vestibulum fringilla bibendum molestie. Ut sollicitudin erat nec auctor tincidunt. Cras ipsum leo, eleifend porttitor odio posuere, varius tincidunt arcu. Phasellus eget massa id sem fringilla efficitur ut viverra tellus. Curabitur ut arcu id augue posuere interdum. Morbi vel hendrerit nisi, id bibendum orci.

Nulla orci elit, commodo sed augue id, sodales placerat turpis. Donec faucibus, felis faucibus egestas imperdiet, arcu nisi luctus quam, porttitor tincidunt ligula dolor nec ipsum. Interdum et malesuada fames ac ante ipsum primis in faucibus. Integer iaculis at est a scelerisque. Quisque sit amet justo urna. Maecenas auctor, mi vel suscipit scelerisque, felis mi vestibulum est, ac hendrerit lectus sapien ac est. In vehicula volutpat arcu, eu imperdiet mi vulputate a. Duis euismod dui nisl, ac hendrerit est euismod sit amet. Phasellus viverra et lorem in pretium. Etiam malesuada augue urna, quis pellentesque augue gravida eu. Pellentesque eget dapibus metus, sit amet ullamcorper risus. Mauris congue purus lacus, id porttitor leo tincidunt egestas. Duis in odio facilisis, commodo nisl at, viverra mi.

Cras eu luctus augue. Maecenas vulputate massa nec justo blandit convallis. Curabitur lobortis, nunc in fringilla mattis, mauris lorem dictum metus, vitae tempor elit massa et nibh. Donec sollicitudin malesuada felis, in laoreet felis. Etiam eget aliquet ante. Integer quis egestas elit. Integer nec dui vel mi congue venenatis quis eu ipsum. Proin auctor sodales neque, tempor semper lacus. Donec at luctus odio. Cras vehicula maximus velit at tristique. Nullam fringilla neque eros, vitae pretium sem auctor vitae. Sed ac porta metus, eu molestie augue. Maecenas ultrices risus orci, et mollis diam aliquam ac. Aenean lobortis sodales arcu.
        </div>
    </body>
</html>
```

</details>

kisvegabor, 2022-03-08:
> I think LV_OBJ_FLAG_OVERFLOW_VISIBLE should have a higher priority, and when the user add this flag, there cannot be any clipping behavior.

I'm thinking the opposite because clip corner is just a mask added by LVGL and the user can add other masks in draw events.
And we can't selectively ignore the masks added by the user if `LV_OBJ_FLAG_OVERFLOW_VISIBLE`.

 
> That behavior would match HTML:

LVGL should work like this with `LV_OBJ_FLAG_OVERFLOW_VISIBLE`. However what if an SVG mask clips the content?

FASTSHIFT, 2022-03-09:
@kisvegabor 
Is there any way to disable only `clip_corner`? Other mask behaviors are reserved.

kisvegabor, 2022-03-10:
> Is there any way to disable only clip_corner?
 
Sure, `..._style_slip_corner` controls it.

kisvegabor, 2022-03-10:
It's also possible to write something like this [here](https://github.com/lvgl/lvgl/blob/45b0f86ae5efb6fc1a006f52d713898b50ecd98b/src/core/lv_obj.c#L548-L563):
```c
if(has_clip_corner(obj) && !has_overfow_visibel(obj)) {
  add_corner_mask();
}
```

But I still believe masks should be "stronger"  than overflow.

FASTSHIFT, 2022-03-13:
> It's also possible to write something like this [here](https://github.com/lvgl/lvgl/blob/45b0f86ae5efb6fc1a006f52d713898b50ecd98b/src/core/lv_obj.c#L548-L563):
>
> ```c
> if(has_clip_corner(obj) && [...]

I think it is more reasonable to follow the specification of the web and reduce the confusion caused by users using `LV_OBJ_FLAG_OVERFLOW_VISIBLE` :)

kisvegabor, 2022-03-16:
See this: https://codepen.io/kisvegabor/pen/VwyvzBP
[image]

The text overflows the red box but it also clipped later.

So it seems on web masks are really stronger than overflow.

FASTSHIFT, 2022-03-31:
> 

OK, I agree to make the clip a higher priority. How to fix this drawing error?

FASTSHIFT, 2022-03-31:
> See this: https://codepen.io/kisvegabor/pen/VwyvzBP [image]
> 
> The text overflows the red box but it also clipped later.
> 
> So it seems on web masks are really stronger than overflow.

OK, I agree to make the clip a higher priority. How to fix this drawing error?

kisvegabor, 2022-04-04:
It seems it's already fixed :smile:

FASTSHIFT, 2022-04-04:
@kisvegabor 
I pulled the latest code. Strangely, I don't see the strange color blocks shown in the video above on the Visual Studio and frame buffer simulators, but there are still color blocks in the SDL Eclipse simulator.

FASTSHIFT, 2022-04-04:
@kisvegabor 
I pulled the latest code, the strange color block shown in the video above still exists in the SDL Eclipse emulator, have you set `lv_indev_set_cursor()`?

kisvegabor, 2022-04-04:
It's interesting that now I see the issue. The cursor was set previously as well.

I've pushed a fix, can you confirm it works well now?

closed 2022-04-04 by kisvegabor

FASTSHIFT, 2022-04-05:
> It's interesting that now I see the issue. The cursor was set previously as well.
> 
> I've pushed a fix, can you confirm it works well now?

It works fine. Thank you.

commit a7f9dfa8 (2022-04-04): fix(refr): consider masks with LV_OBJ_FLAG_OVERFLOW_VISIBLE

fixes: #3151