lvgl/lvgl #3157: Objects are drawn in the default state first

w-dherman, 2022-03-10:
### Perform all steps below and tick them with [x]
- [ x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x] Update lvgl to the latest version
- [ x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

If an objects state is changed directly after creating the object the library seems to draw the object in the default state before the state the object was changed to. This causes a flickering effect.

### To Reproduce

```c
static void myExample(void)
{
    lv_obj_t *container = lv_obj_create(lv_scr_act());
    lv_obj_set_size(container, 320, 480);

    lv_obj_t *button = lv_btn_create(container);
    lv_obj_set_size(button, 200, 75);
    lv_obj_center(button);

    lv_obj_t* label = lv_label_create(button);
    lv_label_set_text(label, "Test");
    lv_obj_center(label);

    lv_obj_add_state(button, LV_STATE_DISABLED);
}
```

### Expected behavior

In the example above I would expect the button to only be drawn one time in the disabled state.

### Screenshots or video

https://user-images.githubusercontent.com/53914160/157726404-d1f9b812-d2a6-456e-b131-47cb4e23726e.mp4

kisvegabor, 2022-03-12:
Hi,

It's because there is a state change transition in the default theme. You can disable it by setting `LV_THEME_DEFAULT_TRANSITION_TIME 0`.

kisvegabor, 2022-03-18:
You are right. I've pushed a fix.

commit 242112b2 (2022-03-18): fix(theme): properly disable transitions if LV_THEME_DEFAULT_TRANSITION_TIME==0

fixes #3157
closed 2022-03-18 by kisvegabor