lvgl/lvgl #3158: With scroll_to_view, it does not return when FLEX_FLOW_ROW is enabled

hasankarainfo, 2022-03-10:
I encountered the related issue in 8.1.0 and 8.3.0

As in the title, when lv_obj_set_flex_flow(pg, LV_FLEX_FLOW_ROW) is active, with the lv_obj_scroll_to_view function, the page scrolls in one direction at first. When it comes to the first object after the last object, scrolling does not occur. 

```	
lv_obj_t *ui_pgsens = 0;
// pgsens
ui_pgsens = lv_obj_create(p->pg_sec);

lv_obj_set_width(ui_pgsens, 240);
lv_obj_set_height(ui_pgsens, 120);

lv_obj_set_x(ui_pgsens, 0);
lv_obj_set_y(ui_pgsens, 0);

lv_obj_set_align(ui_pgsens, LV_ALIGN_TOP_MID);

lv_obj_set_scrollbar_mode(ui_pgsens, LV_SCROLLBAR_MODE_ACTIVE);
lv_obj_set_scroll_dir(ui_pgsens, LV_DIR_ALL);

lv_obj_set_style_radius(ui_pgsens, 0, LV_PART_MAIN | LV_STATE_DEFAULT);
lv_obj_set_style_bg_color(ui_pgsens, lv_color_hex(0xFFFFFF), LV_PART_MAIN | LV_STATE_DEFAULT);
lv_obj_set_style_bg_opa(ui_pgsens, 0, LV_PART_MAIN | LV_STATE_DEFAULT);
lv_obj_set_style_border_width(ui_pgsens, 0, LV_PART_MAIN | LV_STATE_DEFAULT);
lv_obj_set_style_pad_left(ui_pgsens, 5, LV_PART_MAIN | LV_STATE_DEFAULT);
lv_obj_set_style_pad_right(ui_pgsens, 5, LV_PART_MAIN | LV_STATE_DEFAULT);
lv_obj_set_style_pad_top(ui_pgsens, 5, LV_PART_MAIN | LV_STATE_DEFAULT);
lv_obj_set_style_pad_bottom(ui_pgsens, 5, LV_PART_MAIN | LV_STATE_DEFAULT);

lv_obj_set_flex_flow(ui_pgsens, LV_FLEX_FLOW_ROW);
lv_obj_set_style_flex_track_place(ui_pgsens, LV_FLEX_ALIGN_SPACE_BETWEEN, 0);
lv_obj_set_style_flex_cross_place(ui_pgsens, LV_FLEX_ALIGN_CENTER, 0);
lv_obj_set_style_layout(ui_pgsens, LV_LAYOUT_FLEX, 0);
lv_obj_set_style_pad_column(ui_pgsens, 10, 0);
```
but the same feature works fine when the related line is changed as below.
```
lv_obj_set_flex_flow(ui_pgsens, LV_FLEX_FLOW_COLUMN);
```

hasankarainfo, 2022-03-16:
I create sub-objects as below
```
			void *rec_ofs;
			ck_obj_ts *ck = sens_get_next(&rec_ofs);
			for (int i = 0; ck; i++, ck = sens_get_next(&rec_ofs))
			{
				_mscreen_sens_ts *mss = _memory_allocation(sizeof(_mscreen_sens_ts));
				nullset(mss, _mscreen_sens_ts);
				qnode_set_obj(&mss->qn, mss);
				queue_sendforce_node(&p->q_sens, &mss->qn);
				if (rec_ofs)
				{
					mss->ofs_val = rec_ofs;
					mss->rec_val = *(float *)rec_ofs - 0.01;
				}
				mss->ck = ck;

				lv_obj_t *ui_pgsens1a = 0;
				// pgsens1a
				ui_pgsens1a = lv_obj_create(ui_pgsens);

				lv_obj_set_width(ui_pgsens1a, 110);
				lv_obj_set_height(ui_pgsens1a, 110);

				lv_obj_set_x(ui_pgsens1a, 6);
				lv_obj_set_y(ui_pgsens1a, 5);

				lv_obj_clear_flag(ui_pgsens1a, LV_OBJ_FLAG_SCROLLABLE);

				lv_obj_set_style_radius(ui_pgsens1a, 5, LV_PART_MAIN | LV_STATE_DEFAULT);
				lv_obj_set_style_bg_img_src(ui_pgsens1a, &ui_img_img_bg_main_s2_png, LV_PART_MAIN | LV_STATE_DEFAULT);
				lv_obj_set_style_border_width(ui_pgsens1a, 0, LV_PART_MAIN | LV_STATE_DEFAULT);
				lv_obj_set_style_shadow_width(ui_pgsens1a, 20, LV_PART_MAIN | LV_STATE_DEFAULT);
				lv_obj_set_style_shadow_spread(ui_pgsens1a, 1, LV_PART_MAIN | LV_STATE_DEFAULT);
				lv_obj_set_style_pad_left(ui_pgsens1a, 0, LV_PART_MAIN | LV_STATE_DEFAULT);
				lv_obj_set_style_pad_right(ui_pgsens1a, 0, LV_PART_MAIN | LV_STATE_DEFAULT);
				lv_obj_set_style_pad_top(ui_pgsens1a, 0, LV_PART_MAIN | LV_STATE_DEFAULT);
				lv_obj_set_style_pad_bottom(ui_pgsens1a, 0, LV_PART_MAIN | LV_STATE_DEFAULT);

				mss->pg = ui_pgsens1a;
				
				lv_obj_t *ui_head1a = 0;
				// head1a

				ui_head1a = lv_label_create(ui_pgsens1a);

				lv_obj_set_width(ui_head1a, 110);
				lv_obj_set_height(ui_head1a, 40);

				lv_obj_set_x(ui_head1a, 0);
				lv_obj_set_y(ui_head1a, 0);

				lv_obj_set_align(ui_head1a, LV_ALIGN_TOP_MID);

				lv_label_set_long_mode(ui_head1a, LV_LABEL_LONG_SCROLL_CIRCULAR);
				lv_label_set_text(ui_head1a, "Temp SHTC3( C)");

				lv_obj_set_style_text_color(ui_head1a, lv_color_hex(0xFFFFFF), LV_PART_MAIN | LV_STATE_DEFAULT);
				lv_obj_set_style_text_opa(ui_head1a, 255, LV_PART_MAIN | LV_STATE_DEFAULT);
				lv_obj_set_style_text_align(ui_head1a, LV_TEXT_ALIGN_CENTER, LV_PART_MAIN | LV_STATE_DEFAULT);
				lv_obj_set_style_text_font(ui_head1a, &lv_font_montserrat_20, LV_PART_MAIN | LV_STATE_DEFAULT);
				lv_obj_set_style_radius(ui_head1a, 5, LV_PART_MAIN | LV_STATE_DEFAULT);
				lv_obj_set_style_bg_color(ui_head1a, lv_color_hex(0x101EFF), LV_PART_MAIN | LV_STATE_DEFAULT);
				lv_obj_set_style_bg_opa(ui_head1a, 255, LV_PART_MAIN | LV_STATE_DEFAULT);
				lv_obj_set_style_bg_grad_color(ui_head1a, lv_color_hex(0xE8E8E8), LV_PART_MAIN | LV_STATE_DEFAULT);
				lv_obj_set_style_bg_main_stop(ui_head1a, 128, LV_PART_MAIN | LV_STATE_DEFAULT);
				lv_obj_set_style_bg_grad_stop(ui_head1a, 255, LV_PART_MAIN | LV_STATE_DEFAULT);
				lv_obj_set_style_bg_grad_dir(ui_head1a, LV_GRAD_DIR_VER, LV_PART_MAIN | LV_STATE_DEFAULT);

				mss->head = ui_head1a;

				char str[64];
				tb_properties_ts *prop = ck->prop.p;
				sprintf(str, "%s %s (%s)", prop->label, ck->key, prop->units);
				lv_label_set_text(mss->head, str);

				int color_hex = 0;
				sscanf(prop->color, "#%x", &color_hex);
				lv_obj_set_style_bg_color(ui_head1a, lv_color_hex(color_hex), LV_PART_MAIN | LV_STATE_DEFAULT);

				mss->lbl = val_lbl_create(mss->pg, &lv_font_golong_44, 52);
			}
		}
```
I do the transition between objects as follows
```
static void sensor_roller(hyb_ts *hyb)
{
	gui_main_screen_ts *p = hyb->app_param;
	hyb_begin;
	static qnode_ts *qn;
	static int rec_sens, x;
	while (1)
	{
		hyb_delay(100);
		qn = queue_peek_node(&p->q_sens);
		while (qn)
		{
			_mscreen_sens_ts *mss = qnode_get_obj(qn);
			 lv_obj_scroll_to_view(mss->pg, LV_ANIM_ON);
			
			hyb_delay(5000);
			if (rec_sens != p->sensor_list_updated)
			{
				rec_sens = p->sensor_list_updated;
				break;
			}
			qn = qnode_peek_next(qn);
		}
	}

	hyb_end;
}
```
but i should mention i solved my own problem by using different function like below.
```
static void sensor_roller(hyb_ts *hyb)
{
	gui_main_screen_ts *p = hyb->app_param;
	hyb_begin;
	static qnode_ts *qn;
	static int rec_sens, x;
	while (1)
	{
		hyb_delay(100);
		x = 0;
		qn = queue_peek_node(&p->q_sens);
		while (qn)
		{
			lv_obj_scroll_to(p->pg_sensor, x, 0, LV_ANIM_ON);
			
			x += lv_obj_get_width(p->pg_sensor)/2;
			hyb_delay(5000);
			if (rec_sens != p->sensor_list_updated)
			{
				rec_sens = p->sensor_list_updated;
				break;
			}
			qn = qnode_peek_next(qn);
		}
	}

	hyb_end;
}
```

kisvegabor, 2022-03-17:
Still there are some undefined functions in the code (e.g. `sens_get_next`) but I've tried to reproduce it and found a typo in LVGL. Does it work if you replace this line
https://github.com/lvgl/lvgl/blob/4341660b93bda8eafbf836b2974285b39a68409c/src/core/lv_obj_scroll.c#L749
with 
```c
        if(sl - x_scroll < 0) x_scroll = 0;
```
this?

hasankarainfo, 2022-03-17:
when i updated the relevant line, the problem was solved.

closed 2022-03-17 by hasankarainfo

kisvegabor, 2022-03-18:
Thanks for the feedback, I pushed the fix.

commit 7c74f655 (2022-03-18): fix(scroll): fix scroll to view to the left

fixes #3158