lvgl/lvgl #3166: LVGL_Arduino.ino example fails to compile: fatal error: lv_demo.h: No such file or directory

ASMotionLab, 2022-03-13:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq) ✔
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)✔
- [ ] Update lvgl to the latest version✔
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html) ❌

LVGL_Arduino.ino fails to compile: fatal error: lv_demo.h: No such file or directory
[image]

Installed lvgl and lvgl_examples libraries from Arduino lib manager.
Followed "Get Started guide" to the word: https://docs.lvgl.io/master/get-started/arduino.html
Open example from Arduino>File>Examples>lvgl>Arduino>LVGL_Arduino
TFT_eSPI is working, it's demos run fine.
lv_conf.h is next to lvgl folder in Documents/Arduino
[image]

lv_conf.h content is enabled with "1". Colour depth and tick are correct.
lvgl/examples content is copied to lvgl/src/examples
lvgl/demos content is copied to lvgl/src/demos

When i uncommend ""#include <lv_examples.h>"" like suggested in the sketch, compile now fails at lv_examples.h not found.

This is the code (lvgl arduino demo)

```
#include <lvgl.h>
#include <TFT_eSPI.h>
/*If you want to use the LVGL examples,
  make sure to install the lv_examples Arduino library
  and uncomment the following line.
#include <lv_examples.h>
*/

#include <lv_demo.h>

/*Change to your screen resolution*/
static const uint16_t screenWidth  = 480;
static const uint16_t screenHeight = 320;

static lv_disp_draw_buf_t draw_buf;
static lv_color_t buf[ screenWidth * 10 ];

TFT_eSPI tft = TFT_eSPI(screenWidth, screenHeight); /* TFT instance */

#if LV_USE_LOG != 0
/* Serial debugging */
void my_print(const char * buf)
{
    Serial.printf(buf);
    Serial.flush();
}
#endif

/* Display flushing */
void my_disp_flush( lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p )
{
    uint32_t w = ( area->x2 - area->x1 + 1 );
    uint32_t h = ( area->y2 - area->y1 + 1 );

    tft.startWrite();
    tft.setAddrWindow( area->x1, area->y1, w, h );
    tft.pushColors( ( uint16_t * )&color_p->full, w * h, true );
    tft.endWrite();

    lv_disp_flush_ready( disp );
}

/*Read the touchpad*/
void my_touchpad_read( lv_indev_drv_t * indev_driver, lv_indev_data_t * data )
{
    uint16_t touchX, touchY;

    bool touched = tft.getTouch( &touchX, &touchY, 600 );

    if( !touched )
    {
        data->state = LV_INDEV_STATE_REL;
    }
    else
    {
        data->state = LV_INDEV_STATE_PR;

        /*Set the coordinates*/
        data->point.x = touchX;
        data->point.y = touchY;

        Serial.print( "Data x " );
        Serial.println( touchX );

        Serial.print( "Data y " );
        Serial.println( touchY );
    }
}

void setup()
{
    Serial.begin( 115200 ); /* prepare for possible serial debug */

    String LVGL_Arduino = "Hello Arduino! ";
    LVGL_Arduino += String('V') + lv_version_major() + "." + lv_version_minor() + "." + lv_version_patch();

    Serial.println( LVGL_Arduino );
    Serial.println( "I am LVGL_Arduino" );

    lv_init();

#if LV_USE_LOG != 0
    lv_log_register_print_cb( my_print ); /* register print function for debugging */
#endif

    tft.begin();          /* TFT init */
    tft.setRotation( 3 ); /* Landscape orientation, flipped */

    /*Set the touchscreen calibration data,
     the actual data for your display can be acquired using
     the Generic -> Touch_calibrate example from the TFT_eSPI library*/
    uint16_t calData[5] = { 275, 3620, 264, 3532, 1 };
    tft.setTouch( calData );

    lv_disp_draw_buf_init( &draw_buf, buf, NULL, screenWidth * 10 );

    /*Initialize the display*/
    static lv_disp_drv_t disp_drv;
    lv_disp_drv_init( &disp_drv );
    /*Change the following line to your display resolution*/
    disp_drv.hor_res = screenWidth;
    disp_drv.ver_res = screenHeight;
    disp_drv.flush_cb = my_disp_flush;
    disp_drv.draw_buf = &draw_buf;
    lv_disp_drv_register( &disp_drv );

    /*Initialize the (dummy) input device driver*/
    static lv_indev_drv_t indev_drv;
    lv_indev_drv_init( &indev_drv );
    indev_drv.type = LV_INDEV_TYPE_POINTER;
    indev_drv.read_cb = my_touchpad_read;
    lv_indev_drv_register( &indev_drv );

#if 0
    /* Create simple label */
    lv_obj_t *label = lv_label_create( lv_scr_act() );
    lv_label_set_text( label, LVGL_Arduino.c_str() );
    lv_obj_align( label, LV_ALIGN_CENTER, 0, 0 );
#else
    /* Try an example from the lv_examples Arduino library
       make sure to include it as written above.
    lv_example_btn_1();
   */

    // uncomment one of these demos
    lv_demo_widgets();            // OK
    // lv_demo_benchmark();          // OK
    // lv_demo_keypad_encoder();     // works, but I haven't an encoder
    // lv_demo_music();              // NOK
    // lv_demo_printer();
    // lv_demo_stress();             // seems to be OK
#endif
    Serial.println( "Setup done" );
}

void loop()
{
    lv_timer_handler(); /* let the GUI do its work */
    delay( 5 );
}
```

kisvegabor, 2022-03-17:
> Installed lvgl and lvgl_examples libraries from Arduino lib manager.

`lvgl_examples` is already merged into `lvgl`. So you need only LVGL. 

Does it work with `#include <demos/lv_demo.h>`

ljunquera, 2023-01-29:
I had the same problem. I just tried to use the example:
[image]

/Users/[me]/Documents/Arduino/libraries/lv_examples/src/../lv_demo.h:31:29: fatal error: ../lv_demo_conf.h: No such file or directory

The readme had some different instructions, but I do see that the examples are in lvgl. They don't appear in the Arduino UI, like the example above. I am just trying to get a simple "Hello World" and this seemed like a good place to start but it has been hard to find the current way to quick start as some of the legacy instructions have thrown me off.

ljunquera, 2023-01-29:
> #include <demos/lv_demo.h>

Didn't work for me, maybe because of the board "package" I'm using. Do I need this for the example?

kisvegabor, 2023-01-31:
Have you followed the steps [here](https://docs.lvgl.io/master/get-started/platforms/arduino.html#use-the-examples-and-demos)?
[image]

ljunquera, 2023-02-01:
I actually commented `lv_demo_widgets();` out for the moment and made the 0 a 1 to enable the code below:

```
#if 0
    /* Create simple label */
    lv_obj_t *label = lv_label_create( lv_scr_act() );
    lv_label_set_text( label, LVGL_Arduino.c_str() );
    lv_obj_align( label, LV_ALIGN_CENTER, 0, 0 );
#else
```

kisvegabor, 2023-02-01:
Then please follow the step linked above :slightly_smiling_face:

ljunquera, 2023-02-01:
Ok. I followed the instructions.

```
In file included from /Users/lj/Documents/Arduino/libraries/lv_examples/src/lv_demo.h:17:0,
                 from /private/var/folders/b9/jzy3mm9s1yx5sd1jh3lbs2gm0000gn/T/.arduinoIDE-unsaved2023029-353-my40m4.moe4k/LVGL_Arduino/LVGL_Arduino.ino:9:
/Users/lj/Documents/Arduino/libraries/lv_examples/src/../lv_demo.h:31:29: fatal error: ../lv_demo_conf.h: No such file or directory
compilation terminated.
```

kisvegabor, 2023-02-01:
It seems things are messed up a little bit. 

About a year ago, we have merged all the examples and demos into the lvgl repo. So `lv_examples` shouldn't be used. Just use the latest `lvgl` library and no other repo is required.

ljunquera, 2023-02-01:
Ok. Does that mean I should not use the UI, like I did above, to get to the LVGL_Arduino example? Even when I open the LVGL_Arduino.ino from the `lvgl/src/examples/arduino/LVGL_Arduino/` directory I get the `/Users/lj/Documents/Arduino/libraries/lv_examples/src/../lv_demo.h:31:29: fatal error: ../lv_demo_conf.h: No such file or directory` error. Is there a configuration I need to change?

ljunquera, 2023-02-01:
I uninstalled the examples library.

```
/private/var/folders/b9/jzy3mm9s1yx5sd1jh3lbs2gm0000gn/T/.arduinoIDE-unsaved202311-353-l30wbo.rnkc/LVGL_Arduino/LVGL_Arduino.ino:9:21: fatal error: lv_demo.h: No such file or directory
compilation terminated.
```

Fischkopppp, 2023-02-02:
You will find in Arduino\libraries\lv_examples\lv_demo_conf_template.h
Have you tried to copy the file and rename it to v_demo_conf.h? In this file you can turn different demo option on and off.

kisvegabor, 2023-02-04:
> I uninstalled the examples library.

There was a typo in `LVGL_Arduino.ino`. You should include `#include <demos/lv_demos.h>`. See 801d776e7

HapCoderWei, 2023-02-05:
Good job bro! i have been confused by this error for a week, now it solved by your advice. Thanks a lot👍

kisvegabor, 2023-02-05:
Great, happy to hear that it works now :slightly_smiling_face:

WirelessMind, 2023-02-12:
> Have you followed the steps [here](https://docs.lvgl.io/master/get-started/platforms/arduino.html#use-the-examples-and-demos)? [image]

I tried, but it doesn't work! :(

kisvegabor, 2023-02-14:
What error do you see?

WirelessMind, 2023-02-14:
`LVGL_Arduino.ino:8:10: fatal error: demos/lv_demo.h: No such file or directory #include <demos/lv_demo.h>`

WirelessMind, 2023-02-14:
Solved!
I copied demos folder must be copied into lvgl/src. I already tried, but perhaps with a different version.
Right now I'm using the various updated libraries, but the 8.3.0dev version lv_conf.h file that the vendors gave me, even though the library I'm using is the lvgl version 8.3.5.

dougwaterford, 2024-01-03:
Hi, I am attempting to compile the ELCROW HMI-7 demonstration and have this error HMI-7:2:10: fatal error: demos/lv_demos.h: No such file or directory I have followed your posts I have tried different library versions still with same result. Arduino version 1.8.19 OS LINUX MINT current lvgl library version 8.3.11.
I have tried the inclusion in a test file same result.

kisvegabor, 2024-01-03:
Hi,
have you followed [this](https://docs.lvgl.io/8.3/get-started/platforms/arduino.html#use-the-examples-and-demos) and seen [this](https://github.com/lvgl/lvgl/issues/3166#issuecomment-1429642134)?

pgaert, 2024-02-08:
Apologies is this is something simple.  I am also working with an Elecrow ESP32 display and attempting to run the demo program.  Moving the two folders -- demos and examples -- as described above cleared the errors on the relevant includes, and I now receive the following:  

../../src/display/lv_display_private.h: No such file or directory

In response to this, which must exist in some other included code:

#include "../../src/display/lv_display_private.h"

While I have confirmed that this file exists in the described location, this include generates a fatal error.  How can I fix this issue, please?

kisvegabor, 2024-02-08:
> #include "../../src/display/lv_display_private.h"

Where would you like to include it? If to the ino file I'm not what relative path shall be used.

pgaert, 2024-02-08:
I apologize if I was not clear.  This error comes up when compiling, but the include it references is not in the Elecrow demo code.  I am unfamiliar with the structure here, but assume it must in within one of the previously included files?  By this point, the sketch has included lvgl.h, lv_demos.h, and lv_examples.h (and a variety of other header files required for the display and its functions).

kisvegabor, 2024-02-11:
Please share 
- a longer compiler log (attache as a file)
- your *.ino file (attache as a file)
- the LVGL version you are using (see lvgl.h)

smileymiley, 2024-04-21:
@kisvegabor Thanks for posting the fix to the lv_demo.h not found issue
One of the links above no longer works but simply moving Arduino\libraries\lvgl\demos and examples folders to below src fixed it for me
(v8.3.6)

proximacentaury, 2024-04-25:
@kisvgabor Can you suggest a different strategy to chech the availability of the right file on the right place?
You can beleive or not I am trying to start with LVGL almost a months.
I tried to copy the files several times as you mentioned above.
I have reinstalled almost everything into the last few weeks. (Actually I am working with completelly reinstalled Arduino as well right now.)
the result is same.... some missing file... The result of the latest try is...:
_fatal error: ../../src/display/lv_display_private.h: No such file or directory
 #include "../../src/display/lv_display_private.h"_

Maybe a very newbie question.
What is the origin of the relative path ../../src/display/lv_display_private.h ?
The ino file or something different....

kisvegabor, 2024-04-25:
@proximacentaury 
I'm sorry that you had so many issues. I can assure you it should be really simple, but probably we need to update the docs to make it cleaner. 

Please attach 
- your ino file
- a longer the compiler log
- a screenshot about the Arduino `libraries` folder
- check the LVGL version in `lvgl.h`. 
- 
- Let's find where the issue is and update the docs for everyone's happiness :slightly_smiling_face:

proximacentaury, 2024-04-29:
Hello kisvgabor,

I sorry about long silence, but I needed to check many things.
I have Sunton CYD 2.4inch_ESP32-2432S024V version. I have fond some "original" factory documentation with well documented projects.
The documentation contains even the copy of libraries, and customized lvgl.h to LVGL and UserSetup.h to TFTespi.
Long story short I could compile and upload a demo to the board. (2 times from several other attempts)
So at least the board is working with certain setup together with Arduino IDE 2.3.2
The version of the lvgl.h is v8.3.0-dev

There is the plece where I am right now.
The next step is to convert everything to the latest lvgl...

Best Regards

kisvegabor, 2024-05-02:
Good news! 

Updating to v9 should't be too complicated. Please take a look at this [description](https://docs.lvgl.io/9.0/CHANGELOG.html#migration-guide).

kisvegabor, 2024-06-20:
As there were no activity here for a while I'm closing this issue.

closed 2024-06-20 by kisvegabor