lvgl/lvgl #3173: v8.3 discusion

kisvegabor, 2022-03-16:
Hi,

For v8.3 discussion I'm trying a new approach. I merged the not implemented features of v8.2 and the ROADMAP into the list below. This way it's easier to see where LVGL is heading and where the contributors can help to improve LVGL. 

We are not targeting to implement all the features below for v8.3, instead we will see what features will be implemented organically  and we can release v8.3 when we have enough new features. 

I'll evaluate [GitHub Project](https://github.com/lvgl/lvgl/projects?type=beta), as seemingly it can be used to track these tasks. 

As usual, if you have any ideas, remarks, comments or feedback feel free to comment here. 

### Drawing, layouts and styles
- [ ] SVG support: integrate an SVG render library
- [ ] More color formats: 18 or 24 bit, ARGB1555, etc
- [ ] Reconsider color format management for run time color format setting, and custom color format usage. 
- [ ] Different border and/or radius on sides and corners. #2800
- [ ] More grid features. E.g. `repeat(auto-fill, minmax( <value> px, 1fr))`
- [ ] Circle layout.  #2871
- [ ] Support `inherit` and `initial` style values. #3086
- [ ] Hover
- [ ] Color font support #2255 
- [x] Add ARM GPU support #3162

### Widget updates
- [ ] Support dot_begin and dot_middle long modes for labels
- [ ] Add new label alignment modes. #1656
- [ ] Text rotation and write on curved path

### Architecture
- [ ] Rework image decoders and image source handling #3161 
- [ ] Widgets to image buffer for caching
- [ ] New drivers architecture New driver architecture #2720
- [ ] Markup language Markup Language for LVGL #2428
- [ ]  Rework image transformation: #3142
- [ ] Finalize CPP binding. See [lv_binding_cpp](https://github.com/lvgl/lv_binding_cpp)
- [ ] Global cache. See #3116
  
### CI and testing
- [ ] Release scripts
- [ ] Writing unit tests for core widgets #2337
- [ ] Run static analyze (via coverity.io or something else)
- [ ] Platform independent benchmarking  
- [ ] CI test for flash/RAM usage #3127 

### To discuss
- Consider direct binary font format support
- Simplify groups. Discussion: https://forum.lvgl.io/t/lv-group-tabindex/2927/3.
- lv_mem_alloc_aligned(size, align)
- Speed up font decompression
- Support larger images: #1892
- Variable binding improvements like Redux?
- Functional programming support, pure view? See https://www.freecodecamp.org/news/the-revolution-of-pure-views-aed339db7da4/
- Scene support. See [here](https://github.com/lvgl/lvgl/issues/2790#issuecomment-965100911)
- Style components. See [here](https://github.com/lvgl/lvgl/issues/2790#issuecomment-965100911)

X-Ryl669, 2022-03-16:
Concerning image decoder change (& larger images), I've made some progress on lottie format decoding speed up. See [here](https://github.com/espressif/esp-box/issues/8). Basically, if we had some API for a memory allocator capability (like saying what kind of usage we expect to do on the allocated memory) in LVGL, then we can get a 3x speed up in rendering speed in lottie on ESP32. This would also a good improvement on the other platforms with multiple heap and/or DMA engine.

Another interesting improvement would be to have a unique color conversion code in LVGL code base (instead of having multiple instance in the different image decoder, one in rlottie, one in png, etc...) that's optimized for the CPU it's running on. Right now, some CPU architecture don't support unaligned access to memory when the access is not word sized so we have to fallback to memcpy which is very slow. If we had 32 bits access for all color conversion (that means processing pixels 3 by 3 in a typical 32 bits to 24 bits conversion), then we can use "fast" memory and get another increase in rendering speed.

C47D, 2022-03-16:
I would like to finish adding unit tests for existing widgets and clean up the first ones we did.

I'm also interested in adding new color format support.

kisvegabor, 2022-03-17:
@X-Ryl669 
> Basically, if we had some API for a memory allocator capability (like saying what kind of usage we expect to do on the allocated memory) in LVGL, then we can get a 3x speed up in rendering speed in lottie on ESP32. 

If I understood correctly we should to add
```c
lv_mem_alloc_caps(size, flags);
lv_mem_realloc_caps(ptr, size, flags);
lv_mem_free_caps(size, flags);
```
In `lv_conf.h` if `LV_MEM_CUSTOM` is enabled allow redefining the `*_caps` functions too.

> Another interesting improvement would be to have a unique color conversion code in LVGL code base (instead of having multiple instance in the different image decoder, one in rlottie, one in png, etc...) that's optimized for the CPU it's running on.

Something like this?
```
lv_color_convert_inplace(color_array, input_format);
lv_color_convert(color_array_in, color_array_ou, input_format);
```

kisvegabor, 2022-03-17:
@GorgonMeducer 
> Can we consider adding a low-level interface for accelerating tiling as described in https://github.com/lvgl/lvgl/issues/3174 ?

I'll comment in the opened issue.

@C47D 
> I would like to finish adding unit tests for existing widgets and then clean up the first ones we did.

Amazing, Thank you, Carlos! :+1:

kisvegabor, 2022-03-17:
I've create GitHub projects to track the plans. See https://github.com/lvgl/lvgl/projects

X-Ryl669, 2022-03-18:
> If I understood correctly we should to add
> 
> ```c
> lv_mem_alloc_caps(size, flags);
> lv_mem_realloc_caps(ptr, size, flags);
> lv_mem_free_caps(size, flags);
> ```

Yes, with caps/flags being a OR combination of:
```cpp
enum allocator_caps {
    Default = 0,   
    Fast      = 1,   /*!< Usually size limited but fast memory */
    Large    = 2,  /*!< Slow but large memory */
    WordAccess = 4, /*!< Memory block limited to word-sized access */
    DMA     = 8, /*!< DMA optimizable memory block */
    GPU     =16, /*!< GPU memory */
};
```
and a `lv_has_allocator_caps` function to query what are the supported caps for this platform.

> Something like this?
> 
> ```c
> lv_color_convert_inplace(color_array, input_format);
> lv_color_convert(color_array_in, color_array_out, input_format);
> ```

Not exactly (it's missing the image layout: width & stride & height and the destination format: likely RGB565 + Alpha 8 bits, which is different than lv_color_t). 
The latter signature style could be enough, since it's very easy to figure out that the output buffer is the input buffer.

kisvegabor, 2022-03-18:
> and a lv_has_allocator_caps function to query what are the supported caps for this platform.

I think it should be handled in `lv_mem_alloc_caps` to give something which is the closest to requested "config". E.g. in LVGL to get a buffer for image transformation:
```c
lv_color_t * buf = lv_mem_alloc_caps(1024, FAST | GPU);
```

But if this platforms has nothing to do with GPU `lv_mem_alloc_caps` just ignores that flag.

X-Ryl669, 2022-03-18:
Querying the caps would be a good idea to enable/disable some code. For example, if we know (like on ESP32) that there's a fast memory then it's worth allocating into this memory for an animation format, maybe even copying into this memory from the slow memory. If the caps isn't supported, then we'd skip the "optimization" here completely.

mysterywolf, 2022-03-20:
@GorgonMeducer Would you consider to add a documentation in here:
https://docs.lvgl.io/master/get-started/platforms/index.html#
https://github.com/lvgl/lvgl/tree/master/docs/get-started/platforms
to teach people how to use.

kisvegabor, 2022-03-21:
@X-Ryl669 I see your point. We should exactly define what we mean e.g. by "fast". Compared to what? How much faster, etc.

@mysterywolf @GorgonMeducer 
I was thinking about adding a "Draw engines" or "Renderers" menu too for SW, SDL and ARM GPU rendering.   What do you think?

X-Ryl669, 2022-03-21:
It's up to the driver implementer to figure out if it's worth it. For ESP32, I've [measured](https://github.com/espressif/esp-box/issues/8) that parsing the JSON file into rlottie internal structure is 3x faster when those structures are stored in DRAM/IRAM vs SPIRAM. However, since the space in DRAM/IRAM is very limited, we can't store all of them there. So it makes sense to copy them to DRAM (from SPIRAM) when the widget is animating. It'll work fine since you don't have many animation running at the same time usually.

Yet, another platform might not want this optimization (or does not have anything similar). In that case, we need to know if such logic make sense before trying to use it. For example, a GPU based optimization might instead want the *rendered* buffer to be allocated and/or copied to the GPU memory. 

So basically, pseudo code would be something like this:
```
  if (alloc_has_cap(FAST_MEM)) {
     if (!create_lottie_in_fast_mem()) {
        make_some_place_in_fast_mem();  // Free cache ?
        copy_lottie_struct_from_slow_mem_to_fast_mem();
    }
  }
  if (alloc_has_cap(GPU_MEM)) {
    rendering_buffer_gpu = alloc(GPU_MEM);
  } 
  rendering_buffer = rendering_buffer_gpu ? rendering_buffer_gpu : alloc(DEFAULT);
  render();
  if (alloc_has_cap(GPU_MEM) && !rendering_buffer_gpu) {
    copy_buffer_to_gpu(rendering_buffer);
  }
```

GorgonMeducer, 2022-03-21:
> Maybe adding sections [here](https://docs.lvgl.io/master/overview/drawing.html) would be better.
What do you think?

@kisvegabor I like your idea. 

@mysterywolf I have created a Chinese version and wants to collect sufficient feedbacks before writing a English version.

GorgonMeducer, 2022-03-22:
@kisvegabor maybe putting a brief and a reference for renderers in get-started and putting all relevant content in a dedicated place is better.

kisvegabor, 2022-03-24:
Makes sense. I added a new sections for Renderes and GPUs: https://docs.lvgl.io/master/overview/renderers/index.html

Let's add references from other pages if there are some content there.

GorgonMeducer, 2022-03-25:
@kisvegabor Thank you and I will write the arm2d part.

kisvegabor, 2022-05-28:
I'm planning to release v8.3 at the beginning of July and start v9 after that. Do you see any problem with it?

The most important (and still not merged) feature I'd like to have in v8.3 is the [new image decoder interface](https://github.com/lvgl/lvgl/pull/3294) of @X-Ryl669.

embeddedt, 2022-05-28:
Sounds good to me. I will try to complete the `inherit`/`initial` implementation between now and then.

mysterywolf, 2022-05-30:
Hi Please consider about merge this PR before v8.3 https://github.com/lvgl/lvgl/pull/3336 
BTW, would you consider to implement a function so that OS can take over lv_memset and lv_memcpy by using assembly lanuage with different CPU's archtectures to accelerate the speed?

kisvegabor, 2022-05-30:
> BTW, would you consider to implement a function so that OS can take over lv_memset and lv_memcpy by using assembly lanuage with different CPU's archtectures to accelerate the speed?

I'd rather do it in v9. IMO we should implement a more complete and conscious stdlib wrapper for memset, memcpy, memcmp, strcpy(_s), strcmp , malloc, free, realloc, (v)snprinf.

kisvegabor, 2022-07-04:
If there is no objection I'll release v8.3 on Wednesday. (July 4)

mysterywolf, 2022-07-05:
I find that the latest code of LVGL has very serious memory leak. I have tested on VS and QEMU simulators as well as on the STM32F469 Discovery board. All three platforms have the same behavior. RT-Thread found the memory has been exhausted and assert the system. @GorgonMeducer @kisvegabor 

Please confirm this problem. and if this is true, please defer the version release.

https://user-images.githubusercontent.com/34888354/177229666-9bb1419d-47d4-472f-b3f8-c059735e800a.mp4

embeddedt, 2022-07-05:
@mysterywolf I assume by latest you mean https://github.com/lvgl/lvgl/commit/23c8a91a112ed1c64636bb8376c71f6d1b52a719. There was a [fix](https://github.com/lvgl/lvgl/commit/a2bd96023f083777d2c34a690465caba2cd7b5ff) commited for a memory leak 4 days ago so I want to be sure you have that included.

mysterywolf, 2022-07-05:
No, I have include this commit. I confirm that I am using the latest code.
[image]

kisvegabor, 2022-07-05:
Thanks for pointing it out, Meco! 
I've fixed it here: 13c99fc4

Note that, due to some new rendering features `LV_COLOR_SCREEN_TRANSP` should be enabled to correctly handle blending.

GorgonMeducer, 2022-07-05:
@kisvegabor as memory leaking is a serious problem, should I update the cmsis-pack for the fix?

kisvegabor, 2022-07-05:
@GorgonMeducer 

IMO it's enough to update the cmsis-pack before the release tomorrow. What do you think?

GorgonMeducer, 2022-07-05:
Yes! I will do that.

Linjieqiang, 2022-07-06:
Can't wait to see the 8.3 release~!

kisvegabor, 2022-07-06:
>  Thank you, I have tested and there is no memory leaking any more. 

Great!

> By the way, do I need to enable the LV_COLOR_SCREEN_TRANSP when using the music player demo?

Yes, because of some blending with opacity.

mysterywolf, 2022-07-06:
> Yes, because of some blending with opacity.

@kisvegabor Thank you. I have adapted it for new version of RT-Thread BSPs' demo.

kisvegabor, 2022-07-06:
So, v8.3 is released. I wait until tomorrow to announce it on the Forum, LinkedIn, etc to see if any issues comes up.

tore-espressif, 2022-07-11:
> @tore-espressif I've just released v8.3 but it [failed to update](https://github.com/lvgl/lvgl/runs/7210982430?check_suite_focus=true) the Espressif package. Could you take a look at it?

I'm sorry about that  @kisvegabor , I can't see the problem right now, but I'll keep you updated :)

kisvegabor, 2022-11-14:
As v8.3 is released I close this issue.

closed 2022-11-14 by kisvegabor