lvgl/lvgl #3174: There is no low-level interface for accelerating Tiling (paving img in a given area)

GorgonMeducer, 2022-03-16:
### Introduce the problem
Usually, DMA based HW accelerators such as [DMA-350 from Arm](https://developer.arm.com/documentation/102482/0000/Overview/Key-features?lang=en) (and other similar 2D capable DMA) can accelerate tiling easily and hence offload CPU. But in LVGL, this work is done by [repeatedly calling `lv_draw_img()`](https://github.com/lvgl/lvgl/blob/16a1af38733ce8f06ef70615f7b55249d410221d/src/draw/sw/lv_draw_sw_rect.c#L388). 

[image]

### Examples and cases

### Suggested solution
It would be good to add a low-level interface for tiling with mirroring features, i.e. no-mirroring, x-mirroring, y-mirroring and xy-mirroring.

kisvegabor, 2022-03-18:
Tiling seems too high level for me at first look. E.g. a "tile" can have all the image effects, such as:
- rotation
- scale
- re-coloring
- chromakeying
- LVGL masking (e.g. to clip to a cricle)

It seems the API could be used only if a the image should be displayed as it is (i.e. no transformation).

What do you think?

GorgonMeducer, 2022-03-18:
The tiling here just means: repeatedly paving the source image to a target area of the display buffer with at most some mirroring feature.
No scale, no-recolouring.
Arm-2D support tiling with chromakeying and masking.

kisvegabor, 2022-03-24:
I see.  I thin kit covers most of the typical use cases. 

However, let's postpone it little until the transform API is implemented :slightly_smiling_face:

GorgonMeducer, 2022-03-25:
Yeah, we are also waiting for the transform API.

kisvegabor, 2023-10-31:
Closing in favor of #4733.

closed 2023-10-31 by kisvegabor