lvgl/lvgl #3175: How can I set the label to center or change the style

Kingwulin, 2022-03-17:
demo ：[simple-menu-with-custom-header](https://docs.lvgl.io/master/examples.html#simple-menu-with-custom-header)
what I want to do is the images below
I found that was not any action when I change the style of menu or label
would please tell me how to achive it 
Thanks
 
[image]
[image]

kisvegabor, 2022-03-17:
cc @HX2003

Kingwulin, 2022-03-17:
What I want to achieve is the page jump
for example：page1->page2,then press back btn, page2->page1
Is there any other component suitable for this function?

HX2003, 2022-03-18:
> What I want to achieve is the page jump for example：page1->page2,then press back btn, page2->page1 Is there any other component suitable for this function?

Is the question only about styling? How did you change the style?
If I understand correctly, if you just want the labels centered then you can apply `lv_obj_set_flex_align(obj, LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER);` to the menu_page and menu_cont.

closed 2022-04-10 by lvgl-bot