lvgl/lvgl #3177: use a block cursor in textarea widget,

bonzehan, 2022-03-17:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

First create text widget and set a Blue block cursor,then clear LV_STATE_FOCUSED state to hide cursor.
when clearing LV_STATE_FOCUSED state after the initial operation.  the block cursor still be there.
### To Reproduce
	
        ta1 = lv_textarea_create(Mid2MenuRect);
        lv_textarea_set_cursor_pos(ta1, 0);
        lv_textarea_set_text(ta1,"0.123456");
        lv_obj_align(ta1, LV_ALIGN_TOP_MID, x, y);
        lv_obj_set_style_bg_color(ta1, lv_palette_main(LV_PALETTE_BLUE), LV_PART_CURSOR);
        lv_obj_set_style_bg_opa(ta1, LV_OPA_COVER, LV_PART_CURSOR);
        lv_obj_add_flag(ta1, LV_STATE_FOCUSED);
        lv_obj_clear_state(ta1, LV_STATE_FOCUSED); //   use focused  hide cursor

### Expected behavior

Using the default cursor is no problem.The cursor shadow does not appear at the end.
Is there something wrong with how I set the cursor?
### Screenshots or video

[image]
### Suggested solution

I checked that the default cursor properties are border like properties, the function lv_textarea.c --> draw_cursor(), cursor.show 1: hide the cursor(It's actually erasing the border), 0: show it(It's do anything). I don't know if I'm getting this wrong.
so I modified  the function lv_textarea.c --> draw_cursor()
"if(ta->cursor.show == 1) return;" instead "if(ta->cursor.show == 0) return;"
I test it, Both the default cursor and the block cursor seem to work. so.....I don't know if I got it right,Is there a better solution!

kisvegabor, 2022-03-24:
Hi,

I've looked into it and it was tricky issue.  So this what happened:
- In default state the cursor has `bg_opa = 0` as no style is added to it in default state
- `lv_theme_default` add a border in focused state
- Your code added `bg_opa=LV_OPA_COVER` in default state, therefore the cursor became visible in default state too

You can simple solve doing it this way:
```c
  lv_obj_set_style_bg_opa(ta1, LV_OPA_COVER, LV_PART_CURSOR | LV_STATE_FOCUSED);
```

bonzehan, 2022-03-25:
I'm sorry to have caused you trouble.  Thank you for your reply!
I understand, **I need to set the cursor type in focus state!!**
That's the perfect solution.
Thanks again!!!!!!!

kisvegabor, 2022-03-25:
No problem :slightly_smiling_face:

closed 2022-03-25 by kisvegabor