lvgl/lvgl #3186: gif placed in inactive screen consumes CPU

wangyz1997, 2022-03-20:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Hi, I'm using lvgl 8.2 release on ESP32-C3. My project has two screens and I put one gif widget on screen A. After I active screen B, which consits of only labels, I can see a noticeable CPU usage. When I delete the gif widget on screen A, the CPU usage drops to 0%.

### To Reproduce

#### Case 1:
```c
lv_obj_t *screen_A = lv_obj_create(NULL);
lv_obj_t *screen_B = lv_obj_create(NULL);

lv_obj_t *image_gif = lv_gif_create(screen_A);
lv_gif_set_src(image_gif, &image_gif_desc);
lv_obj_align(image_gif, LV_ALIGN_CENTER, 0, 0);

lv_obj_t *label = lv_label_create(screen_B);
lv_obj_set_width(label, 200);
lv_obj_align(label, LV_ALIGN_CENTER, 0, 0);
lv_obj_set_style_text_align(label, LV_TEXT_ALIGN_CENTER, 0);
lv_label_set_text(label, "text on screen B while gif on screen A");

lv_scr_load(screen_B);
```
The CPU usage is about 7~8%, which can be seen below.

#### Case 2:
```c
lv_obj_t *screen_A = lv_obj_create(NULL);
lv_obj_t *screen_B = lv_obj_create(NULL);

// lv_obj_t *image_gif = lv_gif_create(screen_A);
// lv_gif_set_src(image_gif, &image_gif_desc);
// lv_obj_align(image_gif, LV_ALIGN_CENTER, 0, 0);

lv_obj_t *label = lv_label_create(screen_B);
lv_obj_set_width(label, 200);
lv_obj_align(label, LV_ALIGN_CENTER, 0, 0);
lv_obj_set_style_text_align(label, LV_TEXT_ALIGN_CENTER, 0);
lv_label_set_text(label, "text on screen B while gif not on screen A");

lv_scr_load(screen_B);
```
The CPU usage drops to 0%.

### Expected behavior

When the screen with gif widget is inactive, is should not consume CPU.

### Screenshots or video

#### Case 1:

[image]

#### Case 2:

[image]

kisvegabor, 2022-03-21:
Hi,

We could detect if the gif widget is visible or not but I'm not sure it'd really the best to stop its rendering when not visible. Some user might count on that the time passes and the gif will be on a different frame when it's screen is shown again.

Having an `lv_gif_pause()` and `lv_gif_resume()` function seems more flexible to me. 

---

@X-Ryl669 
The new image API could be used e.g. for this purpose.

X-Ryl669, 2022-03-21:
In #2904 it was. The timer function in lv_img deal with the control. You can also rewind the gif (but not play in reverse mode, unfortunately).

wangyz1997, 2022-03-21:
Thanks for your reply. It would be great to have an attribute to control whether to render the gif in the background.

kisvegabor, 2022-05-28:
> GIF is not seekable, so unfortunately, we still need to decode each frame to find out the next frame, so some CPU will still be consumed, even if not rendered (so a lot less CPU will be used).

This is how it works out of the box?

X-Ryl669, 2022-05-30:
Yes. Gif files are made of chunks, there's no central header saying how many frames there is in the file nor their positions in the file to decode them. And since GIF also allow a (transparent) frame to partially overlap a previous frame, you can't decode it from any frame index (unless we write some very clever algorithm that would store each frame position in a table + ensure all pixels are overwritten in each frame and so on). 

So right now, I've added a feature to count the frames (but don't decode them) while decoding the file at first so I can populate the `frames_count` member, IIRC but we still need the timer's callback to process the next frame even if not rendered.

X-Ryl669, 2022-05-30:
> I mean in the new gif decoder, if the gif is not visible now, it's nor rendered, right?

Right.

> But what will happen to the timer period?

The timer is only started if the picture is decoded so only if visible. So until the picture is visible, it's not rendered and the timer is not fired. When the picture becomes visible, it'll be drawn => decoded => timer is started.
If the picture is then hidden, the timer isn't stopped, so its current frame index will continue increasing with the last known frame delay, but decoding will not happen (IIUC, since the timer will call invalidate, but no draw call should be happen for an hidden object, right ?). So it'll just loop like that continuously.

If the user wants to pause the animation, she has to explicitly called the `set_play_mode(PAUSE)` function. 

I think it's the best behavior/compromise between CPU usage and convenience, since if a picture isn't visible, we don't pay the decoding cost. But we don't follow the "true" frame delay anymore either since we don't decode, so we make a small timing error here.

Maybe we can add a check in the timer's callback to ensure if the object is visible and skip this iteration (return) of the timer if it isn't. 

I don't know what's best here, the current behavior means if you move some opaque object over a GIF, the animation will continue without impact (even when the opaque object is above the GIF). But if we skip the iteration, it'll jerk when being visible again.

wangyz1997, 2022-05-30:
Hi, @X-Ryl669 , thanks for your work on the new gif decoder. I think it would be better to add an attribute to the gif image object, which controls decoded it or not when it is invisiable. This is useful since user can configure once to save cpu usage during the object's lifecycle.
And, I consider to leave it as it is when the attribute is not set, which means the gif continues to render when invisiable.

kisvegabor, 2022-05-30:
> IIUC, since the timer will call invalidate, but no draw call should be happen for an hidden object, right ?

Yes, this is how it works. 

> Maybe we can add a check in the timer's callback to ensure if the object is visible and skip this iteration (return) of the timer if it isn't.

I think it should return from the timer `if(!visible && (!seekable || VFR))`.    
- If VFR: we don't know the frame delay so better not doing something stupid.
- If not seekable: it was pointless the increment frame counter as we can't render that frame directly later. 

I can hardly imagine situations where it's important to keep invisible gifs in sync. I consulted with our designer and he has the same opinion. We can think of it as a limitation. This how it works in LVGL. So IMO it fine to keep invisible images "inactive".

X-Ryl669, 2022-05-30:
Ok for me. So how do I get the "visible" flag here (ideally, without requesting the cover check computation again ?)

kisvegabor, 2022-05-31:
`lv_obj_is_visible()` is fine. It just checks the bounding box of the object with the parent and grandparents. The final cover check during rendering might find that this widget is not visible but I think this function is good enough for now.

HaoboGu, 2023-03-10:
I have the same problem, I loaded a gif on scr1 and made another screen scr2 active. gif in scr1 consumes a lot of CPU. Is there any way to prevent the extra CPU consuming?

kisvegabor, 2023-03-11:
In `lv_gif.c` you can add this modification: 
```c
static void next_frame_task_cb(lv_timer_t * t)
{
    lv_obj_t * obj = t->user_data;
    
    if(lv_obj_is_visible(obj) == false) return;   //ADD THIS

   ...
```

HaoboGu, 2023-03-13:
> In `lv_gif.c` you can add this modification:
> 
> ```c
> static void next_frame_task_cb(lv_timer_t * t)
> {
>     lv_obj_t * obj = t->user_data;
>     
>     if(lv_obj_is_visible(obj) == false) return;   //ADD THIS
> 
>    ...
> ```

It works, thanks very much!

kisvegabor, 2024-09-05:
After taking a deeper look at this I realized that probably it should be handled differently and the gif decoder should be part of LVGL's image caching mechanism too. 

Closing it for now. The above-proposed workaround can be used if needed.

closed 2024-09-05 by kisvegabor