lvgl/lvgl #3189: Extern error about const lv_label_class  LVGL Version 8.2.0

Edwin-Dijk, 2022-03-21:
Hi,
I am trying to implement C++ code for my project using LVGL. I changed my projects main to a cpp file and added the extern "C" to the app_main function.
After that I made my own .hpp file for all of my GUI related stuff and included that file in my main.cpp. I however get the following error:
`In file included from ../components/lvgl/lvgl.h:49,
                 from ../main/gui.hpp:28,
                 from ../main/main.cpp:24:
../components/lvgl/src/widgets/lv_label.h:79:1: error: invalid use of 'extern' in linkage specification
 extern const lv_obj_class_t lv_label_class;
 ^~~~~~`
The error will go away when I comment line 79 ( extern const lv_obj_class_t lv_label_class; ) in the label.h file but I think this is not the right way to do it.
Can someone assist me in finding the correct solution to this problem.
LVGL version used: 8.2.0

TLDR:
Changing my main.c to main.cpp
Added a GUI.hpp file and included this in main
Getting an error about an extern const lv_label_class

kisvegabor, 2022-03-21:
Hi,

Is this issue only for the label?

Other widgets have the very similar pattern. E.g. `extern const lv_obj_class_t lv_btn_class;` in `lv_btn.c`.

Edwin-Dijk, 2022-03-21:
Hi there,
Thank you for your reply.
Yes weirdly enough only the label gives me this error, while I do use multiple buttons and sliders in my design. I have got all the files within one project so technically I could remove the extern statement but I wanted to bring this to your attention nonetheless.

embeddedt, 2022-04-06:
Is it possible that your LVGL copy is corrupted? I would expect every widget to give the same error, and `lv_label.h` is included after several other widgets' headers.

Edwin-Dijk, 2022-04-07:
I have tried deleting the entire lvgl directory and redownloading 8.2.0 and inserting that into my project. Unfortunately that didn't solve anything. If there could be another way the lvgl copy was corrupt could you please let me know? :)

closed 2022-04-29 by lvgl-bot

embeddedt, 2022-04-29:
@Edwin-Dijk Unfortunately I don't have any other ideas about what the problem could be. If it is possible to reproduce this in a fresh, clean project that can be shared, we could take a look.