lvgl/lvgl #3191: LV_USE_GPU_NXP_PXP_AUTO_INIT not working in 8.2.0 (was working in 8.0.2)

andrewrt, 2022-03-22:
### Perform all steps below and tick them with [x]
- [x ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
As per: https://docs.lvgl.io/8.2/get-started/nxp.html , 
if you want to use the NXP PXP, you can just set `LV_USE_GPU_NXP_PXP_AUTO_INIT` to 1 along with `LV_USE_GPU_NXP_PXP` and you should be good to go without your own setup. 
This was working for me in LVGL 8.0.2, but when updating to 8.2.0, it fails to compile saying the following symbol isn't defined: 
`lv_draw_nxp_pxp_init`

### To Reproduce
Pull in LVGL 8.2.0 from git repo, then compile existing previously functioning code that used NXP PXP as per:
```c
/*Use NXP's PXP GPU iMX RTxxx platforms*/
#define LV_USE_GPU_NXP_PXP      1
#if LV_USE_GPU_NXP_PXP
#define LV_USE_GPU_NXP_PXP_AUTO_INIT 1
#endif

```

### Expected behavior

Code compiles

kisvegabor, 2022-03-22:
Hi,

The PXP GPU needs to be reimplemented due to some architectural changes in LVGL. 
However, @nicusorcitu from NXP is already working on it in  #3052.

As the fix is on it's way, I close this issue.

closed 2022-03-22 by kisvegabor

nicusorcitu, 2022-03-29:
This issue was already fixed once with https://github.com/lvgl/lvgl/issues/3052.

kisvegabor, 2022-03-29:
> Is this architectural change is for 8.0.2 -> 8.1.0 or 8.1.0 -> 8.2.0?

For 8.1.0->8.2.0