lvgl/lvgl #3194: JavaScript simulator cancel main loop (Emscripten)

leidto, 2022-03-22:
### Introduce the problem
I'd like to change canvas at runtime.

### Examples and cases
`Module.canvas = document.getElementById('canvas2')`
`setTimeout(() => window.startRunning(), 0)`

The code above wrote in the console:
_Aborted(Assertion failed: emscripten_set_main_loop: there can only be one main loop function at once: call emscripten_cancel_main_loop to cancel the previous one before setting a new one with different parameters.)_

I haven't found the **emscripten_set_main_loop()** function on the window object yet. Where should I search or call it?

kisvegabor, 2022-03-23:
@embeddedt know more about it, but it might be related to the upstream [micropython](https://github.com/micropython/micropython). It might be worth asking the question there too.

embeddedt, 2022-03-23:
Hmm, the upstream port is pretty basic and mostly just implements the interpreter in the browser. I don't think they use SDL.

embeddedt, 2022-03-24:
I thought I had answered your question already but apparently not. I believe `emscripten_set_main_loop` is called within the Emscripten SDL implementation. For your use case, I'd suggest that you use an iframe and reload it whenever you want to change canvases, instead of calling `startRunning` again, as it's not very easy to restart Emscripten directly.

leidto, 2022-04-05:
Thank you. Using an iframe is not an option in my case. I'll find another way.