lvgl/lvgl #3208: Can lv_freetype load multiple fonts at the same time?

weixiongmei, 2022-03-26:
### Introduce the problem
Trying to use lv_ft_font_init to load the same font file with different sizes via freetype2, but it doesn't work as expected, the widgets on the screen is not drawing properly if I loaded the font with different size more than one. thanks~~

kisvegabor, 2022-03-29:
cc @guoweilkd

guoweilkd, 2022-03-29:
I have been using it like this and found no problem. Post some pseudo-code ?,

weixiongmei, 2022-04-03:
@guoweilkd , Hi, found the issue, there's a bug in my drawing code. It's fixed now. But one more question, how's the performance of the freetype? Every time when I tried to update the text of a widget that's using freetype font, the usage of the CPU always goes up to 80%-100%, it's supper slow, seems like the cache doesn't work at all... thanks~~

guoweilkd, 2022-04-06:
If you use cache, it is recommended not to use the `FT_FONT_STYLE_BOLD`  macro when create Freetype font. Bold font does not use caching, which severely affects performance.

weixiongmei, 2022-04-06:
@guoweilkd , I didn't use bold style, i just used normal style, lets say you're updating a label with text format hh:mm:ss every 30ms, how many fps you can get on the MCU? I can only get 1 FPS on the ESP32-S3 no matter what size of the font, if you used the pre-compiled font(converted with the LVGL font converter tool), I can get 30 FPS. Thanks~~

guoweilkd, 2022-04-06:
Freetype fonts definitely take longer to render than precompiled fonts.
You can modify these macros(`LV_FREETYPE_CACHE_SIZE `  `LV_FREETYPE_CACHE_FT_FACES `  `LV_FREETYPE_CACHE_FT_SIZES `) to increase the cache size and reduce rendering time

weixiongmei, 2022-04-06:
@guoweilkd , I tried to increased the cache size, but seems like has no effects at all, don't even know the caching is working or not, but the performance is extremly slow, I can only get 1 FPS, or 2-3 fps if I only rendering 1 number "0-9" at the font size 12.....

FASTSHIFT, 2022-04-11:
@weixiongmei 
Is your font cache in an external PSRAM? Have you evaluated the actual read and write speed of PSRAM?

weixiongmei, 2022-04-17:
@FASTSHIFT I think there's something wrong with the cache on the ESP32-S3. when I set the cache size >= 0, it will freeze when performing a cache_lookup, if set the cache size to -1, it seems like working fine, but very slow. How to determine the cache is alloc on the internal ram or external ram? is there a way to manually alloc the cache? thanks~~

FASTSHIFT, 2022-04-18:
> @FASTSHIFT I think there's something wrong with the cache on the ESP32-S3. when I set the cache size >= 0, it will freeze when performing a cache_lookup, if set the cache size to -1, it seems like [...]

The heap used by FreeType can be specified in `include/freetype/config/ftstdlib.h`.

closed 2022-05-11 by lvgl-bot