lvgl/lvgl #3211: Prioritization of GESTURE and CLICKED events

FASTSHIFT, 2022-03-28:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
The gesture event is registered at the root node of the screen, and the click event is registered on the button. When the screen slides, the click event of the button is also triggered.

### To Reproduce
```c
static void scr_event_cb(lv_event_t* event)
{
    LV_LOG_USER("LV_EVENT_GESTURE");
}

static void obj_event_cb(lv_event_t* event)
{
    LV_LOG_USER("LV_EVENT_CLICKED");
}

static void swipe_test(void)
{
    lv_obj_t* obj = lv_btn_create(lv_scr_act());
    lv_obj_set_width(obj, 100);
    lv_obj_center(obj);
    lv_obj_add_event_cb(obj, obj_event_cb, LV_EVENT_CLICKED, NULL);

    lv_obj_add_event_cb(lv_scr_act(), scr_event_cb, LV_EVENT_GESTURE, NULL);
}
```

### Expected behavior
Is it possible to prevent the click event when the gesture event occurs? Or is there any way to avoid it?

### Screenshots or video

https://user-images.githubusercontent.com/26767803/160332393-88442a26-5fcd-4eb5-98a2-24d47c7d4279.mp4

FASTSHIFT, 2022-03-28:
I tried to clear the `LV_OBJ_FLAG_PRESS_LOCK` flag, but the click still triggers.

https://user-images.githubusercontent.com/26767803/160333823-e312a052-fa70-4698-ba3d-301b3655f58f.mp4

kisvegabor, 2022-03-30:
It makes sense. 
We an use `proc->types.pointer.gesture_sent` to skip sending some events if the a gesture was fired.

FASTSHIFT, 2022-03-31:
> gesture_sent

Can this variable be accessed and modified at the application layer? How should it be modified?

kisvegabor, 2022-04-04:
I meant adding `if`s in LVGL itself (`lv_indev.c`) to not fire `PRESSING`, `CLICKED` events, but I've just realized it's not a good idea as it's not sure that the use was interested in the gestures.

I haven't tested it but `lv_indev_get_gesture_dir(lv_indev_act()) == LV_DIR_NONE` should indicate that there was no gesture.

FASTSHIFT, 2022-04-05:
> I meant adding `if`s in LVGL itself (`lv_indev.c`) to not fire `PRESSING`, `CLICKED` events, but I've just realized it's not a good idea as it's not sure that the use was interested in the [...]

Is it possible to add `FLAG` to the user to choose?

kisvegabor, 2022-04-11:
The best would be if we seen that the gesture was "consumed" and if so click, press etc weren't send. But we don't know it s the user should the tell to the indev handler to skip these events.

What about calling `lv_indev_wait_release(lv_indev_act())` from the event handler of the gesture?

FASTSHIFT, 2022-04-19:
> The best would be if we seen that the gesture was "consumed" and if so click, press etc weren't send. But we don't know it s the user should the tell to the indev handler to skip these events.
> 
> What about calling `lv_indev_wait_release(lv_indev_act())` from the event handler of the gesture?

I tested it and it works as expected.

kisvegabor, 2022-04-19:
Great! If you find it a suitable solution, feel free to close this issue. 

If you agree it's ok this way, I'll add not about this method here: https://docs.lvgl.io/master/overview/indev.html#gestures

FASTSHIFT, 2022-04-19:
> Great! If you find it a suitable solution, feel free to close this issue.
> 
> If you agree it's ok this way, I'll add a note about this method here: https://docs.lvgl.io/master/overview/indev.html#gestures

OK

kisvegabor, 2022-04-19:
The docs is updated too.

FASTSHIFT, 2022-05-04:
@kisvegabor 
The widget's state doesn't seem to automatically transition correctly.
lv_btn does not properly exit `LV_STATE_PRESSED` state.

https://user-images.githubusercontent.com/26767803/166685179-95622ec6-bcc0-4348-b0ff-d77776aee590.mp4

```c
static void event_cb(lv_event_t* event)
{
  lv_event_code_t code = lv_event_get_code(event);
  if(code == LV_EVENT_GESTURE)
  {
    LV_LOG_USER("LV_EVENT_GESTURE");
    lv_indev_wait_release(lv_indev_get_act());
  }
  else if(code == LV_EVENT_CLICKED)
  {
    LV_LOG_USER("LV_EVENT_CLICKED");
  }
}

void gesture_test(void)
{
  lv_obj_t* btn = lv_btn_create(lv_scr_act());
  lv_obj_clear_flag(btn, LV_OBJ_FLAG_GESTURE_BUBBLE);
  lv_obj_set_size(btn, 200, 200);
  lv_obj_center(btn);
  lv_obj_add_event_cb(btn, event_cb, LV_EVENT_ALL, NULL);
}
```

kisvegabor, 2022-05-06:
I'm thinking about solving it by sending `LV_EVENT_PRESS_LOST` in `lv_indev_wait_release()` for the active obejct. What do you think?

FASTSHIFT, 2022-05-09:
> I'm thinking about solving it by sending `LV_EVENT_PRESS_LOST` in `lv_indev_wait_release()` for the active object. What do you think?

Will `lv_indev_wait_release()` emit `LV_EVENT_PRESS_LOST` event?

kisvegabor, 2022-05-09:
Yes, I meant that.

kisvegabor, 2022-05-10:
Added, please test it. :slightly_smiling_face:

FASTSHIFT, 2022-05-12:
It works fine, thank you.

closed 2022-05-12 by FASTSHIFT

kisvegabor, 2022-05-12:
Great!

mmar22, 2023-04-18:
Hi boys, nice debaye, but not aplyed in Squareline. I create on screen event clicked and gesture UP.
On gesture always is fired clicked too. ???

kisvegabor, 2023-04-18:
Finally we have decided that you can call `lv_indev_wait_relelase(lv_indev_act())` in the gesture.

mmar22, 2023-04-18:
In Squareline I call nothing. Events is added without call. Your generator need add it in ui.c or perfect is place some checkbox , that define on event wait...

From: Gabor Kiss-Vamosi 
Sent: Tuesday, April 18, 2023 12:48 PM
To: lvgl/lvgl 
Cc: mmar22 ; Comment 
Subject: Re: [lvgl/lvgl] Prioritization of GESTURE and CLICKED events (Issue #3211)

Finally we have decided that you can call lv_indev_wait_relelase(lv_indev_act()) in the gesture.

kisvegabor, 2023-04-19:
Okay, we can add it in SLS.