lvgl/lvgl #3213: lvgl can Gaussian blur be supported?

jianjunxia, 2022-03-28:
Can lvgl support Gaussian blur? In this way, we can better deal with the characteristics of some pictures

kisvegabor, 2022-03-28:
For performance reasons now you can perform only box-blur. However. in practice performing box-blur twice looks quite similar to Gaussian blur.

---
We use GitHub Issues for development-related discussions.

Please ask your question in the [Forum](https://forum.lvgl.io) to make it visible for a larger community. You can simply log in with your GitHub account.

closed 2022-03-28 by kisvegabor

jianjunxia, 2022-03-28:
> For performance reasons now you can perform only box-blur. However. in practice performing box-blur twice looks quite similar to Gaussian blur.
>
> We use GitHub Issues for development-related [...]

thanks

jianjunxia, 2022-03-28:
> 

why you say "For performance reasons now you can perform only box-blur"?
and do you have a useful third-party library or example? i want to test
thanks！

kisvegabor, 2022-03-31:
> why you say "For performance reasons now you can perform only box-blur"?

Because box-blur is much simpler, thus faster.

> and do you have a useful third-party library or example? 

Please open a Forum topic, link it here, and I'll answer there.

jianjunxia, 2022-04-01:
> > why you say "For performance reasons now you can perform only box-blur"?
> 
> Because box-blur is much simpler, thus faster.
> 
> > and do you have a useful third-party library or example?
> 
> Please open a Forum topic, link it here, and I'll answer there.

it's here https://forum.lvgl.io/t/how-to-implement-this-feature-in-lvgl-with-background-filter-blur-60px-in-css/8591?u=jianjunxia