lvgl/lvgl #3214: Large option list in roller has incorrect behavior

jbarrena, 2022-03-28:
Hi,

I am using two rollers to input time. The minutes roller has 60 options and hour 12 options. The minutes roller is printed incorrectly, like if it had some spaces between options.

The code below reproduces tyhe bug in simulator:
```c
lv_roller_set_options(me->rlrHour,"01\n02\n03\n04\n05\n06\n07\n08\n09\n10\n11\n12",
        	LV_ROLLER_MODE_INFINITE);
    /*else
    	lv_roller_set_options(me->rlrHour,"01\n02\n03\n04\n05\n06\n07\n08\n09\n10\n11\n12\n13\n14\n15\n16\n17\n18\n19\n20\n21\n22\n23\n24",
        	LV_ROLLER_MODE_NORMAL);*/
    lv_roller_set_visible_row_count(me->rlrHour, 1);
    lv_obj_set_height(me->rlrHour, 36); // Force height because it does not manage well font height
    lv_roller_set_selected(me->rlrHour, 0, LV_ANIM_OFF); // Select by default first item
    lv_group_add_obj(me->grpText, me->rlrHour);
    lv_obj_add_event_cb(me->rlrHour, onTimeEvent, LV_EVENT_KEY, me);
```
This is how it prints the options in the rollers. Expected is the roller in the left, and the right one represents the incorrect behavior.
[image]

Thanks for your help.

Julen.

embeddedt, 2022-03-28:
Sometimes enabling [`LV_USE_LARGE_COORD`](https://github.com/lvgl/lvgl/blob/9a0a74813d86c897859a33f49d477352ebf5823b/lv_conf_template.h#L323) fixes this, although I'm surprised you'd see the issue with only 60 options.

jbarrena, 2022-03-30:
Thank you for your help! It works now.

As I described, with 60 point in LV_ROLLER_MODE_INFINITE it didn't work properly.

Thanks again!

Julen.

closed 2022-03-30 by jbarrena

SpaceTeddy, 2022-08-17:
same here.
with ESP32 and roller >29 items output is not like it should.

sers,
Chris

kisvegabor, 2022-08-19:
I've pushed an update which automatically calculates the required number of extra "pages" in infinite mode. It targets 1000 px height but adds 3 pages at least .

commit 878e3d33 (2022-08-19): feat(roller): remove LV_ROLLER_INF_PAGES and calculate it dynamically

fixes #3214