lvgl/lvgl #3219: "Scene" support

kisvegabor, 2022-03-29:
### Introduce the problem
Imagine there is a popup showing different states of a process:
- overview + start button
- waiting for the system to start the process (API call pushes to the next state)
- progress indication (arc + label, updated via API)
- error screen
- ready screen

At this moment it's very cumbersome to implement:
1. Create a popup
2. On "Start" click event clean the content, and show waiting state
3. On API call go to progress indication
4. On API call update the arc
5. At any stage go to error screen if requested, and clean up
6. Go to ready screen when requested and clean up

So popup is adjusted from many unrelated sources which is hard to manage and maintain and quite error prone. Besides the popup's code is scattered to many places although this popup is still one "component", it just have different "states" (or rather "scenes"). . 

### Suggested solution
I already mentioned [here](https://github.com/lvgl/lvgl/issues/2790#issuecomment-965100911) that I suggest introducing **Scenes**.  It's a very simple but very powerful thing. 

Each widget has an `int scene_active` field and there is an `lv_obj_set_scene(obj, scene_id)` function which sends `LV_EVENT_SCENE_UNLOAD`, sets the `obj->scene_active=scene_id` and sends `LV_EVENT_SCENE_LOAD` to the widget. 

The above mentioned popup can be handled very easily with this concept: all stages are scenes and on `LV_EVENT_SCENE_LOAD` the the required widgets are created and  on `LV_EVENT_SCENE_UNLOAD` the previous content is removed. Everything is kept in one place and there is nice API to select the content to show (e.g. `lv_obj_set_scene(popup, POPUP_SCENE_ERROR)`).

Some other use cases:
- There is a screen with "error" and "normal" scenes: on error, some styles are changed, and an error message is displayed. On "normal" these are reverted.
- A list has 2 scenes: hidden, visible. On scene load instead of hiding/unhiding, any complex animation can be started.
- There is a button component which can have dark, light, blue, flat designs. Each can be scene and the user could simply select between the designs.

---
Some alternative ideas about the implementation:

It can be implemented outside of the LVGL core like this:
```c
void lv_obj_add_scene_handler(lv_obj_t * obj, lv_event_cb_t event_cb)
{
   lv_scene_dsc_t * dsc = lv_mem_alloc(sizeof(lv_scene_dsc_t));
   dsc->scene_act = 0;
   lv_obj_add_event_cb(obj, scene_handler_event_cb, LV_EVENT_ALL, dsc); //Just to free dsc
   lv_obj_add_event_cb(obj, event_cb, LV_EVENT_ALL, dsc);
}

static void scene_handler_event_cb(lv_event_t * e)
{
  lv_event_code_t code = lv_event_get_code(e);
  lv_scene_dsc_t * dsc = lv_event_get_user_date(e);
  if(code == LV_EVENT_DELETED) {
    lv_mem_free(dsc);
  }
}

void lv_obj_set_scene(lv_obj_t * obj, uint32_t scene_id)
{
  lv_scene_dsc_t * dsc = lv_obj_get_event_user_data(obj, scene_handler_event_cb);
  dsc->scene_next = scene_id;
  lv_event_send(obj, LV_EVENT_SCENE_UNLOAD, NULL);
  dsc->scene_act = scene_id;
  lv_event_send(obj, LV_EVENT_SCENE_LOAD, NULL);
}

uint32_t lv_obj_get_scene(lv_obj_t * obj)
{
  lv_scene_dsc_t * dsc = lv_obj_get_event_user_data(obj, scene_handler_event_cb);
  return dsc->scene_act;
}
```
This way this feature can go to a separate file and can be enabled/disabled in `lv_conf.h` in a clean way. 

What do you think?

FASTSHIFT, 2022-03-31:
I implemented a similar [page manager](https://github.com/FASTSHIFT/X-TRACK/tree/main/Software/X-Track/USER/App/Utils/PageManager), hope it helps you :)

kisvegabor, 2022-04-04:
Thanks for sharing! 
It seems similar to [lv_fragment](https://docs.lvgl.io/master/others/fragment.html), a feature added by @mariotaku  a few month ago.

The main difference I see between page_manager/fragment vs scene, is that scene is more abstract and lower level because it
- can act on a single widget too (e.g. a slider in error or normal scene)
- can act on the current content (e.g. change style or size or text, value, etc of some widgets without recreating them)
- can act on "foreign" widgets too. (e.g on incorrect password put the "NEXT" button the "inactive" scene)

However, push/pop/stack etc is out of the scope of this feature. It can be built on top of it if required. 
At least this is how I imagined it.  :slightly_smiling_face:

kisvegabor, 2022-04-06:
> IIUC, the screen concept is the opposite, it's like a dynamic filter on everything, so you don't pay for creation/destruction between pages, but you pay for more memory usage since you've allocated everything upfront.

It's up to you. If you want to re-implement something like fragment pages with scenes, on `SCENE_UNLOAD/LOAD` you can either 
- clean the parent and create new content 
- hide the page, and unhide an other
- delete some widget, create others and  adjust the remaining ones (i.e. rearrange the page)  

But there is no stack to "pop back" to the 3rd earlier state. If you need stack too, probably you need the fragment widget or build the history yourself.

kisvegabor, 2022-05-07:
Not stale

xlbao123, 2023-01-31:
It seems "scene" can be used to display one screen's different state. In my experience it's really useful, especially since markup language is supported later to describe the ui interface.

kisvegabor, 2023-10-26:
Closing as it can be realized by lv_observer.

closed 2023-10-26 by kisvegabor