lvgl/lvgl #3239: lv_draw_sw_letter may use NULL g.resolved_font

johado, 2022-03-30:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

I'm using v8.2.0.
When using a truetype font and letters not supported by the font, the application will hang due to 
LV_ASSERT_NULL(font_p); in lv_font.c:lv_font_get_glyph_bitmap()
It seems the freetype implementation returns "found" but sets is_placeholder 
lv_font_get_glyph() returns true, without setting dsc_out->resolved_font

### To Reproduce
```c
void font_bug(void)
{
  static lv_ft_info_t info;
  /* From lvgl/examples/libs/freetype/ */
  info.name = "./arial.ttf";
  info.weight = 28;
  info.style = FT_FONT_STYLE_NORMAL;
  info.mem = NULL;
  if(!lv_ft_font_init(&info)) {
      LV_LOG_ERROR("create failed.");
  }
  lv_obj_t * scr = lv_scr_act();
  lv_obj_t *label = lv_label_create(scr);
  lv_obj_set_style_text_font(label, info.font, LV_STATE_DEFAULT);
  lv_obj_set_style_bg_color(label, lv_color_hex(0x030303), LV_STATE_DEFAULT);
  lv_label_set_text_fmt(label, "test: %s", LV_SYMBOL_UP);
}
```

```

#0  lv_font_get_glyph_bitmap (font_p=0x0, letter=61559) at lvgl-native-8.2.0/src/font/lv_font.c:51
#1  0x000055555559da17 in lv_draw_sw_letter (draw_ctx=0x5555558d4e50, dsc=0x7fffffffd4b0, pos_p=<optimized out>, letter=61559) at lvgl-native-8.2.0/src/draw/sw/lv_draw_sw_letter.c:126
#2  0x000055555556b327 in lv_draw_letter (letter=<optimized out>, pos_p=0x7fffffffd48c, dsc=0x7fffffffd4b0, draw_ctx=0x5555558d4e50) at lvgl-native-8.2.0/src/draw/lv_draw_label.c:367
#3  lv_draw_label (draw_ctx=draw_ctx@entry=0x5555558d4e50, dsc=dsc@entry=0x7fffffffd5b0, coords=coords@entry=0x7fffffffd5a0, txt=0x5555558ffd70 "test: ", hint=hint@entry=0x0) at lvgl-native-8.2.0/src/draw/lv_draw_label.c:302
#4  0x000055555559259e in draw_main (e=<optimized out>) at lvgl-native-8.2.0/src/widgets/lv_label.c:859
#5  lv_label_event (class_p=<optimized out>, e=<optimized out>) at lvgl-native-8.2.0/src/widgets/lv_label.c:786
#6  lv_label_event (class_p=<optimized out>, e=<optimized out>) at lvgl-native-8.2.0/src/widgets/lv_label.c:734
#7  0x000055555555b90f in lv_obj_event_base (class_p=class_p@entry=0x0, e=e@entry=0x7fffffffd660) at lvgl-native-8.2.0/src/core/lv_event.c:98
#8  0x000055555555b9ea in event_send_core (e=e@entry=0x7fffffffd660) at lvgl-native-8.2.0/src/core/lv_event.c:458
#9  0x000055555555bbe4 in lv_event_send (obj=obj@entry=0x5555558fc6a0, event_code=event_code@entry=LV_EVENT_DRAW_MAIN, param=param@entry=0x5555558d4e50) at lvgl-native-8.2.0/src/core/lv_event.c:75
#10 0x0000555555569a7c in lv_refr_obj (draw_ctx=draw_ctx@entry=0x5555558d4e50, obj=0x5555558fc6a0) at lvgl-native-8.2.0/src/core/lv_refr.c:146
#11 0x0000555555569ae4 in lv_refr_obj (draw_ctx=draw_ctx@entry=0x5555558d4e50, obj=obj@entry=0x5555558fcbe0) at lvgl-native-8.2.0/src/core/lv_refr.c:180
#12 0x0000555555569b78 in lv_refr_obj_and_children (draw_ctx=draw_ctx@entry=0x5555558d4e50, top_obj=top_obj@entry=0x5555558fcbe0) at lvgl-native-8.2.0/src/core/lv_refr.c:720
#13 0x0000555555569cfd in lv_refr_area_part (draw_ctx=draw_ctx@entry=0x5555558d4e50) at lvgl-native-8.2.0/src/core/lv_refr.c:652
#14 0x000055555556a464 in lv_refr_area (area_p=<optimized out>) at lvgl-native-8.2.0/src/core/lv_refr.c:571
#15 lv_refr_areas () at lvgl-native-8.2.0/src/core/lv_refr.c:511
#16 _lv_disp_refr_timer (tmr=<optimized out>) at lvgl-native-8.2.0/src/core/lv_refr.c:313
#17 0x00005555555896a7 in lv_timer_exec (timer=0x5555558fb930) at lvgl-native-8.2.0/src/misc/lv_timer.c:313
#18 lv_timer_handler () at lvgl-native-8.2.0/src/misc/lv_timer.c:109
#19 0x00005555555585f5 in main (argc=<optimized out>, argv=<optimized out>) at main.c:124
(gdb) 
(gdb) l
46       * @param letter an UNICODE character code
47       * @return pointer to the bitmap of the letter
48       */
49      const uint8_t * lv_font_get_glyph_bitmap(const lv_font_t * font_p, uint32_t letter)
50      {
51          LV_ASSERT_NULL(font_p);
52          return font_p->get_glyph_bitmap(font_p, letter);
53      }
54
```

### Expected behavior
Never hang, use any fallback fonts available.
With the following patch it works.
```
--- a/src/draw/sw/lv_draw_sw_letter.c
+++ b/src/draw/sw/lv_draw_sw_letter.c
@@ -108,6 +108,9 @@ void lv_draw_sw_letter(lv_draw_ctx_t * draw_ctx, const lv_draw_label_dsc_t * dsc
         return;
     }
 
+    if (g.resolved_font == NULL) {
+      return;
+    }
     /*Don't draw anything if the character is empty. E.g. space*/
     if((g.box_h == 0) || (g.box_w == 0)) return;
 
diff --git a/src/font/lv_font.c b/src/font/lv_font.c
index 0e7a2fe74..bde55aa22 100644
--- a/src/font/lv_font.c
+++ b/src/font/lv_font.c
@@ -77,6 +77,9 @@ bool lv_font_get_glyph_dsc(const lv_font_t * font_p, lv_font_glyph_dsc_t * dsc_o
         }
         f = f->fallback;
     }
+    if (dsc_out->resolved_font == NULL) {
+      found = false;
+    }
     return found;
 }
```

### Screenshots or video

kisvegabor, 2022-03-31:
Please, try with the version from `master`. I think it was already fixed.

johado, 2022-03-31:
Thanks, that seems to work!
Any ETA on that fix ending up in a release (8.2.1 or 8.3.0) ?

johado, 2022-03-31:
One thing I noticed is that scrolling gets very slow if i initialize like 10 truetype fonts in both NORMAL and BOLD and use them all.
(fallbacks mainly to get asian and arabic glyphs as well)
Tested using SDL2 on a Linux machine. top says 1 to 1.3 % CPU usage, but occasionally spikes at 80%.
I haven't enabled any caching, so I'll experiment with that.

kisvegabor, 2022-04-04:
> Any ETA on that fix ending up in a release (8.2.1 or 8.3.0) ?

There is  no plan for v8.2.1 now. My guess is that v8.3 can be out in 2-3 month.

> One thing I noticed is that scrolling gets very slow if i initialize like 10 truetype fonts in both NORMAL and BOLD and use them all.
(fallbacks mainly to get asian and arabic glyphs as well)

@guoweilkd might know more about it.

guoweilkd, 2022-04-06:
> One thing I noticed is that scrolling gets very slow if i initialize like 10 truetype fonts in both NORMAL and BOLD and use them all.
> (fallbacks mainly to get asian and arabic glyphs as well)

If you use cache, it is recommended not to use the FT_FONT_STYLE_BOLD macro when create Freetype font. Bold font does not use caching, which severely affects performance. If you must use bold font, use bold font files(xxx.ttf) instead.