lvgl/lvgl #3242: How can I get lv_timer_Handler() to execute in a new thread

chuifengji, 2022-03-31:
### Introduce the problem

How can I get lv_timer_Handler() to execute in a new thread?
I packaged LVGL as a DLL for QuickJS, QuickJS is single-threaded, so 'While (1)' on the main thread blocks the JS code

thanks~

### Examples and cases

**I tried the following code：**

```
static void init_handler()
{
  while (1)
  {
    /* Periodically call the lv_task handler.
    * It could be done in a timer interrupt or an OS task too.*/
    lv_timer_handler();
    usleep(5 * 1000); 
  }
}

int main()
{

  /*Initialize LVGL*/
  lv_init();

  /*Initialize the HAL (display, input devices, tick) for LVGL*/
  hal_init();

  view_init();

  pthread_t pid;
  int res = pthread_create(&pid, NULL, init_handler, &pid);
}
```

**But an error was thrown：**

```Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'nextEventMatchingMask should only be called from the Main Thread!```
### Suggested solution

embeddedt, 2022-03-31:
It would be cleanest to fire `lv_timer_handler` from some sort of QuickJS-provided periodic timer. Since JavaScript is an event-driven language, there should be a way of doing this. If you try to put LVGL and the JavaScript runtime on separate threads, you will have to start dealing with mutexes as LVGL APIs could be called from both threads at once.

----------------------
We use GitHub issues for development-related discussions.
Please use the [forum](https://forum.lvgl.io/) to ask questions.

closed 2022-03-31 by embeddedt

chuifengji, 2022-04-01:
agree, thanks~